nextcloud/nextcloudpi #1649: ncp-restore:"tar: Unexpected EOF in archive" restore backup not possible NextcloudPi

jannissi, 2022-10-24:
Hi there,

i set up an new ncpi and want to restore my former data and files. 
Unfortunately it is not possible. I set up the new ncp, ran apt-update and apt-upgrade (following this instruction: https://help.nextcloud.com/t/how-to-restore-nextcloudpi-backup-using-web-panel/126312)
And this is what happens:

[ nc-restore ] (Mon Oct 24 06:57:38 BST 2022)
check free space...
tar: Unexpected EOF in archive
tar: Error is not recoverable: exiting now
extracting backup file /media/backup/nextcloud-bkp_20221022_1666404010.tar...

I also tried to restore using ncp-config.

Can you help me?

Thanks

jannissi, 2022-10-26:
Hello,

i still have this problem and i really want to restore my Nextcloud. So could please someone help me?

schoetju, 2022-10-28:
Hi Jannissi, 
looks like your backup-file is corrupted somehow. Is your old source-system is still available? If yes, please create a new backup.

In general: Please read [Forum Github Which to use when](https://help.nextcloud.com/t/forum-github-or-chat-which-to-use-when/126318)  Your issue is support-related (= Forum). It could turn into a bug-report (=Github) later on, if you/we find out, that the backup _always_ is corrupted e.g. under special configuration / circumstances. 

Anyway - good luck with your backup/restore! 
(FYI: I've been running into similar issues a few days ago when moving from 32 to 62 Bit RPI-setup. Luckily I could do a new backup on the old 32Bit-setup, that could be restored to the new 64Bit-Setup without any issues)