nextcloud/nextcloudpi #1650: Latest Docker image Creates a huge docker-startup-backups folder

luismauriz1, 2022-10-31:
Hi,
i updateded my NCP Docker Container to the latest version with PHP 8.1.
On two different Systems with Ubuntu 22.04 and X86 architecture I got the same issue after a day
After a reboot of the System it creates a "docker-startup-backups" folder in the /var/www folder. In this folder is a full backup of the system. This is too huge for my drive so it gets completly full. This causes a crash of the mysql service and i have to setup the container new. Is this a common problem or a feature ?
Thanks!

svenb1234, 2022-11-02:
That is #1609 which was amrked as wontfix.

luismauriz1, 2022-11-07:
Thanks! yes that would be great! Bc when I install a new nextcloud and disable backups then it creates them without my agree ;)