nextcloud/nextcloudpi #1652: Internal Server Error

jiyang1018, 2022-11-03:
I completed deleted the previous docker, and all the files left in appdata directory, pulled the latest, and started over. It worked for the past week until last night. This morning, all my nextcloud clients couldn't connect, and I believe the kill switch on my server got triggered; I restarted my unraid server. After all dockers were running, I got this error first
[image]
then it cycled between this and the previous one a little bit
[image]
now it gives only this:

```
Internal Server Error

The server encountered an internal error and was unable to complete your request.
Please contact the server administrator if this error reappears multiple times, please include the technical details below in your report.
More details can be found in the server log.
```

log seems to be a repeatition of this with different "reqId" and "time"
`{"reqId":"Y2QEMoVcE1MMqBJs2Nd3ZgAAAEY","level":3,"time":"2022-11-03T18:10:58+00:00","remoteAddr":"69.75.39.170","user":"--","app":"remote","method":"GET","url":"/status.php","message":"WRONGPASS invalid username-password pair","userAgent":"Mozilla/5.0 (Windows) mirall/3.4.0stable-Win64 (build 20211129) (Nextcloud, windows-10.0.19044 ClientArchitecture: x86_64 OsArchitecture: x86_64)","version":"24.0.5.1","exception":{"Exception":"RedisException","Message":"WRONGPASS invalid username-password pair","Code":0,"Trace":[{"file":"/data/nextcloud/lib/private/RedisFactory.php","line":143,"function":"auth","class":"Redis","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data/nextcloud/lib/private/RedisFactory.php","line":178,"function":"create","class":"OC\\RedisFactory","type":"->"},{"file":"/data/nextcloud/lib/private/Memcache/Redis.php","line":43,"function":"getInstance","class":"OC\\RedisFactory","type":"->"},{"file":"/data/nextcloud/lib/private/Memcache/Factory.php","line":118,"function":"__construct","class":"OC\\Memcache\\Redis","type":"->"},{"file":"/data/nextcloud/lib/private/Server.php","line":1106,"function":"createLocking","class":"OC\\Memcache\\Factory","type":"->"},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":162,"function":"OC\\{closure}","class":"OC\\Server","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php","line":122,"function":"OC\\AppFramework\\Utility\\{closure}","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":129,"function":"offsetGet","class":"Pimple\\Container","type":"->"},{"file":"/data/nextcloud/lib/private/ServerContainer.php","line":136,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":57,"function":"query","class":"OC\\ServerContainer","type":"->"},{"file":"/data/nextcloud/lib/private/Server.php","line":2082,"function":"get","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/data/nextcloud/lib/private/Files/View.php","line":122,"function":"getLockingProvider","class":"OC\\Server","type":"->"},{"file":"/data/nextcloud/lib/private/Server.php","line":454,"function":"__construct","class":"OC\\Files\\View","type":"->"},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":162,"function":"OC\\{closure}","class":"OC\\Server","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php","line":122,"function":"OC\\AppFramework\\Utility\\{closure}","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":129,"function":"offsetGet","class":"Pimple\\Container","type":"->"},{"file":"/data/nextcloud/lib/private/ServerContainer.php","line":136,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":57,"function":"query","class":"OC\\ServerContainer","type":"->"},{"file":"/data/nextcloud/lib/private/Server.php","line":1445,"function":"get","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/data/nextcloud/lib/base.php","line":602,"function":"boot","class":"OC\\Server","type":"->"},{"file":"/data/nextcloud/lib/base.php","line":1111,"function":"init","class":"OC","type":"::"},{"file":"/data/nextcloud/status.php","line":37,"args":["/data/nextcloud/lib/base.php"],"function":"require_once"}],"File":"/data/nextcloud/lib/private/RedisFactory.php","Line":143,"CustomMessage":"--"}}`

I tried status.php, and it returned:

```
Bad Request
Your browser sent a request that this server could not understand.
Reason: You're speaking plain HTTP to an SSL-enabled server port.
Instead use the HTTPS scheme to access this URL, please.
```

Please let me know what info is needed to solve this issue. Thanks in advance.

monsieurhannes, 2022-11-15:
Can you run this in your CLI: `docker exec -it nextcloudpi ncp-report`? You can post the output here.

jiyang1018, 2022-11-16:
> Can you run this in your CLI: `docker exec -it nextcloudpi ncp-report`? You can post the output here.

```
<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.50.2
NextCloudPi image    NextCloudPi_docker_09-29-22
OS                   Debian GNU/Linux 11. 5.19.14-Unraid (x86_64)
automount            no
USB devices          sda 
datadir              /data/nextcloud/data
data in SD           no
data filesystem      fuseblk
data disk usage      30T/52T
rootfs usage         9.1G/30G
swapfile             none
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    24.0.5.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
public IP            ***REMOVED SENSITIVE VALUE***
public IP            ***REMOVED SENSITIVE VALUE***
Port check 80        closed
Port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               12days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```
```

ZendaiOwl, 2023-01-15:
It looks like something has happened with the redis server, the username & password doesn't match anymore

hyphenc, 2023-02-25:
Hi all,

I think the same thing happened to me today when I tried upgrading my ncp v1.50.1 container running nextcloud 24.0.8.
(First I upgraded the container and then nextcloud within the container. This method probably isn't the proper way to do it but anyhow...)

After upgrading I tried accessing my nextcloud and it sporadically threw HTTP 500s at me, when I looked at the logs I found a bunch of these: RedisException: WRONGPASS and NOAUTH (see attached logs).

It seems Redis changed their auth scheme in v6.x and a 'default' user now exists. From what I've read when this user is not supplied the 'default' user should be implicitly used, but apparently that doesn't happen or otherwise I wouldn't be getting these auth errors.

Then I looked into the [source code](https://github.com/nextcloud/server/blob/2ff0c972c98a14f4dbd81d7923b119770600918f/lib/private/RedisFactory.php#L77) and it seems that nextcloud _can_ in fact send a username, I just didn't have that parameter set in my `config.php` because it wasn't needed before.

working (_for me_) redis conf:
```php
'redis' =>                           
array (                                                                 
    'host' => '/var/run/redis/redis.sock',
    'user' => 'default',     
    'port' => 0,
    'timeout' => 0.0,
    'password' => 'PASSWORD',
)
```

I hope this helps.

Relevant logs:

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.51.0
NextCloudPi image    NextcloudPi_docker_01-22-23
OS                   Debian GNU/Linux 11. 5.15.0-58-generic (x86_64)
automount            no
USB devices          none
datadir              /data/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      35G/75G
rootfs usage         35G/75G
swapfile             none
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    25.0.3.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
public IP            ***REMOVED SENSITIVE VALUE***
public IP            ***REMOVED SENSITIVE VALUE***
Port check 80        closed
Port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               24days

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"Y_ocjeWftqlVc_qRgKM2tQAAAEQ","level":3,"time":"2023-02-25T14:34:53+00:00","remoteAddr":"172.17.0.1","user":"--","app":"core","method":"GET","url":"/avatar/ncp/64?v=11","message":"WRONGPASS invalid username-password pair","userAgent":"Mozilla/5.0 (X11; Linux x86_64; rv:109.0) Gecko/20100101 Firefox/110.0","version":"25.0.3.2","exception":{"Exception":"RedisException","Message":"WRONGPASS invalid username-password pair","Code":0,"Trace":[{"file":"/data/nextcloud/lib/private/RedisFactory.php","line":143,"function":"auth","class":"Redis","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data/nextcloud/lib/private/RedisFactory.php","line":178,"function":"create","class":"OC\\RedisFactory","type":"->"},{"file":"/data/nextcloud/lib/private/Memcache/Redis.php","line":43,"function":"getInstance","class":"OC\\RedisFactory","type":"->"},{"file":"/data/nextcloud/lib/private/Memcache/Factory.php","line":118,"function":"__construct","class":"OC\\Memcache\\Redis","type":"->"},{"file":"/data/nextcloud/lib/private/Server.php","line":1120,"function":"createLocking","class":"OC\\Memcache\\Factory","type":"->"},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":162,"function":"OC\\{closure}","class":"OC\\Server","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php","line":122,"function":"OC\\AppFramework\\Utility\\{closure}","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":129,"function":"offsetGet","class":"Pimple\\Container","type":"->"},{"file":"/data/nextcloud/lib/private/ServerContainer.php","line":136,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":57,"function":"query","class":"OC\\ServerContainer","type":"->"},{"file":"/data/nextcloud/lib/private/Server.php","line":2070,"function":"get","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/data/nextcloud/lib/private/Files/View.php","line":117,"function":"getLockingProvider","class":"OC\\Server","type":"->"},{"file":"/data/nextcloud/lib/private/Server.php","line":462,"function":"__construct","class":"OC\\Files\\View","type":"->"},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":162,"function":"OC\\{closure}","class":"OC\\Server","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php","line":122,"function":"OC\\AppFramework\\Utility\\{closure}","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer. [... cut ...] :"::"},{"file":"/data/nextcloud/lib/private/legacy/OC_Template.php","line":78,"function":"initTemplateEngine","class":"OC_Template","type":"::"},{"file":"/data/nextcloud/lib/private/legacy/OC_Template.php","line":323,"function":"__construct","class":"OC_Template","type":"->"},{"file":"/data/nextcloud/index.php","line":74,"function":"printExceptionErrorPage","class":"OC_Template","type":"::"}],"File":"/data/nextcloud/lib/private/Memcache/Redis.php","Line":48,"CustomMessage":"--"}}

```
</details>

feutl, 2023-03-14:
Are those two somehow releated, I do have the same issues and struggling in finding the reason, but if someone helps, perhaps I can provide some information.
https://github.com/nextcloud/nextcloudpi/issues/1740#issuecomment-1466188049

feutl, 2023-03-22:
@hyphenc I tried to add the "default" user but it did not change much, actually nothing.
How did you came up with "default" as the username and was there anything else to do? I am really struggling in getting this fixed, especially I do not have a clue why this even popped up and it does create it kind of problems on my instance for all users.

Thanks

feutl, 2023-03-28:
After a reboot of my NCP docker I have the same issue again.
The docker installation changes the redis password every reboot, so I assume that this messes things up if the user is being set because even with the default user I get the error again :(

kozfelipe, 2023-04-10:
Hey guys. Im kinda dumb using dockers. How can I edit those confs if docker's editors are not available (neither its repos)