nextcloud/nextcloudpi #1655: Autoupload from android app suddenly stops

Correntable, 2022-11-07:
**System information**
NextCloudPi image    NextCloudPi_RaspberryPi_v1.50.3.img
OS                   Debian GNU/Linux 11. 5.10.92-v8+ (aarch64)
automount            no
USB devices          none
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      4.3G/59G
rootfs usage         4.3G/59G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    24.0.5.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
public IP            ***REMOVED SENSITIVE VALUE***
public IP            ***REMOVED SENSITIVE VALUE***
Port check 80        closed
Port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
Interface            wlan0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               1:47

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "7": "nextcloudpi",
            "5": "nextcloudpi.local",
            "8": "nextcloudpi.lan",
            "3": "correntrpi.hopto.org",
            "11": "93.42.69.117",
            "1": "192.168.1.64",
            "14": "nextcloudpi"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "24.0.5.1",
        "overwrite.cli.url": "https:\/\/correntrpi.hopto.org\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
        },
        "tempdirectory": "\/var\/www\/nextcloud\/data\/tmp",
        "mail_smtpmode": "sendmail",
        "mail_smtpauthtype": "LOGIN",
        "mail_from_address": "***REMOVED SENSITIVE VALUE***",
        "mail_domain": "***REMOVED SENSITIVE VALUE***",
        "preview_max_x": "2048",
        "preview_max_y": "2048",
        "jpeg_quality": "60",
        "overwriteprotocol": "https",
        "trusted_proxies": "***REMOVED SENSITIVE VALUE***",
        "htaccess.RewriteBase": "\/",
        "maintenance": false

I recently installed NextcloudPi on my Reaspberry Pi 4 Model B and setted the autoupload of my picture folders on my Pixel 4a with LineageOS 19.1, Nextcloud app 3.22.3. The upload started regularly but when it reached 500MB of items uploaded crashed and the app report "Server unavailable". Checking from web browser denies this: the server is up and running but not reachable from my phone. I noticed that rebooting my phone makes the upload to restart but after few MB uploaded it crashes again.

Jannik44, 2022-11-13:
Im pretty sure thats because the phone sets the app in energysaving mode, you can disable that in the app settings

joshtrichards, 2025-05-31:
This is an old Issue. Are you still experiencing this behavior?