nextcloud/nextcloudpi #1657: SSH root on Odroid HC4 not working

youwrote, 2022-11-09:
Hi,

I am using NCP on an Odroid HC4 and cannot activate the root user for SSH anymore. I tried to activate it via ncp-web to do some maintenance recently, as I have done before, but ncp-web won't let me set the root user. I also cannot create any other user for SSH via ncp-web. In an earlier issue linked below (1), nacho advised on using root as SSH user with Odroid. 

I cannot exactly tell since when this issue is occurring. Below, I also give some info on my setup. If you need more info, I'll be happy to provide it. I put this here as an issue and not on the forums because of reference (1). If it doesn't belong here, please lmk.

`[ SSH ] (Wed Nov 9 09:34:19 UTC 2022)
Refusing to use the root user for SSH. It's insecure`

(1) https://github.com/nextcloud/nextcloudpi/issues/711#issuecomment-436836624

Info on setup:
NextCloudPi version | v1.50.3
NextCloudPi image | NextCloudPi_11-08-21
OS | Armbian 21.08.3 Buster \l . 5.10.60-meson64 (aarch64)
auto updates enabled
original installation through softy

ABUCKY0, 2022-11-26:
In the cloud client thingy, there is a panel to enable the SSH user password thing. Upon enabling that, sign in through SSH and change your password when prompted. Then you should be good to go.

gilbert-grape, 2023-01-23:
> In the cloud client thingy, there is a panel to enable the SSH user password thing. Upon enabling that, sign in through SSH and change your password when prompted. Then you should be good to go.

Where exactly can I change the user? What is cloud client thingy?

guentercoffee, 2023-01-23:
> > In the cloud client thingy, there is a panel to enable the SSH user password thing. Upon enabling that, sign in through SSH and change your password when prompted. Then you should be good to go.
> 
> Where exactly can I change the user? What is cloud client thingy?

This proposed solution from @ABUCKY0 doesn't work, since you can't set a password for user "root" via the web interface, nor can you create another user in that manner. This is also why I didn't mark this issue as solved after his answer.

youwrote, 2023-01-23:
> > In the cloud client thingy, there is a panel to enable the SSH user password thing. Upon enabling that, sign in through SSH and change your password when prompted. Then you should be good to go.
> 
> Where exactly can I change the user? What is cloud client thingy?

This proposed solution from @ABUCKY0 doesn't work, since you can't set a password for user "root" via the web interface, nor can you create another user in that manner. This is also why I didn't mark this issue as solved after his answer. 

I wrote my reply from the wrong account earlier, sorry.

closed 2023-01-23 by theCalcaholic