nextcloud/nextcloudpi #1662: Unable to restore NC23 backup NCP 1.49.1 to NCP 1.5.02  - not compatible with > PHP 8.0

Bobmorton-TX, 2022-11-14:
I am trying to restore a backup from NCP 1.49.1 NC 23.0.9.1 to NCP 1.50.2
the restore works but I am getting this error message in NCP-backend and when trying to access NC:
`This version of Nextcloud is not compatible with > PHP 8.0.
You are currently running 8.1.10.`

I am unable to upgrade NCP 1.49.1 any further due to old distribution
when trying to upgrade this docker I get
`WARNING: Docker images should be updated by replacing the container from the latest docker image (refer to the documentation for instructions: https://docs.nextcloudpi.com). If you are sure that you know what you are doing, you can still execute the update script by running it like this:`

OLD DOCKER

```

NextCloudPi version  v1.49.1
NextCloudPi image    NextCloudPi_10-07-21
OS                   Debian GNU/Linux 10. 4.2.8 (aarch64)
automount            no
USB devices          sdg 
datadir              /data/nextcloud/data
data in SD           yes
data filesystem      overlayfs
data disk usage      1.7T/2.9T
rootfs usage         1.7T/2.9T
swapfile             /share/CE_CACHEDEV1_DATA/.swap/qnap_swap
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    23.0.9.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
Internet check       ok
public IP            ***REMOVED SENSITIVE VALUE***
public IP            ***REMOVED SENSITIVE VALUE***
Port check 80        closed
Port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              192.168.40.1
Interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               2days

NEW DOCKER
<summary>NextCloudPi diagnostics</summary>

NextCloudPi version  v1.50.2
NextCloudPi image    NextCloudPi_docker_09-29-22
OS                   Debian GNU/Linux 11. 4.2.8 (aarch64)
automount            no
USB devices          sdg 
datadir              /data/nextcloud/data
data in SD           yes
data filesystem      overlayfs
data disk usage      1.7T/2.9T
rootfs usage         1.7T/2.9T
swapfile             /share/CE_CACHEDEV1_DATA/.swap/qnap_swap
dbdir                /data/database
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
public IP            ***REMOVED SENSITIVE VALUE***
public IP            ***REMOVED SENSITIVE VALUE***
Port check 80        closed
Port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               2days

```

monsieurhannes, 2022-11-15:
You have to replace (delete) the old docker container. They can't "live" side by side. 
Basically, you start the docker container with `v1.49.1`, load the backup, run some updates in the container I guess to get NC to the latest version possible.
Run `docker stop nextcloudpi`, `docker rm nextcloudpi`, `docker pull ownyourbits/nextcloudpi:latest` and then you start it again.

Is that the way you did it, or why is it that both of your container above have a uptime of 2 days?

Bobmorton-TX, 2022-11-15:
I was testing the update procedure as the site needs to stay online. so several docker independent containers were running. 

I was able to create a backup of the 
` NextCloudPi version  v1.49.1
NextCloudPi image    NextCloudPi_10-07-21
OS                   Debian GNU/Linux 10. 4.2.8 (aarch64)
`
and import this into new docker with v1.49.1 run NC updates and then create backup again and import into docker with v1.50.2

monsieurhannes, 2022-11-15:
But you have two instances connected to the same database simultaneously? I guess you'll always have a short downtime during the update unless you run everything (database and files) as identical copies...

Can you run `ncp-report` in the container?

Bobmorton-TX, 2022-11-15:
I have two dockers running both with their own database and files within the container.
sure a short down time is fine. 
the problem seems to be the different PHP versions in the docker containers and the requirements of the different NC versions. I think with my described way above I can proceed even though this does not seem optimal

theCalcaholic, 2022-12-17:
Backups need to be restored with a compatible NCP version to the one they were created from (first and second version number must be equal). With incompatible versions, it *can* work but is not recommended.
After restoring the backup you can follow the usual update process

closed 2022-12-17 by theCalcaholic