nextcloud/nextcloudpi #1663: NextCloudPi not yet initialized, trying again in a few seconds

PeterPilley, 2022-11-22:
I have been through all the open and closed tickets and the forum but dont have an answer so am opening a new issue

**System information**
Docker version 20.10.21, build baeda1f
docker-compose version 1.29.2, build 5becea4c

Docker Images tested 1.50.2, 1.50, 1.49,1.48.3

I previously had this running with 1.50.2 and then had to do a redeploy on docker now all I received is the not yet initialised message.

I have tried:
* Accessing the mariadb server within the image, access works, tables and database exist
* I have tried using a standard docker volume as opposed to a local folder for ncdata and same result
* I have used a mapping to nextcloudpi.local

I have run ncp-update within the image (allowing the update) and can see there is an udpate there to 1.50.3 (is there an updated docker image somewhere?)

Checking the activation page I can see multiple requests to ncp-launcher.php and replies (200).

action: 
launch
ref: 
nc-admin
config: 
{ "PASSWORD":"*","CONFIRM" :"**","USER"    : "ncp" }
csrf_token: 
*

and response { "token": "iPQba5a4SVRnD+H/H6FsX2Cr0Q72UWIv0I3P2l4W0XE=", "ref": "nc-admin", "output": "" ,  "ret": "1" }

I can see an error in apache2/error.log regarding the access provisions set in the apache2 conf file

access check of 'localhost' to /ncp-launcher.php failed

I am stuck I think I have tried everything, Any ideas?

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.50.2
NextCloudPi image    NextCloudPi_docker_09-29-22
OS                   Debian GNU/Linux 11. 5.4.0-125-generic (x86_64)
automount            no
USB devices          sdd 
datadir              /data/nextcloud/data
data in SD           no
data filesystem      ext2/ext3
data disk usage      582G/916G
rootfs usage         72G/136G
swapfile             /swapfile
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    24.0.5.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       no
public IP            ***REMOVED SENSITIVE VALUE***
public IP            ***REMOVED SENSITIVE VALUE***
Port check 80        closed
Port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               85days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "7": "nextcloudpi",
            "5": "nextcloudpi.local",
            "8": "nextcloudpi.lan",
            "3": "c2ab77cf152f",
            "11": "206.83.103.115",
            "1": "172.80.19.2",
            "14": "c2ab77cf152f"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "24.0.5.1",
        "overwrite.cli.url": "https:\/\/c2ab77cf152f\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
        },
        "tempdirectory": "\/var\/www\/nextcloud\/data\/tmp",
        "mail_smtpmode": "sendmail",
        "mail_smtpauthtype": "LOGIN",
        "mail_from_address": "***REMOVED SENSITIVE VALUE***",
        "mail_domain": "***REMOVED SENSITIVE VALUE***",
        "preview_max_x": "2048",
        "preview_max_y": "2048",
        "jpeg_quality": "60",
        "overwriteprotocol": "https"
    }
}

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Wed Nov 23 10:56:14.926730 2022] [authz_host:error] [pid 130:tid 140060549105408] [client 172.80.19.1:37516] AH01753: access check of 'localhost' to /img/favicon.png failed, reason: unable to get the remote host name
[Wed Nov 23 10:56:26.184628 2022] [authz_host:error] [pid 130:tid 140060549105408] [client 172.80.19.1:52922] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Nov 23 10:56:36.266613 2022] [authz_host:error] [pid 130:tid 140060549105408] [client 172.80.19.1:39208] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Nov 23 10:56:46.704259 2022] [authz_host:error] [pid 130:tid 140060549105408] [client 172.80.19.1:59874] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Nov 23 10:56:57.136345 2022] [authz_host:error] [pid 129:tid 140060515518208] [client 172.80.19.1:47682] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Nov 23 10:57:07.248116 2022] [authz_host:error] [pid 129:tid 140060515518208] [client 172.80.19.1:55138] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Nov 23 10:57:17.340687 2022] [authz_host:error] [pid 129:tid 140060515518208] [client 172.80.19.1:47216] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Nov 23 10:57:27.425251 2022] [authz_host:error] [pid 130:tid 140060549105408] [client 172.80.19.1:60448] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Nov 23 10:57:37.523957 2022] [authz_host:error] [pid 130:tid 140060549105408] [client 172.80.19.1:47640] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Nov 23 10:57:47.944559 2022] [authz_host:error] [pid 130:tid 140060549105408] [client 172.80.19.1:33126] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Nov 23 10:57:58.374916 2022] [authz_host:error] [pid 130:tid 140060549105408] [client 172.80.19.1:35890] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Nov 23 10:58:08.460437 2022] [authz_host:error] [pid 130:tid 140060549105408] [client 172.80.19.1:57020] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Nov 23 10:58:18.558789 2022] [authz_host:error] [pid 130:tid 140060549105408] [client 172.80.19.1:40038] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Nov 23 10:58:28.974685 2022] [authz_host:error] [pid 130:tid 140060549105408] [client 172.80.19.1:34026] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Nov 23 10:58:39.058942 2022] [authz_host:error] [pid 129:tid 140060515518208] [client 172.80.19.1:34140] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Nov 23 10:58:49.161254 2022] [authz_host:error] [pid 130:tid 140060549105408] [client 172.80.19.1:57460] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Nov 23 10:58:59.580072 2022] [authz_host:error] [pid 130:tid 140060549105408] [client 172.80.19.1:46670] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Nov 23 10:59:10.026599 2022] [authz_host:error] [pid 129:tid 140060515518208] [client 172.80.19.1:33526] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Nov 23 10:59:20.114348 2022] [authz_host:error] [pid 129:tid 140060515518208] [client 172.80.19.1:40334] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Nov 23 10:59:30.506882 2022] [authz_host:error] [pid 130:tid 140060549105408] [client 172.80.19.1:59576] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name

```
</details>

<details>
<summary>Database logs</summary>

```

tail: cannot open '/var/log/mysql/*.log' for reading: No such file or directory

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

```
</details>

OfficialMuffin, 2022-11-28:
Having a similar issue here. Unable to successfully activate NextcloudPi.
```

Doctrine\DBAL\Exception: Failed to connect to the database: An exception occurred in the driver: SQLSTATE[HY000] [2002] No such file or directory in /data-ro/nextcloud/lib/private/DB/Connection.php:139
Stack trace:
#0 /data-ro/nextcloud/3rdparty/doctrine/dbal/src/Connection.php(1519): OC\DB\Connection->connect()
#1 /data-ro/nextcloud/3rdparty/doctrine/dbal/src/Connection.php(1041): Doctrine\DBAL\Connection->getWrappedConnection()
#2 /data-ro/nextcloud/lib/private/DB/Connection.php(261): Doctrine\DBAL\Connection->executeQuery()
#3 /data-ro/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php(345): OC\DB\Connection->executeQuery()
#4 /data-ro/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php(281): Doctrine\DBAL\Query\QueryBuilder->execute()
#5 /data-ro/nextcloud/lib/private/AppConfig.php(411): OC\DB\QueryBuilder\QueryBuilder->execute()
#6 /data-ro/nextcloud/lib/private/AppConfig.php(176): OC\AppConfig->loadConfigValues()
#7 /data-ro/nextcloud/lib/private/AppConfig.php(367): OC\AppConfig->getApps()
#8 /data-ro/nextcloud/lib/private/legacy/OC_App.php(983): OC\AppConfig->getValues()
#9 /data-ro/nextcloud/lib/private/Server.php(713): OC_App::getAppVersions()
#10 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(162): OC\Server->OC\{closure}()
#11 /data-ro/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(122): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#12 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(129): Pimple\Container->offsetGet()
#13 /data-ro/nextcloud/lib/private/ServerContainer.php(136): OC\AppFramework\Utility\SimpleContainer->query()
#14 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(57): OC\ServerContainer->query()
#15 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(184): OC\AppFramework\Utility\SimpleContainer->get()
#16 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(162): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#17 /data-ro/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#18 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(129): Pimple\Container->offsetGet()
#19 /data-ro/nextcloud/lib/private/ServerContainer.php(136): OC\AppFramework\Utility\SimpleContainer->query()
#20 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(57): OC\ServerContainer->query()
#21 /data-ro/nextcloud/lib/private/Server.php(1105): OC\AppFramework\Utility\SimpleContainer->get()
#22 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(162): OC\Server->OC\{closure}()
#23 /data-ro/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(122): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#24 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(129): Pimple\Container->offsetGet()
#25 /data-ro/nextcloud/lib/private/ServerContainer.php(136): OC\AppFramework\Utility\SimpleContainer->query()
#26 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(57): OC\ServerContainer->query()
#27 /data-ro/nextcloud/lib/private/Server.php(2082): OC\AppFramework\Utility\SimpleContainer->get()
#28 /data-ro/nextcloud/lib/private/Files/View.php(122): OC\Server->getLockingProvider()
#29 /data-ro/nextcloud/lib/private/Server.php(454): OC\Files\View->__construct()
#30 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(162): OC\Server->OC\{closure}()
#31 /data-ro/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(122): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#32 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(129): Pimple\Container->offsetGet()
#33 /data-ro/nextcloud/lib/private/ServerContainer.php(136): OC\AppFramework\Utility\SimpleContainer->query()
#34 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(57): OC\ServerContainer->query()
#35 /data-ro/nextcloud/lib/private/Server.php(1445): OC\AppFramework\Utility\SimpleContainer->get()
#36 /data-ro/nextcloud/lib/base.php(602): OC\Server->boot()
#37 /data-ro/nextcloud/lib/base.php(1111): OC::init()
#38 /data-ro/nextcloud/console.php(48): require_once('...')
#39 /data-ro/nextcloud/occ(11): require_once('...')
#40 {main}

```

The-XOR, 2022-12-30:
I literally spent hours, and these are my conclusions:
At least on raspberry OS (also tried dietpi, same issue) the redis service does NOT start because the redis-server.pid file has an insecure mask, 677 instead of 644.

I modified the  /etc/init.d/redis-server file (insider the container) and added the line:
		chmod 644 $PIDFILE        
after the line with "chown redis:redis $RUNDIR $PIDFILE"
(to clarify, the file appears like:

       case "$1" in
		start)
		echo -n "Starting $DESC: "
		mkdir -p $RUNDIR
		touch $PIDFILE
		chown redis:redis $RUNDIR $PIDFILE
		chmod 644 $PIDFILE        <------- ADD THIS LINE <-----------
		chmod 755 $RUNDIR

Restarted the container, crossed fingers... everything works.

PeterPilley, 2023-02-26:
Seeing this same issue with latest docker image, tried the solution above and unfortunately I dont believe it is a redis-server issue.

Confirmed redis-server was running before trying solution

PeterPilley, 2023-02-27:
Have made some headway today, the only way I could continue was to remove ncp-activation.conf and move ncp.conf and nextcloud.conf into the apache2/sites-enabled.

I tried following the direction of 
* authz require ip 127.0.0.1 (No luck)
* I checked the write permissions of everything comparing it to a known good
* I recreated multiple times the docker volume 

Any help is really appreciated as I am sure I am not the only one having this issue

NoanWasTaken, 2023-10-24:
Did someone found the solution ? I'm still having this problem here.