nextcloud/nextcloudpi #1665: NextCloud GUI blocked when I download a file with IDM (Internet Download Manager)

luke6464, 2022-11-24:
Hi I’m using NextCloud (NextCloudPi 1.50.2 - NC 24.0.5) installed on Rasperry Pi4 with Docker and Portainer.
When I try to download a file with IDM (Internet Download Manager) as soon as the download starts, the NextCloud GUI remains blocked for all users for a long time. If I download the files without IDM everything works perfectly. Is it possible that the block is caused by the overload of the multiple connections that IDM creates to download the file? … is there a way to set the NextCloud server to prevent or limit multiple connections on the same file?

I tried to change several php-fpm configuration parameters:
pm = static
pm.max_children = 100
pm.start_servers = 12
pm.min_spare_servers = 6
pm.max_spare_servers = 18

but the problem has not been solved.
At this point I think it's a NextCloudPi bug or a wrong configuration of the NextCloud server that I can't understand.
I must clarify that I am not an advanced user of Linux!
Thanks for your help.

sunjam, 2022-11-26:
You'd need to provide logs to know if the issue is with NextcloudPi, so [please post to the support forum and fill out the template](https://help.nextcloud.com/c/support/7).  **But...**
Since it happens using IDM my guess is your internet connection simply cannot handle it.  Try a [speed test](https://librespeed.org/).  Also consider whether you are using a really slow ncp disk (sd card) or something faster (SSD) in addition to a slow wifi card (2.4g) vs something faster (gb lan)

luke6464, 2022-11-27:
> You'd need to provide logs to know if the issue is with NextcloudPi, so [please post to the support forum and fill out the template](https://help.nextcloud.com/c/support/7). **But...** Since it [...]

Hi, it is impossible that it is the internet connection since I have a 1000 Mbit/s professional fiber connection with guaranteed minimum bandwidth and with other connected servers that manage hundreds of simultaneous connections ... the same RasberryPi 4 on which it is also installed NextCloud with the FTP service is able to transfer files at a speed of 50-60MByte/s ... so I would say that the problem only concerns NextCloud ...
tanks

sunjam, 2022-11-28:
What is your boot disk?  SSD or SD card or thumb drive?

Also, IDM would be run off a client device, yes?  You are possibly saturating that device (wifi 2.4g or 5g) as opposed to the rest of the network

Since IDM is making nextcloud unresponsive, try taking some networking stats locally on client device.

luke6464, 2022-11-29:
> What is your boot disk? SSD or SD card or thumb drive?
>
> Also, IDM would be run off a client device, yes? You are possibly saturating that device (wifi 2.4g or 5g) as opposed to the rest of the [...]

Nextcloud runs on Raspberry Pi4 8GB Ram on SD Card boot device.
The Client PC from which I log in is an Intel i9 with 64GB RAM with a 1Gbit LAN port but connected on a 10Gbit wired LAN ... the whole LAN is wired with professional connections capable of supporting up to 10Gbit of data transfer. When the Nextcloud GUI crashes only Nextcloud crashes...all the rest of the network and clients are working fine and I'm able to transfer with other services without problems.
For this reason I am increasingly convinced that it is a specific problem that NextCloud has only with IDM and that it does not depend on other problems related to the network or connected devices.
Tanks

luke6464, 2022-11-29:
I performed other tests and I noticed that during the download with IDM the transfer speed is always high and if at the same time I perform other downloads with the FTP service from the same device (Raspberry) everything works perfectly, both the download with FTP and the one with IDM they proceed in parallel at the maximum possible speed without any system blockage ... the only thing that gets stuck is only the Nextcloud GUI interface.
For this reason I am increasingly convinced that it is some specific Nextcloud BUG with IDM.
Everything else works smoothly even when transferring files using other services at the same time. The Raspberry never crashes… only the NextCloud GUI crashes.
Tanks

sunjam, 2022-11-29:
Internet Download Manager has nothing to do with nextcloudpi or nextcloud.
It is a browser plugin and windows app.

sunjam, 2022-11-29:
> yes that's what I thought too initially but then seeing that only the
> Nextcloud GUI crashes when IDM starts talking to the Nextcloud server while
> everything else continues to function normally I don't know what else to

How exactly is IDM talking to the nextcloud server? Can you give a specific
example.

luke6464, 2022-11-29:
The example is the following: a NextCloud user normally accesses the NextCloud web interface by logging in then chooses a file to download and after clicking on the file the connection is intercepted by IDM which starts multiple connections to download the file. As soon as IDM starts, the NextCloud GUI hangs for all users until IDM ends or even if IDM is stopped first, the GUI hangs for several minutes. Meanwhile any other type of access to other services on the Raspberry (ex. SMB, FTP) for example with openmediavault continue to work perfectly without any significant slowdown. This is the strangest thing! ... the Raspberry System continues to work efficiently for the other hosted services.
tanks

theCalcaholic, 2023-02-07:
Possibly related to #1624 

To be honest, though, this is more of a support request for a very specific hardware/software configuration that we frankly don't have the capacity to invest into.
If you find a solution that is related to a change in NCP, I'd love to know about it in order to incorporate it, but I can't provide any in-depth testing/troubleshooting myself.

closed 2023-02-07 by theCalcaholic

uranus829, 2023-12-28:
Hello, has the problem with idm and nexcloud been resolved? I have this problem too.