nextcloud/nextcloudpi #1670: Allow RC updates

darguez, 2022-12-03:
When I try to update nextcloud to a RC version, there is an error due to the version does not have only numbers:

25.0.1 is fine
25.0.2rc2 shows the error:
```
Current Nextcloud version 25.0.1.1
Available Nextcloud version 25.0.2rc2
/usr/local/etc/library.sh: line 429: [: 2rc2: integer expression expected
/usr/local/etc/library.sh: line 429: [: 2rc2: integer expression expected
Download Nextcloud 25.0.2rc2...
Error downloading
Clean up...
```

sunjam, 2022-12-06:
Spelling might differ. Check their github releases. Might include 25.0.2RC2 or some other variation

darguez, 2022-12-06:
I don't think it is a misspelling issue. Look at the log, an integer is expected in the last part of the version because a numeric comparison is made to determine if the version is newer than current version.

ZendaiOwl, 2023-01-15:
Correct, the library.sh file doesn't work with rc versions, it isn't written for that unfortunately

its0nlyfun, 2024-01-29:
I currently have this error message, too. I'm not really sure if the system is updating properly or not after that.
>[ nc-update ] (Mon Jan 29 07:46:27 GMT 2024)
> Downloading updates
> Performing updates
> /usr/local/etc/library.sh: line 494: [: 0sh: integer expression expected
> /usr/local/etc/library.sh: line 494: [: 0sh: integer expression expected
> /usr/local/etc/library.sh: line 494: [: 0sh: integer expression expected
> Nextcloud is already latest version

theCalcaholic, 2024-02-01:
@its0nlyfun That's a separate issue and should be fixed with 1.53.1 :)