nextcloud/nextcloudpi #1671: Nextcloud Pi - Blank screen - Docker version

Tchiimy, 2022-12-03:
Hello,

I'm running NextcloudPi (docker ARM version), on a RPI4.

After the update of this container it seems that the NextcloudPi interface is "blank' : 

[image]

It seems to be something similar to [this](https://github.com/nextcloud/nextcloudpi/issues/287#issuecomment-343726717) but I don't know if someone had the same problem with the docker version.

Any ideas ?

sunjam, 2022-12-09:
Did you fix you linked above solve your problem?

Tchiimy, 2022-12-09:
> Did you fix you linked above solve your problem?
> 
>     * Or, try reloading the browser.
> 
>     * Or, try clearing your cache.
> 
> 
> Personally, I never trust the webui and only use `ncp-config` directly over ssh, but I'm old school like that.

Yeah I tried these bit it didn't worked for me, so I directly stop the maintenance mode of Nextcloud by modifing the maintenance mode in the config.php file :

` sudo nano /portainer/Files/AppData/Config/ncdata/nextcloud/config/config.php`

`'maintenance' => false`

But I'm curious to see if others have the same problem as me or no !

powerflo, 2023-01-06:
The same issue came up when I tested v1.50.5 (thecalcaholic/ncp-internal-arm64:3851423807-testing).

In the logs I found the following error message:
`
[Fri Jan 06 16:04:55.354686 2023] [proxy_fcgi:error] [pid 331:tid 281473324396928] [client 192.168.2.212:61945] AH01071: Got error 'PHP message: PHP Warning:  scandir(/usr/local/bin/ncp/ncp-backup): Failed to open directory: Not a directory in /var/www/ncp-web/elements.php on line 178PHP message: PHP Warning:  scandir(): (errno 20): Not a directory in /var/www/ncp-web/elements.php on line 178PHP message: PHP Fatal error:  Uncaught TypeError: array_diff(): Argument #1 ($array) must be of type array, bool given in /var/www/ncp-web/elements.php:178\nStack trace:\n#0 /var/www/ncp-web/elements.php(178): array_diff()\n#1 /var/www/ncp-web/index.php(236): print_sidebar()\n#2 {main}\n  thrown in /var/www/ncp-web/elements.php on line 178', referer: https://192.168.2.100:44443/wizard/
`

The elements.php script tries to open /usr/local/bin/ncp/ncp-backup as a directory, but this is not a directory.
Here the content of /usr/local/bin/ncp:
```
# ls -la /usr/local/bin/ncp/           
total 36
drwxr-xr-x 9 root root 4096 Jan  6 01:42 .
drwxr-xr-x 3 root root 4096 Jan  6 01:44 ..
drwxr-xr-x 2 root root 4096 Jan  6 01:45 BACKUPS
drwxr-xr-x 2 root root 4096 Jan  6 01:46 CONFIG
drwxr-xr-x 2 root root 4096 Jan  6 01:46 NETWORKING
drwxr-xr-x 2 root root 4096 Jan  6 01:45 SECURITY
drwxr-xr-x 2 root root 4096 Jan  6 01:45 SYSTEM
drwxr-xr-x 2 root root 4096 Jan  6 01:45 TOOLS
drwxr-xr-x 2 root root 4096 Jan  6 01:45 UPDATES
lrwxrwxrwx 1 root root   25 Jan  6 01:40 ncp-backup -> /usr/local/bin/ncp-backup
lrwxrwxrwx 1 root root   26 Jan  6 01:42 ncp-restore -> /usr/local/bin/ncp-restore
```

Could this be the reason for the blank web interface?

ZendaiOwl, 2023-01-06:
Can confirm this issue on Docker image for amd64 as well, there is indeed an issue with the PHP array function trying to read a directory but is pointing to a file (or maybe a symlink?)

theCalcaholic, 2023-01-06:
The original issue is something different, since the bug encountered by @powerflo and @ZendaiOwl is not even in master yet

ZendaiOwl, 2023-01-06:
Oh my bad, was just confirming the issue that powerflo encountered on the testing arm64 image happened on the amd64 as well :) 

You had already started fixing that when I wrote tho

theCalcaholic, 2023-01-07:
Don't worry, I just wanted to have this documented in the issue :)

theCalcaholic, 2023-01-21:
@Tchiimy Do you still encounter this issue? If so, please add the output of `ncp-report`

Tchiimy, 2023-01-24:
> ncp-report

Hmm I've pulled the latest image but I still have this problem : 

[image]

Is it linked with my system or firefox ?

My `ncp-report` : 

`<--! Paste this in GitHub report -->

<details>
<summary>NextcloudPi diagnostics</summary>

```

NextcloudPi version  v1.51.0
NextcloudPi image    NextcloudPi_docker_01-22-23
OS                   Debian GNU/Linux 11. 5.15.84-v8+ (aarch64)
automount            no
USB devices          sda sdb 
datadir              /data/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      95G/235G
rootfs usage         95G/235G
swapfile             /dev/zram1
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    24.0.5.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              closed
Port 443             closed
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               6:50

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "7": "nextcloudpi",
            "5": "nextcloudpi.local",
            "8": "nextcloudpi.lan",
            "3": "ca11270d5d99",
            "11": "77.204.154.52",
            "1": "172.17.0.3",
            "14": "ca11270d5d99",
[... 83 lines cut ...]
```

{"reqId":"Y86k3r4pVqzEAKszsM_NMQAAAAk","level":3,"time":"2023-01-23T15:16:48+00:00","remoteAddr":"10.201.174.4","user":"ncp","app":"drawio","method":"DELETE","url":"/remote.php/dav/files/ncp/4TB/Joplin/locks/1_1_7b72c30cd6c645e6949d26950079d442.json","message":"Can't delete preview for file: /ncp/files/4TB/Joplin/locks/1_1_7b72c30cd6c645e6949d26950079d442.json","userAgent":"Joplin/1.0","version":"24.0.5.1","exception":{"Exception":"OCP\\Files\\NotFoundException","Message":"/appdata_oc4emqbveg6b/drawio/previews","Code":0,"Trace":[{"function":"get","class":"OC\\Files\\Node\\Root","type":"->"},{"file":"/data/nextcloud/lib/private/Files/Node/LazyFolder.php","line":72,"function":"call_user_func_array"},{"file":"/data/nextcloud/lib/private/Files/Node/LazyFolder.php","line":149,"function":"__call","class":"OC\\Files\\Node\\LazyFolder","type":"->"},{"file":"/data/nextcloud/lib/private/Files/AppData/AppData.php","line":141,"function":"get","class":"OC\\Files\\Node\\LazyFolder","type":"->"},{"file":"/data/nextcloud/apps/drawio/lib/Listeners/FileDeleteListener.php","line":39,"function":"getFolder","class":"OC\\Files\\AppData\\AppData","type":"->"},{"file":"/data/nextcloud/lib/private/EventDispatcher/ServiceEventListener.php","line":87,"function":"handle","class":"OCA\\Drawio\\Listeners\\FileDeleteListener","type":"->"},{"file":"/data/nextcloud/3rdparty/symfony/event-dispatcher/EventDispatcher.php","line":251,"function":"__invoke","class":"OC\\EventDispatcher\\ServiceEventListener","type":"->"},{"file":"/data/nextcloud/3rdparty/symfony/event-dispatcher/EventDispatcher.php","line":73,"function":"callListeners","class":"Symfony\\Component\\EventDispatcher\\EventDispatcher","type":"->"},{"file":"/data/nextcloud/lib/private/EventDispatcher/EventDispatcher.php","line":88,"function":"dispatch","class":"Symfony\\Component\\EventDispatcher\\EventDispatcher","type":"->"},{"file":"/data/nextcloud/lib/private/EventDispatcher/EventDispatcher.php","line":100,"function":"dispatch","class":"OC\\EventDispatcher\\EventDispatcher","type":"->"},{"file":"/data/nextcloud/lib/private/Files/Node/HookConnector.php","line":156,"function":"dispatchTyped","class":"OC\\EventDispatcher\\EventDispatcher","type":"->"},{"file":"/data/nextcloud/lib/private/legacy/OC_Hook.php","line":106,"function":"postDelete","class":"OC\\Files\\Node\\HookConnector","type":"->"},{"file":"/data/nextcloud/lib/private/Files/View.php","line":1290,"function":"emit","class":"OC_Hook","type":"::"},{"file":"/data/nextcloud/lib/private/Files/View.php","line":1218,"function":"runHooks","class":"OC\\Files\\View","type":"->"},{"file":"/data/nextcloud/lib/private/Files/View.php","line":733,"function":"basicOperation","class":"OC\\Files\\View","type":"->"},{"file":"/data/nextcloud/apps/dav/lib/Connector/Sabre/File.php","line":523,"function":"unlink","class":"OC\\Files\\View","type":"->"},{"file":"/data/nextcloud/3rdparty/sabre/dav/lib/DAV/Tree.php","line":179,"function":"delete","class":"OCA\\DAV\\Connector\\Sabre\\File","type":"->"},{"file":"/data/nextcloud/3rdparty/sabre/dav/lib/DAV/CorePlugin.php","line":281,"function":"delete","class":"Sabre\\DAV\\Tree","type":"->"},{"file":"/data/nextcloud/3rdparty/sabre/event/lib/WildcardEmitterTrait.php","line":89,"function":"httpDelete","class":"Sabre\\DAV\\CorePlugin","type":"->"},{"file":"/data/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":472,"function":"emit","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/data/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":253,"function":"invokeMethod","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/data/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":321,"function":"start","class" [... cut ...] ,"function":"get","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/data/nextcloud/lib/base.php","line":602,"function":"boot","class":"OC\\Server","type":"->"},{"file":"/data/nextcloud/lib/base.php","line":1111,"function":"init","class":"OC","type":"::"},{"file":"/data/nextcloud/status.php","line":37,"args":["/data/nextcloud/lib/base.php"],"function":"require_once"}],"File":"/data/nextcloud/lib/private/RedisFactory.php","Line":143,"CustomMessage":"--"}}

```
</details>`

Tchiimy, 2023-01-24:
> Can you give me the output of `cat /var/log/apache2/error.log` I believe there's an invalid file in the config directory of the ncp apps (`/data/ncp`)

> cat /var/log/apache2/error.log

Hmm I tried to do it via the command console of the container (portainer) and I got this : 

`root@ca11270d5d99:/# cat /var/log/apache2/error.log
[Tue Jan 24 14:35:42.541769 2023] [ssl:error] [pid 324:tid 547642064912] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=localhost / issuer: CN=localhost / serial: 1050478643D922999C4D0F784DB1775639C0A8D0 / notbefore: Jan 22 01:34:41 2023 GMT / notafter: Jan 19 01:34:41 2033 GMT]
[Tue Jan 24 14:35:42.541876 2023] [ssl:error] [pid 324:tid 547642064912] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Tue Jan 24 14:35:42.569590 2023] [ssl:error] [pid 325:tid 547642064912] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=localhost / issuer: CN=localhost / serial: 1050478643D922999C4D0F784DB1775639C0A8D0 / notbefore: Jan 22 01:34:41 2023 GMT / notafter: Jan 19 01:34:41 2033 GMT]
[Tue Jan 24 14:35:42.569664 2023] [ssl:error] [pid 325:tid 547642064912] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Tue Jan 24 14:35:42.575066 2023] [mpm_event:notice] [pid 325:tid 547642064912] AH00489: Apache/2.4.54 (Debian) OpenSSL/1.1.1n configured -- resuming normal operations
[Tue Jan 24 14:35:42.575182 2023] [core:notice] [pid 325:tid 547642064912] AH00094: Command line: '/usr/sbin/apache2'
[Tue Jan 24 14:35:43.617377 2023] [authz_host:error] [pid 329:tid 547314712960] [client 172.17.0.1:34334] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name
[Tue Jan 24 14:35:49.654471 2023] [authz_host:error] [pid 330:tid 547314712960] [client 172.17.0.1:34350] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name
[Tue Jan 24 14:35:52.720875 2023] [authz_host:error] [pid 329:tid 547632361856] [client 10.201.174.4:56362] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name
[Tue Jan 24 14:35:53.140953 2023] [authz_host:error] [pid 329:tid 547632361856] [client 10.201.174.4:56362] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name
[Tue Jan 24 14:35:53.141610 2023] [authz_host:error] [pid 329:tid 547632361856] [client 10.201.174.4:56362] AH01753: access check of 'localhost' to /index.php failed, reason: unable to get the remote host name
[Tue Jan 24 14:35:53.363177 2023] [authz_host:error] [pid 329:tid 547632361856] [client 10.201.174.4:56362] AH01753: access check of 'localhost' to /index.php failed, reason: unable to get the remote host name
[Tue Jan 24 14:35:54.859076 2023] [authz_host:error] [pid 330:tid 546903667072] [client 172.17.0.1:56944] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name
[Tue Jan 24 14:35:57.122295 2023] [proxy_fcgi:error] [pid 329:tid 547632361856] [client 10.201.174.4:56362] AH01071: Got error 'PHP message: PHP Warning:  Trying to access array offset on value of type null in /var/www/ncp-web/elements.php on line 190PHP message: PHP Fatal error:  Uncaught TypeError: sizeof(): Argument #1 ($value) must be of type Countable|array, null given in /var/www/ncp-web/elements.php:190\nStack trace:\n#0 /var/www/ncp-web/index.php(236): print_sidebar()\n#1 {main}\n  thrown in /var/www/ncp-web/elements.php on line 190'
[Tue Jan 24 14:35:57.282576 2023] [authz_host:error] [pid 329:tid 547632361856] [client 10.201.174.4:56362] AH01753: access check of 'localhost' to /img/view-close.svg failed, reason: unable to get the remote host name, referer: https://192.168.0.21:4444/css/ncp.css
[Tue Jan 24 14:36:00.353169 2023] [authz_host:error] [pid 329:tid 547632361856] [client 10.201.174.4:56362] AH01753: access check of 'localhost' to /ncp-output.php failed, reason: unable to get the remote host name, referer: https://192.168.0.21:4444/wizard/
[Tue Jan 24 14:36:00.500710 2023] [authz_host:error] [pid 329:tid 547272749440] [client 172.17.0.1:55514] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name
[Tue Jan 24 14:36:02.934874 2023] [authz_host:error] [pid 329:tid 547623969152] [client 10.201.174.4:56362] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name, referer: https://192.168.0.21:4444/wizard/
[Tue Jan 24 14:36:03.168216 2023] [authz_host:error] [pid 329:tid 547623969152] [client 10.201.174.4:56362] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name, referer: https://192.168.0.21:4444/wizard/
[Tue Jan 24 14:36:03.168626 2023] [authz_host:error] [pid 329:tid 547623969152] [client 10.201.174.4:56362] AH01753: access check of 'localhost' to /index.php failed, reason: unable to get the remote host name, referer: https://192.168.0.21:4444/wizard/
[Tue Jan 24 14:36:03.342724 2023] [authz_host:error] [pid 329:tid 547623969152] [client 10.201.174.4:56362] AH01753: access check of 'localhost' to /index.php failed, reason: unable to get the remote host name, referer: https://192.168.0.21:4444/wizard/
[Tue Jan 24 14:36:07.145120 2023] [proxy_fcgi:error] [pid 329:tid 547623969152] [client 10.201.174.4:56362] AH01071: Got error 'PHP message: PHP Warning:  Trying to access array offset on value of type null in /var/www/ncp-web/elements.php on line 190PHP message: PHP Fatal  [... cut ...]
[Tue Jan 24 14:38:05.081577 2023] [authz_host:error] [pid 329:tid 547180429696] [client 172.17.0.1:45858] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name
[Tue Jan 24 14:38:10.275559 2023] [authz_host:error] [pid 329:tid 547255964032] [client 172.17.0.1:37368] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name
[Tue Jan 24 14:38:15.782863 2023] [authz_host:error] [pid 330:tid 547289534848] [client 172.17.0.1:37382] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name`

How else am I supposed to access it ?

theCalcaholic, 2023-01-24:
Accessing it like that is totally fine.
That log didn't help, unfortunately. Can you provide the output of `bash -c 'for cfg in /usr/local/etc/ncp-config.d/*; do echo "$cfg"; jq "." "$cfg" > /dev/null; done'`?

Please use multiline code blocks for pasting the output. The syntax is like this:

    ```
    your 
    multiline
    code
    ```

theCalcaholic, 2023-01-24:
Hm okay, looks about right...

I have improved the sensitive of the command a bit. Can you retry with `bash -c 'for cfg in /usr/local/etc/ncp-config.d/*; do echo "$cfg"; { jq ".params" "$cfg" || echo "error" >&2; } | grep null > /dev/null && echo "missing params"; done'` and give me the result?

ChronoWerX82, 2023-01-26:
Here is the same problem since 1.51.0

Tchiimy, 2023-01-26:
> 

Hmm I have this : 

```
root@ca11270d5d99:/# bash -c 'ls -1 /usr/local/bin/ncp/*/*.sh'
/usr/local/bin/ncp/BACKUPS/nc-backup-auto.sh
/usr/local/bin/ncp/BACKUPS/nc-backup.sh
/usr/local/bin/ncp/BACKUPS/nc-export-ncp.sh
/usr/local/bin/ncp/BACKUPS/nc-import-ncp.sh
/usr/local/bin/ncp/BACKUPS/nc-restore.sh
/usr/local/bin/ncp/BACKUPS/nc-rsync-auto.sh
/usr/local/bin/ncp/BACKUPS/nc-rsync.sh
/usr/local/bin/ncp/CONFIG/nc-admin.sh
/usr/local/bin/ncp/CONFIG/nc-httpsonly.sh
/usr/local/bin/ncp/CONFIG/nc-limits.sh
/usr/local/bin/ncp/CONFIG/nc-passwd.sh
/usr/local/bin/ncp/CONFIG/nc-prettyURL.sh
/usr/local/bin/ncp/CONFIG/nc-previews-auto.sh
/usr/local/bin/ncp/CONFIG/nc-scan-auto.sh
/usr/local/bin/ncp/CONFIG/nc-trusted-domains.sh
/usr/local/bin/ncp/CONFIG/nc-webui.sh
/usr/local/bin/ncp/NETWORKING/dnsmasq.sh
/usr/local/bin/ncp/NETWORKING/duckDNS.sh
/usr/local/bin/ncp/NETWORKING/freeDNS.sh
/usr/local/bin/ncp/NETWORKING/letsencrypt.sh
/usr/local/bin/ncp/NETWORKING/namecheapDNS.sh
/usr/local/bin/ncp/NETWORKING/nc-forward-ports.sh
/usr/local/bin/ncp/NETWORKING/nc-trusted-proxies.sh
/usr/local/bin/ncp/NETWORKING/no-ip.sh
/usr/local/bin/ncp/NETWORKING/spDYN.sh
/usr/local/bin/ncp/SECURITY/modsecurity.sh
/usr/local/bin/ncp/SECURITY/nc-encrypt.sh
/usr/local/bin/ncp/SYSTEM/metrics.sh
/usr/local/bin/ncp/SYSTEM/nc-info.sh
/usr/local/bin/ncp/TOOLS/clear-php-opcache.sh
/usr/local/bin/ncp/TOOLS/nc-fix-permissions.sh
/usr/local/bin/ncp/TOOLS/nc-maintenance.sh
/usr/local/bin/ncp/TOOLS/nc-previews.sh
/usr/local/bin/ncp/TOOLS/nc-scan.sh
/usr/local/bin/ncp/UPDATES/nc-autoupdate-nc.sh
/usr/local/bin/ncp/UPDATES/nc-notify-updates.sh
/usr/local/bin/ncp/UPDATES/nc-update-nc-apps-auto.sh
/usr/local/bin/ncp/UPDATES/nc-update-nc-apps.sh
/usr/local/bin/ncp/UPDATES/nc-update-nextcloud.sh
/usr/local/bin/ncp/UPDATES/unattended-upgrades.sh
```

It seems that I miss some scripts no ?

ChronoWerX82, 2023-01-26:
Here with the same problem since 1.51.0

```
/local/bin/ncp/*/*.sh'
/usr/local/bin/ncp/BACKUPS/nc-backup-auto.sh
/usr/local/bin/ncp/BACKUPS/nc-backup.sh
/usr/local/bin/ncp/BACKUPS/nc-export-ncp.sh
/usr/local/bin/ncp/BACKUPS/nc-import-ncp.sh
/usr/local/bin/ncp/BACKUPS/nc-restore.sh
/usr/local/bin/ncp/BACKUPS/nc-rsync-auto.sh
/usr/local/bin/ncp/BACKUPS/nc-rsync.sh
/usr/local/bin/ncp/CONFIG/nc-admin.sh
/usr/local/bin/ncp/CONFIG/nc-httpsonly.sh
/usr/local/bin/ncp/CONFIG/nc-limits.sh
/usr/local/bin/ncp/CONFIG/nc-passwd.sh
/usr/local/bin/ncp/CONFIG/nc-prettyURL.sh
/usr/local/bin/ncp/CONFIG/nc-previews-auto.sh
/usr/local/bin/ncp/CONFIG/nc-scan-auto.sh
/usr/local/bin/ncp/CONFIG/nc-trusted-domains.sh
/usr/local/bin/ncp/CONFIG/nc-webui.sh
/usr/local/bin/ncp/NETWORKING/dnsmasq.sh
/usr/local/bin/ncp/NETWORKING/duckDNS.sh
/usr/local/bin/ncp/NETWORKING/freeDNS.sh
/usr/local/bin/ncp/NETWORKING/letsencrypt.sh
/usr/local/bin/ncp/NETWORKING/namecheapDNS.sh
/usr/local/bin/ncp/NETWORKING/nc-forward-ports.sh
/usr/local/bin/ncp/NETWORKING/nc-trusted-proxies.sh
/usr/local/bin/ncp/NETWORKING/no-ip.sh
/usr/local/bin/ncp/NETWORKING/spDYN.sh
/usr/local/bin/ncp/SECURITY/modsecurity.sh
/usr/local/bin/ncp/SECURITY/nc-encrypt.sh
/usr/local/bin/ncp/SYSTEM/metrics.sh
/usr/local/bin/ncp/SYSTEM/nc-info.sh
/usr/local/bin/ncp/TOOLS/clear-php-opcache.sh
/usr/local/bin/ncp/TOOLS/nc-fix-permissions.sh
/usr/local/bin/ncp/TOOLS/nc-maintenance.sh
/usr/local/bin/ncp/TOOLS/nc-previews.sh
/usr/local/bin/ncp/TOOLS/nc-scan.sh
/usr/local/bin/ncp/UPDATES/nc-autoupdate-nc.sh
/usr/local/bin/ncp/UPDATES/nc-notify-updates.sh
/usr/local/bin/ncp/UPDATES/nc-update-nc-apps-auto.sh
/usr/local/bin/ncp/UPDATES/nc-update-nc-apps.sh
/usr/local/bin/ncp/UPDATES/nc-update-nextcloud.sh
/usr/local/bin/ncp/UPDATES/unattended-upgrades.sh
```

Some scripts miss too

theCalcaholic, 2023-01-26:
Ah sorry, the missing scripts are, because I gave you the non-docker list. That's fine and they should not be added.

The error would not be caused by missing scripts, but rather by additional scripts that exist in the bin directory but miss the config file in /usr/local/etc/ncp-config.d

theCalcaholic, 2023-01-26:
Okay, I don't think we are getting anywhere with this without additional debug output. I have prepared a version of elements.php that should do two things:
1) It should skip the app that's causing the issue for you, therefore making the web UI work again
2) It should log the file/app that causes the issue to /var/log/apache2/error.log

Please replace your existing /var/www/ncp-web/elements.php with the file attached and fix its permissions with `sudo chown www-data: /var/www/ncp-web/elements.php` and then provide the output of the following command:

```sh
cat /var/log/apache2/error.log
```
[elements.php.zip](https://github.com/nextcloud/nextcloudpi/files/10511917/elements.php.zip)

ChronoWerX82, 2023-01-26:
Now working here again

```
[Thu Jan 26 13:02:09.645383 2023] [ssl:error] [pid 330:tid 548350873616] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=localhost / issuer: CN=localhost / serial: 1050478643D922999C4D0F784DB1775639C0A8D0 / notbefore: Jan 22 01:34:41 2023 GMT / notafter: Jan 19 01:34:41 2033 GMT]
[Thu Jan 26 13:02:09.645449 2023] [ssl:error] [pid 330:tid 548350873616] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Thu Jan 26 13:02:09.669322 2023] [ssl:error] [pid 331:tid 548350873616] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=localhost / issuer: CN=localhost / serial: 1050478643D922999C4D0F784DB1775639C0A8D0 / notbefore: Jan 22 01:34:41 2023 GMT / notafter: Jan 19 01:34:41 2033 GMT]
[Thu Jan 26 13:02:09.669389 2023] [ssl:error] [pid 331:tid 548350873616] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Thu Jan 26 13:02:09.674445 2023] [mpm_event:notice] [pid 331:tid 548350873616] AH00489: Apache/2.4.54 (Debian) OpenSSL/1.1.1n configured -- resuming normal operations
[Thu Jan 26 13:02:09.674590 2023] [core:notice] [pid 331:tid 548350873616] AH00094: Command line: '/usr/sbin/apache2'
[Thu Jan 26 13:02:34.033157 2023] [proxy_fcgi:error] [pid 337:tid 548190101888] [client 192.168.178.10:53042] AH01071: Got error 'PHP message: PHP Warning:  file_get_contents(/usr/local/etc/ncp-config.d/nc-restore-snapshot.cfg): Failed to open stream: No such file or directory in /var/www/ncp-web/elements.php on line 183PHP message: PHP Warning:  Trying to access array offset on value of type null in /var/www/ncp-web/elements.php on line 190PHP message: PHP Fatal error:  Uncaught TypeError: sizeof(): Argument #1 ($value) must be of type Countable|array, null given in /var/www/ncp-web/elements.php:190\nStack trace:\n#0 /var/www/ncp-web/index.php(236): print_sidebar()\n#1 {main}\n  thrown in /var/www/ncp-web/elements.php on line 190'
[Thu Jan 26 13:03:04.520519 2023] [proxy_fcgi:error] [pid 337:tid 548164923776] [client 192.168.178.10:53053] AH01071: Got error 'PHP message: PHP Warning:  file_get_contents(/usr/local/etc/ncp-config.d/nc-restore-snapshot.cfg): Failed to open stream: No such file or directory in /var/www/ncp-web/elements.php on line 183PHP message: PHP Warning:  Trying to access array offset on value of type null in /var/www/ncp-web/elements.php on line 190PHP message: PHP Fatal error:  Uncaught TypeError: sizeof(): Argument #1 ($value) must be of type Countable|array, null given in /var/www/ncp-web/elements.php:190\nStack trace:\n#0 /var/www/ncp-web/index.php(236): print_sidebar()\n#1 {main}\n  thrown in /var/www/ncp-web/elements.php on line 190', referer: https://192.168.178.11:4443/wizard/
[Thu Jan 26 13:03:24.966601 2023] [proxy_fcgi:error] [pid 337:tid 548299207040] [client 192.168.178.10:53075] AH01071: Got error 'PHP message: PHP Warning:  file_get_contents(/usr/local/etc/ncp-config.d/nc-restore-snapshot.cfg): Failed to open stream: No such file or directory in /var/www/ncp-web/elements.php on line 183PHP message: PHP Warning:  Trying to access array offset on value of type null in /var/www/ncp-web/elements.php on line 190PHP message: PHP Fatal error:  Uncaught TypeError: sizeof(): Argument #1 ($value) must be of type Countable|array, null given in /var/www/ncp-web/elements.php:190\nStack trace:\n#0 /var/www/ncp-web/index.php(236): print_sidebar()\n#1 {main}\n  thrown in /var/www/ncp-web/elements.php on line 190'
[Thu Jan 26 13:06:37.503312 2023] [proxy_fcgi:error] [pid 337:tid 548257243520] [client 192.168.178.10:53391] AH01071: Got error 'PHP message: PHP Warning:  file_get_contents(/usr/local/etc/ncp-config.d/nc-restore-snapshot.cfg): Failed to open stream: No such file or directory in /var/www/ncp-web/elements.php on line 183PHP message: PHP Warning:  Trying to access array offset on value of type null in /var/www/ncp-web/elements.php on line 190PHP message: PHP Fatal error:  Uncaught TypeError: sizeof(): Argument #1 ($value) must be of type Countable|array, null given in /var/www/ncp-web/elements.php:190\nStack trace:\n#0 /var/www/ncp-web/index.php(236): print_sidebar()\n#1 {main}\n  thrown in /var/www/ncp-web/elements.php on line 190'
[Thu Jan 26 13:08:18.920382 2023] [mpm_event:notice] [pid 331:tid 548350873616] AH00492: caught SIGWINCH, shutting down gracefully
[Thu Jan 26 13:08:18.955312 2023] [proxy_fcgi:error] [pid 337:tid 548139721088] [client 192.168.178.10:53053] AH01067: Failed to read FastCGI header, referer: https://192.168.178.11:4443/wizard/
[Thu Jan 26 13:08:49.495806 2023] [ssl:error] [pid 327:tid 548498112528] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=localhost / issuer: CN=localhost / serial: 1050478643D922999C4D0F784DB1775639C0A8D0 / notbefore: Jan 22 01:34:41 2023 GMT / notafter: Jan 19 01:34:41 2033 GMT]
[Thu Jan 26 13:08:49.496146 2023] [ssl:error] [pid 327:tid 548498112528] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Thu Jan 26 13:08:49.518203 2023] [ssl:error] [pid 328:tid 548498112528] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=localhost / issuer: CN=localhost / serial: 1050478643D922999C4D0F784DB1775639C0A8D0 / notbefore: Jan 22 01:34:41 2023 GMT / notafter: Jan 19 01:34:41 2033 GMT]
[Thu Jan 26 13:08:49.518270 2023] [ssl:error] [pid 328:tid 548498112528] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[... cut ...] offset on value of type null in /var/www/ncp-web/elements.php on line 144PHP message: PHP Warning:  Trying to access array offset on value of type null in /var/www/ncp-web/elements.php on line 145PHP message: PHP Warning:  Trying to access array offset on value of type null in /var/www/ncp-web/elements.php on line 146PHP message: PHP Warning:  Trying to access array offset on value of type null in /var/www/ncp-web/elements.php on line 21PHP message: PHP Warning:  foreach() argument must be o
```

theCalcaholic, 2023-01-26:
@ChronoWerX82 Okay, so the issue was `nc-swapfile`! Can you give me the output of `cat /usr/local/etc/ncp-config.d/nc-swapfile.cfg`, please?

ChronoWerX82, 2023-01-26:
```
cat: /usr/local/etc/ncp-config.d/nc-swapfile.cfg: No such file or directory
```

theCalcaholic, 2023-01-27:
@ChronoWerX82 nc-swapfile.sh is one of the scripts that shouldn't be present in the first place on docker.
 Could you remove the following scripts from /data/bin?
 
 ```
nc-autoupdate-ncp
nc-update
nc-automount
nc-format-USB
nc-datadir
nc-database
nc-ramlogs
nc-swapfile
nc-static-IP
nc-wifi
UFW
nc-snapshot
nc-snapshot-auto
nc-snapshot-sync
nc-restore-snapshot
nc-audit
nc-hdd-monitor
nc-hdd-test
nc-zram
SSH
fail2ban
NFS
nc-nextcloud
nc-init
samba
```

ChronoWerX82, 2023-01-27:
No one of these files are in the foldere

[image]

theCalcaholic, 2023-01-27:
Sorry, I was unprecise - they are in the various subfolders of /usr/local/bin/ncp

ChronoWerX82, 2023-01-27:
These folders are all clean

```
/usr/local/bin/ncp/BACKUPS/nc-backup-auto.sh
/usr/local/bin/ncp/BACKUPS/nc-backup.sh
/usr/local/bin/ncp/BACKUPS/nc-export-ncp.sh
/usr/local/bin/ncp/BACKUPS/nc-import-ncp.sh
/usr/local/bin/ncp/BACKUPS/nc-restore.sh
/usr/local/bin/ncp/BACKUPS/nc-rsync-auto.sh
/usr/local/bin/ncp/BACKUPS/nc-rsync.sh
/usr/local/bin/ncp/CONFIG/nc-admin.sh
/usr/local/bin/ncp/CONFIG/nc-httpsonly.sh
/usr/local/bin/ncp/CONFIG/nc-limits.sh
/usr/local/bin/ncp/CONFIG/nc-passwd.sh
/usr/local/bin/ncp/CONFIG/nc-prettyURL.sh
/usr/local/bin/ncp/CONFIG/nc-previews-auto.sh
/usr/local/bin/ncp/CONFIG/nc-scan-auto.sh
/usr/local/bin/ncp/CONFIG/nc-trusted-domains.sh
/usr/local/bin/ncp/CONFIG/nc-webui.sh
/usr/local/bin/ncp/NETWORKING/dnsmasq.sh
/usr/local/bin/ncp/NETWORKING/duckDNS.sh
/usr/local/bin/ncp/NETWORKING/freeDNS.sh
/usr/local/bin/ncp/NETWORKING/letsencrypt.sh
/usr/local/bin/ncp/NETWORKING/namecheapDNS.sh
/usr/local/bin/ncp/NETWORKING/nc-forward-ports.sh
/usr/local/bin/ncp/NETWORKING/nc-trusted-proxies.sh
/usr/local/bin/ncp/NETWORKING/no-ip.sh
/usr/local/bin/ncp/NETWORKING/spDYN.sh
/usr/local/bin/ncp/SECURITY/modsecurity.sh
/usr/local/bin/ncp/SECURITY/nc-encrypt.sh
/usr/local/bin/ncp/SYSTEM/metrics.sh
/usr/local/bin/ncp/SYSTEM/nc-info.sh
/usr/local/bin/ncp/TOOLS/clear-php-opcache.sh
/usr/local/bin/ncp/TOOLS/nc-fix-permissions.sh
/usr/local/bin/ncp/TOOLS/nc-maintenance.sh
/usr/local/bin/ncp/TOOLS/nc-previews.sh
/usr/local/bin/ncp/TOOLS/nc-scan.sh
/usr/local/bin/ncp/UPDATES/nc-autoupdate-nc.sh
/usr/local/bin/ncp/UPDATES/nc-notify-updates.sh
/usr/local/bin/ncp/UPDATES/nc-update-nc-apps-auto.sh
/usr/local/bin/ncp/UPDATES/nc-update-nc-apps.sh
/usr/local/bin/ncp/UPDATES/nc-update-nextcloud.sh
/usr/local/bin/ncp/UPDATES/unattended-upgrades.sh
```

ChronoWerX82, 2023-01-30:
@theCalcaholic any news or solutions?