nextcloud/nextcloudpi #1672: coredump /usr/sbin/php-fpm8.1

mabod, 2022-12-03:
I started using nextcloudpi via docker-compose on an up-to-date endeavouros.  I see severavl coredumps for` /usr/sbin/php-fpm8.1`. 

Just one example out of 156 coredumps from the last 4 days:

```
# coredumpctl info 16775
           PID: 16775 (php-fpm8.1)
           UID: 33 (http)
           GID: 33 (http)
        Signal: 11 (SEGV)
     Timestamp: Sat 2022-12-03 17:33:50 CET (20min ago)
  Command Line: $'php-fpm: pool www'
    Executable: /usr/sbin/php-fpm8.1
 Control Group: /system.slice/docker-60963f79d0e89de4a1754d3cd572d5b1c022b6eaca00f64f844daaf11a54fea7.scope
          Unit: docker-60963f79d0e89de4a1754d3cd572d5b1c022b6eaca00f64f844daaf11a54fea7.scope
         Slice: system.slice
       Boot ID: b96d7fb0b0d84463aa28875582a04c3c
    Machine ID: 4bd88beaa35549b5922de02c8064cbf1
      Hostname: 60963f79d0e8
       Storage: /var/lib/systemd/coredump/core.php-fpm8\x2e1.33.b96d7fb0b0d84463aa28875582a04c3c.16775.1670085230000000.zst (inaccessible)
       Message: Process 16775 (php-fpm8.1) of user 33 dumped core.
                
                Stack trace of thread 113:
                #0  0x00007f776d51425c n/a (/usr/lib/php/20210902/opcache.so + 0x1c25c)
                ELF object binary architecture: AMD x86-64
```

Versions:

```
docker 1:20.10.21-1
ownyourbits/nextcloudpi-x86   latest 
NextCloudPi 24.0.5
```

```
docker -compose.yml
# cat docker-compose.yml     
version: '3'
services:
  nextcloudpi:
    image: ownyourbits/nextcloudpi-x86
    command: "${IP}"
    restart: always
    ports:
     - "8080:80"
     - "443:443"
     - "4443:4443"
    volumes:
     - /mnt/zstore/nextcloudpi/data:/data
     - /etc/localtime:/etc/localtime:ro
    # for nc-encrypt
    devices:
      - /dev/fuse:/dev/fuse
    # for nc-encrypt # NOTE: take a look at this https://github.com/docker/for-linux/issues/321#issuecomment-677744121
    cap_add:
      - SYS_ADMIN
    container_name: nextcloudpi

volumes:
  ncdata:

```

theCalcaholic, 2023-01-21:
This is likely due to issues with the PHP opcache JIT that was introduced in PHP 8. It's been causing a lot of segfaults for many users. v1.51.0 brings 

1) an adjusted PHP opcache config, that may help with some of these issues
2) A ncp-config tool to clear the opcache, which should help with these issues at least for some time, whenever executed.

Overall, if your core dumps are caused by the JIT segfaults, this is unfortunately an upstream issue. What you could do (also for verifying that your issues are indeed caused by the JIT), is to set `opcache.jit=off` in `/usr/local/etc/ncp-templates/php/opcache.ini.sh` (will be overwritten on container replacement)

theCalcaholic, 2023-01-21:
Probable duplicate of #1697

theCalcaholic, 2023-02-07:
Closing because it likely is a duplicate and a fix has been provided

closed 2023-02-07 by theCalcaholic