nextcloud/nextcloudpi #1680: Moving data to USB fails with PHP compatibility error

tbreuss, 2022-12-30:
When using TUI or GUI moving data to USB fails with the following error:

~~~
[ nc-datadir ] (Fri Dec 30 17:12:07 GMT 2022)
This version of Nextcloud is not compatible with PHP>=8.2.<br/>You are currently running 8.2.0.
~~~

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.50.4
NextCloudPi image    NextCloudPi_RaspberryPi_v1.50.3.img
OS                   Debian GNU/Linux 11. 5.15.76-v8+ (aarch64)
automount            yes
USB devices          sda
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      13G/15G
rootfs usage         13G/15G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
public IP            ***REMOVED SENSITIVE VALUE***
public IP            ***REMOVED SENSITIVE VALUE***
Port check 80        closed
Port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               2:43

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

This version of Nextcloud is not compatible with PHP>=8.2.<br/>You are currently running 8.2.0.
```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Fri Dec 30 16:57:48.058231 2022] [authz_host:error] [pid 890:tid 548224475520] [client 192.168.1.105:54991] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Fri Dec 30 16:57:48.085916 2022] [authz_host:error] [pid 890:tid 548224475520] [client 192.168.1.105:54991] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Fri Dec 30 17:03:10.572858 2022] [authz_host:error] [pid 890:tid 548224475520] [client 192.168.1.105:54991] AH01753: access check of 'localhost' to /wizard failed, reason: unable to get the remote host name
[Fri Dec 30 17:03:10.580155 2022] [authz_host:error] [pid 890:tid 548224475520] [client 192.168.1.105:54991] AH01753: access check of 'localhost' to /wizard/ failed, reason: unable to get the remote host name
[Fri Dec 30 17:03:10.922911 2022] [authz_host:error] [pid 890:tid 548224475520] [client 192.168.1.105:54991] AH01753: access check of 'localhost' to /wizard/ failed, reason: unable to get the remote host name
[Fri Dec 30 17:03:10.927689 2022] [authz_host:error] [pid 890:tid 548224475520] [client 192.168.1.105:54991] AH01753: access check of 'localhost' to /wizard/index.php failed, reason: unable to get the remote host name
[Fri Dec 30 17:03:11.266166 2022] [authz_host:error] [pid 890:tid 548224475520] [client 192.168.1.105:54991] AH01753: access check of 'localhost' to /wizard/index.php failed, reason: unable to get the remote host name
[Fri Dec 30 17:03:11.305443 2022] [authz_host:error] [pid 890:tid 548224475520] [client 192.168.1.105:54991] AH01753: access check of 'localhost' to /wizard/bootstrap/css/bootstrap.min.css failed, reason: unable to get the remote host name, referer: https://nextcloudpi.local:4443/wizard/
[... 16 lines cut ...]
<details>
<summary>Database logs</summary>

```

tail: cannot open '/var/log/mysql/*.log' for reading: No such file or directory

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"WSh2kte1pEtgNZlLqFNR","level":3,"time":"2022-10-03T22:35:02+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Not installed","userAgent":"--","version":"","exception":{"Exception":"Exception","Message":"Not installed","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/base.php","line":670,"function":"checkInstalled","class":"OC","type":"::"},{"file":"/var/www/nextcloud/lib/base.php","line":1111,"function":"init","class":"OC","type":"::"},{"file":"/var/www/nextcloud/cron.php","line":43,"args":["/var/www/nextcloud/lib/base.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/base.php","Line":282,"CustomMessage":"--"}}
{"reqId":"7ABKHBYpe6d9FUiQFGvl","level":3,"time":"2022-12-28T16:55:10+00:00","remoteAddr":"","user":"--","app":"core","method":"","url":"--","message":"Error while running background job (class: OC\\Preview\\BackgroundCleanupJob, arguments: )","userAgent":"--","version":"24.0.5.1","exception":{"Exception":"RedisException","Message":"AUTH failed while reconnecting","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/Memcache/Redis.php","line":103,"function":"set","class":"Redis","type":"->"},{"file":"/var/www/nextcloud/lib/private/Lock/MemcacheLockingProvider.php","line":83,"function":"add","class":"OC\\Memcache\\Redis","type":"->"},{"file":"/var/www/nextcloud/lib/private/Files/Storage/Common.php","line":758,"function":"acquireLock","class":"OC\\Lock\\MemcacheLockingProvider","type":"->"},{"file":"/var/www/nextcloud/lib/private/Files/Cache/Scanner.php","line":335,"function":"acquireLock","class":"OC\\Files\\Storage\\Common","type":"->"},{"file":"/var/www/nextcloud/lib/private/Files/Cache/LocalRootScanner.php","line":39,"function":"scan","class":"OC\\Files\\Cache\\Scanner","type":"->"},{"file":"/var/www/nextcloud/lib/private/Files/View.php","line":1350,"function":"scan","class":"OC\\Files\\Cache\\LocalRootScanner","type":"->"},{"file":"/var/www/nextcloud/lib/private/Files/View.php","line":1390,"function":"getCacheEntry","class":"OC\\Files\\View","type":"->"},{"file":"/var/www/nextcloud/lib/private/Files/Node/Root.php","line":205,"function":"getFileInfo","class":"OC\\Files\\View","type":"->"},{"function":"get","class":"OC\\Files\\Node\\Root","type":"->"},{"file":"/var/www/nextcloud/lib/private/Files/Node/LazyFolder.php","line":72,"function":"call_user_func_array"},{"file":"/var/www/nextcloud/lib/private/Files/Node/LazyFolder.php","line":149,"function":"__call","class":"OC\\Files\\Node\\LazyFolder","type":"->"},{"file":"/var/www/nextcloud/lib/private/Files/AppData/AppData.php","line":87,"function":"get","class":"OC\\Files\\Node\\LazyFolder","type":"->"},{"file":"/var/www/nextcloud/lib/private/Preview/Storage/Root.php","line":90,"function":"getAppDataRootFolder","class":"OC\\Files\\AppData\\AppData","type":"->"},{"file":"/var/www/nextcloud/lib/private/Preview/BackgroundCleanupJob.php","line":137,"function":"getStorageId","class":"OC\\Preview\\Storage\\Root","type":"->"},{"file":"/var/www/nextcloud/lib/private/Preview/BackgroundCleanupJob.php","line":78,"function":"getNewPreviewLocations","class":"OC\\Preview\\BackgroundCleanupJob","type":"->"},{"file":"/var/www/nextcloud/lib/private/Preview/BackgroundCleanupJob.php","line":64,"function":"getDeletedFiles","class":"OC\\Preview\\BackgroundCleanupJob","type":"->"},{"file":"/var/www/nextcloud/lib/private/BackgroundJob/Job.php","line":54,"function":"run","class":"OC\\Preview\\BackgroundCleanupJob","type":"->"},{"file":"/var/www/nextcloud/lib/private/BackgroundJob/TimedJob.php","line":60,"function":"execute","class":"OC\\BackgroundJob\\Job","type":"->"},{"file":"/var/www/nextcloud/cron.php","line":151,"function":"execute","class":"OC\\BackgroundJob\\TimedJob","type":"->"}],"File":"/var/www/nextcloud/lib/private/Memcache/Redis.php","Line":103,"message":"Error while running background job (class: OC\\Preview\\BackgroundCleanupJob, arguments: )","CustomMessage":"Error while running background job (class: OC\\Preview\\BackgroundCleanupJob, arguments: )"}}
{"reqId":"7ABKHBYpe6d9FUiQFGvl","level":3,"time":"2022-12-28T16:55:11+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Redis server /var/run/redis/redis.sock:0 went away","userAgent":"--","version":"24.0.5.1","exception":{"Exception":"RedisException","Message":"Redis server /var/run/redis/redis.sock:0 went away","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/Memcache/Redis.php","line":48,"function":"get","class":"Redis","type":"->"},{"file":"/var/www/nextcloud/lib/autoloader.php","line":154,"function":"get","class":"OC\\Memcache\\Redis","type":"->"},{"function":"load","class":"OC\\Autoloader","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":114,"function":"__construct","class":"ReflectionClass","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":133," [... cut ...] cloud/apps/dav/lib/Server.php","line":358,"function":"exec","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/var/www/nextcloud/apps/dav/appinfo/v2/remote.php","line":35,"function":"exec","class":"OCA\\DAV\\Server","type":"->"},{"file":"/var/www/nextcloud/remote.php","line":166,"args":["/var/www/nextcloud/apps/dav/appinfo/v2/remote.php"],"function":"require_once"}],"File":"/var/www/nextcloud/3rdparty/doctrine/dbal/src/Driver/PDO/Statement.php","Line":92}},"CustomMessage":"--"}}

```
</details>

staticdev, 2022-12-30:
@tbreuss the latest version of NC still supports until PHP 8.1, the machine you are running may have installed/updated PHP to 8.2 and will make NC not work.

tbreuss, 2022-12-30:
> @tbreuss the latest version of NC still supports until PHP 8.1, the machine you are running may have installed/updated PHP to 8.2 and will make NC not work.

Oh, that was probably me doing n `apt upgrade'. Thanks for the hint.

tbreuss, 2022-12-31:
Me again. I think to problem that I've encountered have to do with the differences between `apt uprade` vs `apt full-upgrade`. Unfortunately the [official documention](https://www.raspberrypi.com/documentation/computers/os.html) is a bit confusing. It is actually recommended to do a `full-upgrade`, but this probably led to the above problem.

Maybe this comment will help someone else with the same problem.

staticdev, 2022-12-31:
Good to know @tbreuss, should we close the issue then?

tbreuss, 2022-12-31:
Yes, I close the issue. Thanks!

closed 2022-12-31 by tbreuss