nextcloud/nextcloudpi #1685: Installing NCP from install.sh on raspbian fails because user pi is not there

Iolaum, 2023-01-04:
## Issue description

I was trying to install NextcloudPi on  a clean Raspberry Pi OS 64 Lite install. The process failed with the following error:

```
$ curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | sudo bash
...
usermod: user 'pi' does not exist

```

## Looking into the issue

This is likely caused by [this code](https://github.com/nextcloud/nextcloudpi/blob/fb8414fa640042430c0e3035151d64fd3da77c14/ncp.sh#L29-L34)

```
  test -f /usr/bin/raspi-config && {
    sed -i '/Change User Password/i"0 NextCloudPi Configuration" "Configuration of NextCloudPi" \\' /usr/bin/raspi-config
    sed -i '/1\\ \*) do_change_pass ;;/i0\\ *) ncp-config ;;'                                       /usr/bin/raspi-config
    # Disable raspberry pi default user
    usermod pi -s /sbin/nologin
  }
```

The latest raspberry pi OS instructions [suggest creating a user defined username and password](https://www.raspberrypi.com/documentation/computers/configuration.html#configuring-a-user).

ZendaiOwl, 2023-01-04:
An issue about this already exists

See: #1674 

Pending PR with a suggested fix: #1682

Iolaum, 2023-01-04:
Nice catch, will close this then as a duplicate.

closed 2023-01-04 by Iolaum

ZendaiOwl, 2023-01-04:
Thank you :pray: