nextcloud/nextcloudpi #1692: nc-backup fails due to Nextcloud is not compatible with PHP>=8.2. issue

michuvon, 2023-01-09:
My Nextcloud(pi) is working fine but auto-backups and manual backups fail. When running backups manually the following error comes up:

```
Running nc-backup
check free space...
This version of Nextcloud is not compatible with PHP>=8.2.<br/>You are currently running 8.2.0.Abort...
This version of Nextcloud is not compatible with PHP>=8.2.<br/>You are currently running 8.2.0.This version of Nextcloud is not compatible with PHP>=8.2.<br/>You are currently running 8.2.0.Done. Press any key...
```

- My NCP takes a automatic backup every 7 days.
- I have auto updates enabled.
- The last successfull backup was on the 1.1.2023
- Today I just updated NCP to version 1.50.5 but the issue persisted.

Seems like the:
> This version of Nextcloud is not compatible with PHP>=8.2.
issue is also coming up in this issue: #1664 

---

> <--! Paste this in GitHub report -->
> 
> <details>
> <summary>NextCloudPi diagnostics</summary>
> 
> ```
> 
> NextCloudPi version  v1.50.5
> NextCloudPi image    NextCloudPi_02-09-22
> OS                   Debian GNU/Linux 11. 4.19.0-20-amd64 (x86_64)
> automount            yes
> USB devices          sdb 
> datadir              /media/ncdata/ncdata/data
> data in SD           yes
> data filesystem      ext2/ext3
> data disk usage      300G/503G
> rootfs usage         300G/503G
> swapfile             /var/swap
> dbdir                /var/lib/mysql
> Nextcloud check      error
> HTTPD service        up
> PHP service          up
> MariaDB service      up
> Redis service        up
> HPB service          down
> Postfix service      up
> Internet check       ok
> public IP            ***REMOVED SENSITIVE VALUE***
> public IP            ***REMOVED SENSITIVE VALUE***
> Port check 80        closed
> Port check 443       closed
> IP                   ***REMOVED SENSITIVE VALUE***
> gateway              ***REMOVED SENSITIVE VALUE***
> Interface            ens18
> certificates         ***REMOVED SENSITIVE VALUE***
> NAT loopback         no
> Uptime               4:34
> 
> ```
> </details>
> 
> 
> <details>
> <summary>Nextcloud configuration</summary>
> 
> ```
> 
> This version of Nextcloud is not compatible with PHP>=8.2.<br/>You are currently running 8.2.0.
> ```
> </details>
> 
> 
> <details>
> <summary>HTTPd logs</summary>
> 
> ```
> 
> [Mon Jan 09 00:00:02.530253 2023] [ssl:warn] [pid 506:tid 139697910865216] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
> [Mon Jan 09 00:00:02.530331 2023] [ssl:error] [pid 506:tid 139697910865216] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=NextCloudPi / issuer: CN=NextCloudPi / serial: 170D1668CEDDEEA2C439118EB999D3AF17EBE686 / notbefore: Feb  9 12:39:36 2022 GMT / notafter: Feb  7 12:39:36 2032 GMT]
> [Mon Jan 09 00:00:02.530339 2023] [ssl:error] [pid 506:tid 139697910865216] AH02604: Unable to configure certificate localhost:4443:0 for stapling
> [Mon Jan 09 00:00:02.531545 2023] [mpm_event:notice] [pid 506:tid 139697910865216] AH00489: Apache/2.4.54 (Debian) OpenSSL/1.1.1n configured -- resuming normal operations
> [Mon Jan 09 00:00:02.531559 2023] [core:notice] [pid 506:tid 139697910865216] AH00094: Command line: '/usr/sbin/apache2'
> 
> ```
> </details>
> 
> 
> <details>
> <summary>Database logs</summary>
> 
> ```
> 
> 2022-04-08 18:39:10 0 [Note] InnoDB: 10.3.34 started; log sequence number 3897183870; transaction id 8792780
> 2022-04-08 18:39:10 0 [Note] InnoDB: Loading buffer pool(s) from /var/lib/mysql/ib_buffer_pool
> 2022-04-08 18:39:10 0 [Note] Plugin 'FEEDBACK' is disabled.
> 2022-04-08 18:39:10 0 [Note] Server socket created on IP: '127.0.0.1'.
> 2022-04-08 18:39:10 0 [Note] Reading of all Master_info entries succeeded
> 2022-04-08 18:39:10 0 [Note] Added new Master_info '' to hash table
> 2022-04-08 18:39:10 0 [Note] /usr/sbin/mysqld: ready for connections.
> Version: '10.3.34-MariaDB-0+deb10u1'  socket: '/run/mysqld/mysqld.sock'  port: 3306  Debian 10
> 2022-04-08 18:39:11 0 [Note] InnoDB: Buffer pool(s) load completed at 220408 18:39:11
> 2022-04-08 18:48:42 0 [Note] /usr/sbin/mysqld (initiated by: unknown): Normal shutdown
> 2022-04-08 18:48:42 0 [Note] Event Scheduler: Purging the queue. 0 events
> 2022-04-08 18:48:42 0 [Note] InnoDB: FTS optimize thread exiting.
> 2022-04-08 18:48:42 0 [Note] InnoDB: Starting shutdown...
> 2022-04-08 18:48:42 0 [Note] InnoDB: Dumping buffer pool(s) to /var/lib/mysql/ib_buffer_pool
> 2022-04-08 18:48:42 0 [Note] InnoDB: Instance 0, restricted to 32768 pages due to innodb_buf_pool_dump_pct=25
> 2022-04-08 18:48:42 0 [Note] InnoDB: Buffer pool(s) dump completed at 220408 18:48:42
> 2022-04-08 18:48:43 0 [Note] InnoDB: Removed temporary tablespace data file: "ibtmp1"
> 2022-04-08 18:48:43 0 [Note] InnoDB: Shutdown completed; log sequence number 3898249611; transaction id 8794492
> 2022-04-08 18:48:43 0 [Note] /usr/sbin/mysqld: Shutdown complete

ZendaiOwl, 2023-01-14:
I think the updates are installing the PHP 8.2 version as in the issue you linked and we need to implement a package hold on the PHP packages so they don't get updated/upgraded to the 8.2 version from 8.1

theCalcaholic, 2023-01-21:
Fixed in the upcoming v1.51.0

closed 2023-01-21 by theCalcaholic