nextcloud/nextcloudpi #1693: Unable to login on a fresh install of v1.50.5 in LXC

negulescus, 2023-01-09:
Hello,

I did a fresh install of a Proxmox LXC instance for the today's version 1.50.5.

I see that root and webadmin account are no longer allowed:

https://github.com/nextcloud/nextcloudpi/blob/master/ncp.sh
line 132

  ## NCP USER FOR AUTHENTICATION
  id -u "$WEBADMIN" &>/dev/null || useradd --home-dir /nonexistent "$WEBADMIN"
  echo -e "$WEBPASSWD\n$WEBPASSWD" | passwd "$WEBADMIN"
  chsh -s /usr/sbin/nologin "$WEBADMIN"
  chsh -s /usr/sbin/nologin root

**What user credentials shall one use when logging in into the machine?
What am I missing?**

Thank you,

**System information**
Unable to provide as cannot login.

ZendaiOwl, 2023-01-14:
Can you access the web interface? The SSH activation in the admin panel should allow you to create a user to use

negulescus, 2023-01-14:
For the love of God, I cannot find that SSH option in the nextcloudpi admin panel. (I know it was available in the older versions.)
But, even if I activate SSH, _what good is a phone call if you are unable to speak?_ The **root** and **ncp** users are having thier shells pointed to "usr/sbin/nologin" (see the code snippet above).

ZendaiOwl, 2023-01-14:
The SSH activation will create a _new_ user for you that you can use for access. 

What happens if you try to access the container directly? 

`lxc exec ncp bash`

tteck, 2023-01-14:
@negulescus this could be of interest
https://github.com/tteck/Proxmox/discussions/925#discussioncomment-4686625

negulescus, 2023-01-14:
@ZendaiOwl 

- Thank you for the questions but, unfortunately I don't have access to that configuration to test as I have hacked the install script so it doesn't change the shell.

@tteck 

- Thank you too. The provided solution would works, but, imho, shouldn't nextcloudpi installation script be corrected, especially as SSH enablement is not available in the nextcloudpi web administration pages.

ZendaiOwl, 2023-01-16:
Yes you're right, it isn't available in the admin panel anymore, I missed I had fixed the issue that was preventing those from being disabled so that isn't available in the LXC and Docker images anymore because SSH access isn't necessary on base Docker or LXC images. It's possible to access them directly with the CLI and execute commands through the same CLI as well, for the regular installs the root user should be disabled for login per normal and users should make their own user to login with the SSH activation.

The exception here becomes the web interface in Proxmox that uses noVNC, SPICE or xterm.js for access to the container terminal, the cli in Proxmox can still access it as normal. 

I opened a discussion question for @tteck if we can reference his install script for Proxmox on the website and in the README & let people use that instead for a Proxmox installation since he sets the configurations for a Proxmox install in that script which will allow you to access it in the Proxmox web interface.

theCalcaholic, 2023-01-21:
Thanks to @tteck for fixing this issue upstream. I think, this is the right place to solve it, because the issue is specific to the *combination of LXC and Proxmox*. Usually, we keep the SSH app disabled for all containers - docker, LXD and LXC - because you can easily run commands as any user via the container daemon. But I was unaware, that the Proxmox UI uses a different mechanism to provide shell access which requires the user to have a login shell.

closed 2023-01-21 by theCalcaholic