nextcloud/nextcloudpi #1695: Nextcloud has quit its work - welcomes you with a Service not available (Error 503) page

Krawei, 2023-01-09:
Dear community,

my NCPi-Instance has stopped to work.

nc-error log tells me:
[Mon Jan 09 22:38:48.091323 2023] [proxy_fcgi:error] [pid 3004:tid 547991392640] [client 192.168.2.209:49028] AH01067: Failed to read FastCGI header
[Mon Jan 09 22:38:48.091461 2023] [proxy_fcgi:error] [pid 3004:tid 547991392640] (104)Connection reset by peer: [client 192.168.2.209:49028] AH01075: Error dispatching request to : 

Output of "ncp-report":
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.50.5
NextCloudPi image    NextCloudPi_03-02-22
OS                   Debian GNU/Linux 11. 5.15.84-v8+ (aarch64)
automount            no
USB devices          sda 
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      28G/917G
rootfs usage         28G/917G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    24.0.8.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
Internet check       ok
public IP            ***REMOVED SENSITIVE VALUE***
public IP            ***REMOVED SENSITIVE VALUE***
Port check 80        closed
Port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               3:30

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "7": "nextcloudpi",
            "5": "nextcloudpi.local",
            "8": "nextcloudpi.lan",
            "3": "nextcloudpi",
            "11": "1.1.1.1",
            "1": "192.168.2.5",
            "14": "localhost",
            "20": "nc.daheim.dedyn.io",
            "21": "192.168.2.5",
            "22": "my.domain.xa"
        },
[... 89 lines cut ...]
{"reqId":"oVv3pDndurPX0zgRFsrF","level":4,"time":"2023-01-08T00:59:27+00:00","remoteAddr":"","user":"--","app":"serverinfo","method":"","url":"--","message":"Error during app loading: Redis server /var/run/redis/redis.sock:0 went away","userAgent":"--","version":"24.0.8.2","exception":{"Exception":"RedisException","Message":"Redis server /var/run/redis/redis.sock:0 went away","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/Memcache/Redis.php","line":48,"function":"get","class":"Redis","type":"->"},{"file":"/var/www/nextcloud/lib/autoloader.php","line":154,"function":"get","class":"OC\\Memcache\\Redis","type":"->"},{"function":"load","class":"OC\\Autoloader","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Bootstrap/Coordinator.php","line":207,"function":"class_exists"},{"file":"/var/www/nextcloud/lib/private/legacy/OC_App.php","line":172,"function":"isBootable","class":"OC\\AppFramework\\Bootstrap\\Coordinator","type":"->"},{"file":"/var/www/nextcloud/lib/private/legacy/OC_App.php","line":140,"function":"loadApp","class":"OC_App","type":"::"},{"file":"/var/www/nextcloud/lib/private/Console/Application.php","line":119,"function":"loadApps","class":"OC_App","type":"::"},{"file":"/var/www/nextcloud/console.php","line":99,"function":"loadCommands","class":"OC\\Console\\Application","type":"->"},{"file":"/var/www/nextcloud/occ","line":11,"args":["/var/www/nextcloud/console.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/private/Memcache/Redis.php","Line":48,"message":"Error during app loading: Redis server /var/run/redis/redis.sock:0 went away","exception":{},"CustomMessage":"Error during app loading: Redis server /var/run/redis/redis.sock:0 went away"}}
{"reqId":"FsYNDtkeyF9AToxKoOik","level":2,"time":"2023-01-08T05:26:03+00:00","remoteAddr":"","user":"--","app":"appstoreFetcher","method":"","url":"--","message":"Could not connect to appstore: cURL error 28: Operation timed out after 60001 milliseconds with 0 bytes received (see https://curl.haxx.se/libcurl/c/libcurl-errors.html) for https://apps.nextcloud.com/api/v1/apps.json","userAgent":"--","version":"24.0.8.2","data":{"app":"appstoreFetcher"}}
{"reqId":"Y7q_chJ4Ij8YEGaIASuTsAAAAw0","level":2,"time":"2023-01-08T13:04:52+00:00","remoteAddr":"192.168.2.247","user":"--","app":"no app in context","method":"POST","url":"/index.php/login","message":"Login failed: clarissa (Remote IP: 192.168.2.247)","userAgent":"iPhone (4) (Nextcloud iOS)","version":"24.0.8.2","data":[]}
{"reqId":"Y7q_ihJ4Ij8YEGaIASuTtAAADw0","level":2,"time":"2023-01-08T13:05:16+00:00","remoteAddr":"192.168.2.247","user":"--","app":"no app in context","method":"POST","url":"/index.php/login","message":"Login failed: clarissa (Remote IP: 192.168.2.247)","userAgent":"iPhone (4) (Nextcloud iOS)","version":"24.0.8.2","data":[]}
{"reqId":"Y7q_3W4wMOIHM87P5mevvwAAQAE","level":2,"time":"2023-01-08T13:06:39+00:00","remoteAddr":"192.168.2.250","user":"--","app":"no app in context","method":"POST","url":"/index.php/login","message":"Login failed: Roger (Remote IP: 192.168.2.250)","userAgent":"Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_6) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/13.1 Safari/605.1.15","version":"24.0.8.2","data":[]}
{"reqId":"Y7q_8W4wMOIHM87P5mevwgAASgE","level":2,"time":"2023-01-08T13:06:57+00:00","remoteAddr":"192.168.2.250","user":"--","app":"no app in context","method":"POST","url":"/index.php/login","message":"Login failed: Roger Rabbit (Remote IP: 192.168.2.250)","userAgent":"Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_6) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/13.1 Safari/605.1.15","version":"24.0.8.2","data":[]}
{"reqId":"Y7rANG4wMOIHM87P5mev3wAARAU","level":2,"time":"2023-01-08T13:08:05+00:00","remoteAddr":"192.168.2.128","user":"--","app":"no app in context","method":"POST","url":"/index.php/login","message":"Login failed: Roger (Remote IP: 192.168.2.128)","userAgent":"Mozilla/5.0 (Android 12; Mobile; rv:108.0) Gecko/108.0 Firefox/108.0","version":"24.0.8.2","data":[]}
{"reqId":"Y7rA0xJ4Ij8YEGaIASuUEQAAEiE","level":2,"time":"2023-01-08T13:10:45+00:00","remoteAddr":"192.168.2.250","user":"--","app":"core","method":"POST","url":"/index.php/login","message":"Login failed: 'Roger' (Remote IP: '192.168.2.250')","userAgent":"Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_6) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/13.1 Safari/605.1.15","version":"24.0.8.2","data":{"app":"core"}}
{"reqId":"Y7rA0xJ4Ij8YEGaIASuUEQAAEiE","level":2,"time":"2023-01-08T13:10:45+00:00","remoteAddr":"192.168.2.250","user":"--","app":"no app in context","method":"POST","url":"/index.php/login","message":"Login failed: nobody@notyourbusiness.xa (Remote IP: 192.168.2.250)","userAgent":"Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_6) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/13.1 Safari/605.1.15","version":"24.0.8.2","data":[]}
{"reqId":"Y7rB_G4wMOIHM87P5mewawAASQU","level":2,"time":"2023-01-08T13:15:42+00:00","remoteAddr":"192.168.2.247","user":"--","app":"no app in context","method":"POST","url":"/index.php/login","message":"Login failed: clarissa (Remote IP: 192.168.2.247)","userAgent":"iPhone (4) (Nextcloud iOS)","version":"24.0.8.2","data":[]}
{"reqId":"Y7rCyBJ4Ij8YEGaIASuUxAAABgM","level":2,"time":"2023-01-08T13:19:06+00:00","remoteAddr":"192.168.2.247","user":"clarissa","app":"core","method":"OPTIONS","url":"/remote.php/dav/principals/users/clarissa/","message":"Login failed: 'clarissa' (Remote IP: '192.168.2.247')","userAgent":"iOS/15.7.2 (19H218) dataaccessd/1.0","version":"24.0.8.2","data":{"app":"core"}}
{"reqId":"QtPFsprV21UamGvP3LEL","level":3,"time":"2023-01-08T21:05:02+00:00","remoteAddr":"","user":"--","app":"core","method":"","url":"--","message":"Error while running background job (class: OCA\\ServerInfo\\Jobs\\UpdateStorageStats, arguments: )","userAgent":"--","version":"24.0.8.2","exception":{"Exception":"OC\\DB\\Exceptions\\DbalException","Message":"An exception occurred while executing a query: SQLSTATE[08S01]: Communication link failure: 1053 Server shutdown in progress","Code":1053,"Trace":[{"file":"/var/www/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php","line":296,"function":"wrap","class":"OC\\DB\\Exceptions\\DbalException","type":"::"},{"file":"/var/www/nextcloud/apps/serverinfo/lib/StorageStatistics.php","line":85,"function":"executeQuery", [... cut ...]
{"reqId":"qwZs1dMnxR2xmTIcnskP","level":2,"time":"2023-01-09T05:26:03+00:00","remoteAddr":"","user":"--","app":"appstoreFetcher","method":"","url":"--","message":"Could not connect to appstore: cURL error 28: Operation timed out after 60002 milliseconds with 0 bytes received (see https://curl.haxx.se/libcurl/c/libcurl-errors.html) for https://apps.nextcloud.com/api/v1/apps.json","userAgent":"--","version":"24.0.8.2","data":{"app":"appstoreFetcher"}}

```
</details>

Can anyone tell me how to fix this issue? Is there a known problem?

ZendaiOwl, 2023-01-15:
I don't know, from the logs it seems the connection to redis went away, have you tried a reboot? 

You might want to post to the forum for support questions like this since it isn't immediately obvious what is wrong here without more investigation and the GitHub issues aren't really for support, the forum is.

filmoff, 2023-01-17:
Hi,

it is my first post on github and hope it is ok if I post this here:

I think I have the same problem. It is strange, when i start my windows nextcloud client (3.6.4) and it starts to sync a bigger amount of files I get the same failure. If the windows nextcloud client is closed, nextcloud starts to work again. But every time I start the nextcloud client it crashes.

Krawei, 2023-01-19:
I've solved the issue on my own.

I had 3 different versions of php installed on my system

- php8.0
- php8.1
- php 8.2
.
The active php-branch was 8.1, checked/set via
`update-alternatives --config-php`.

I issued a backup of my nextcloud instance via
`ncp-config` => Backup (you can use ncp-backup, too I think) and noted every installed php8.1-package via
`dpkg -l | grep -i php8.1`
Afterwards I've deinstalled any php-branch higher than php8.0 via
`apt purge php8.2* php8.1*`
and reboot my pie.

The upcoming check was unsatisfying, my apache showed my the "clear text php code page" when I tried to browse to my NC URL. So I re-installed the noted php8.1 packages mentioned above, noticed that my Nextcloud was reborn (with the default php-values I guess) after the upcoming reboot, and checked the NC admin settings page .

Here the NC told me to tweak the "max-size"-Values (
Error alert: "The PHP memory limit is below the recommended value of 512MB"), so I did in the 
- /etc/php/8.1/{cli,fpm}/php.ini and incresed them to 1G both.

The last "icing on the cake"-step was to enable HTTP/2 via
```
apt install php8.1-fpm (if not installed)

a2dismod php8.1 && a2dismod mpm_prefork && a2enmod proxy_fcgi setenvif mpm_event && systemctl restart apache2

a2enconf php8.1-fpm && systemctl restart apache2

```
and voila - everything works fast and decent for me :)

So I close this issue.

closed 2023-01-19 by Krawei