nextcloud/nextcloudpi #1697: OCC Segmentation fault after upgrading to v1.50.5

user251022, 2023-01-11:
After upgrading from last version to v1.50.5 the automatic file scan does not work any more. In the logs i found a Segmentation fault error. 

`Running nc-scan
/usr/local/bin/ncc: line 3: 2505817 Segmentation fault      ${SUDO} php /var/www/nextcloud/occ "$@"
/usr/local/bin/ncc: line 3: 2505820 Segmentation fault      ${SUDO} php /var/www/nextcloud/occ "$@"
Done. Press any key...`

appcoders, 2023-01-11:
Did you upgrade your system with apt upgrade? I got an issue after 8.2 was automatically installed.
```
apt remove php8.2-common php8.2-opcache php8.2-cli php8.2-phpdbg php8.2-readline
sudo bash -c 'source /usr/local/etc/library.sh && clear_opcache'

```

Nextcloud works but still is very slow.

This was the update output of apt:

```
   php8.1 (8.1.13-1+0~20221126.29+debian11~1.gbpfee7cc => 8.1.14-1+0~20230107.31+debian11~1.gbpef7e32)
   php8.1-bcmath (8.1.13-1+0~20221126.29+debian11~1.gbpfee7cc => 8.1.14-1+0~20230107.31+debian11~1.gbpef7e32)
   php8.1-bz2 (8.1.13-1+0~20221126.29+debian11~1.gbpfee7cc => 8.1.14-1+0~20230107.31+debian11~1.gbpef7e32)
   php8.1-cli (8.1.13-1+0~20221126.29+debian11~1.gbpfee7cc => 8.1.14-1+0~20230107.31+debian11~1.gbpef7e32)
   php8.1-common (8.1.13-1+0~20221126.29+debian11~1.gbpfee7cc => 8.1.14-1+0~20230107.31+debian11~1.gbpef7e32)
   php8.1-curl (8.1.13-1+0~20221126.29+debian11~1.gbpfee7cc => 8.1.14-1+0~20230107.31+debian11~1.gbpef7e32)
   php8.1-fpm (8.1.13-1+0~20221126.29+debian11~1.gbpfee7cc => 8.1.14-1+0~20230107.31+debian11~1.gbpef7e32)
   php8.1-gd (8.1.13-1+0~20221126.29+debian11~1.gbpfee7cc => 8.1.14-1+0~20230107.31+debian11~1.gbpef7e32)
   php8.1-gmp (8.1.13-1+0~20221126.29+debian11~1.gbpfee7cc => 8.1.14-1+0~20230107.31+debian11~1.gbpef7e32)
   php8.1-intl (8.1.13-1+0~20221126.29+debian11~1.gbpfee7cc => 8.1.14-1+0~20230107.31+debian11~1.gbpef7e32)
   php8.1-ldap (8.1.13-1+0~20221126.29+debian11~1.gbpfee7cc => 8.1.14-1+0~20230107.31+debian11~1.gbpef7e32)
   php8.1-mbstring (8.1.13-1+0~20221126.29+debian11~1.gbpfee7cc => 8.1.14-1+0~20230107.31+debian11~1.gbpef7e32)
   php8.1-mysql (8.1.13-1+0~20221126.29+debian11~1.gbpfee7cc => 8.1.14-1+0~20230107.31+debian11~1.gbpef7e32)
   php8.1-opcache (8.1.13-1+0~20221126.29+debian11~1.gbpfee7cc => 8.1.14-1+0~20230107.31+debian11~1.gbpef7e32)
   php8.1-readline (8.1.13-1+0~20221126.29+debian11~1.gbpfee7cc => 8.1.14-1+0~20230107.31+debian11~1.gbpef7e32)
   php8.1-xml (8.1.13-1+0~20221126.29+debian11~1.gbpfee7cc => 8.1.14-1+0~20230107.31+debian11~1.gbpef7e32)
   php8.1-zip (8.1.13-1+0~20221126.29+debian11~1.gbpfee7cc => 8.1.14-1+0~20230107.31+debian11~1.gbpef7e32)
   php8.2-cli (8.2.0-1+0~20221208.9+debian11~1.gbp03fa1d => 8.2.1-1+0~20230107.11+debian11~1.gbp326758)
   php8.2-common (8.2.0-1+0~20221208.9+debian11~1.gbp03fa1d => 8.2.1-1+0~20230107.11+debian11~1.gbp326758)
   php8.2-opcache (8.2.0-1+0~20221208.9+debian11~1.gbp03fa1d => 8.2.1-1+0~20230107.11+debian11~1.gbp326758)
   php8.2-phpdbg (8.2.0-1+0~20221208.9+debian11~1.gbp03fa1d => 8.2.1-1+0~20230107.11+debian11~1.gbp326758)
   php8.2-readline (8.2.0-1+0~20221208.9+debian11~1.gbp03fa1d => 8.2.1-1+0~20230107.11+debian11~1.gbp326758)
```

user251022, 2023-01-12:
That worked well - Thank!

appcoders, 2023-01-13:
After upgrading this evening from
php8.1:arm64 (8.1.14-1+0~20230107.31+debian11~1.gbpef7e32 to  8.1.14-2+0~20230113.32+debian11~1.gbp6a972c
after apt upgrade  everything works again.

closed 2023-01-14 by theCalcaholic