nextcloud/nextcloudpi #1700: lxc import fails Error: Backup is missing index.yaml

r1ckdu, 2023-01-13:
**System information**
Intel NUC7CJYH Ubuntu 22.02 LTS

When importing the latest release (lxc import NextcloudPi_LXD_v1.50.5.tar.gz)  the process reports:
Error: Backup is missing index.yaml

How can we get rid of this?

ZendaiOwl, 2023-01-13:
Could you give it a try with `lxc image import` and see if that works?

r1ckdu, 2023-01-13:
That does not return an error but lxc list is empty and lxc start ncp reports that the instance does not exist

ZendaiOwl, 2023-01-13:
`lxc image list images:`

https://www.cyberciti.biz/faq/how-to-list-vm-images-in-lxd-linux-containers/

ZendaiOwl, 2023-01-14:
I think the commands in the readme are not correct. I'll look into it when I have some time to spare, or if someone else comes along before then and can do it

theCalcaholic, 2023-01-14:
For reference, this is how we do it in the CICD pipeline:

```sh
lxc image import "NextcloudPi_LXD_v1.50.5.tar.gz" --alias "nextcloudpi"
lxc launch "nextcloudpi" ncp
```

So that *should* work

r1ckdu, 2023-01-14:
I tried with a bit different naming but yeah it works!

r1ckdu, 2023-01-14:
after doing the launch one can start it with

`lxc start ncp`

ZendaiOwl, 2023-01-16:
I've updated the commands on the website and an update for the readme is pending in a pull request so I'm closing this issue

closed 2023-01-16 by ZendaiOwl