nextcloud/nextcloudpi #1702: Updated to NCPi version 24.0.5 but updater shows incompatibility for version 24.0.8

OfficialMuffin, 2023-01-14:
This [update](https://github.com/nextcloud/nextcloudpi/releases/tag/v1.50.5) shows that it updates NC to version 24.0.8, however, the web updater shows that NextcloudPi is incompatible for this new version of NC

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.50.5
NextCloudPi image    NextCloudPi_RaspberryPi_v1.50.3.img
OS                   . 5.10.92-v8+ (aarch64)
automount            yes
USB devices          sda sdb
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      350G/917G
rootfs usage         350G/917G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
public IP            ***REMOVED SENSITIVE VALUE***
public IP            ***REMOVED SENSITIVE VALUE***
Port check 80        closed
Port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               2days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

This version of Nextcloud is not compatible with PHP>=8.2.<br/>You are currently running 8.2.1.
```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Sat Jan 14 00:00:02.348562 2023] [ssl:warn] [pid 808:tid 547632530496] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Sat Jan 14 00:00:02.348830 2023] [ssl:error] [pid 808:tid 547632530496] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=fv-az504-71 / issuer: CN=fv-az504-71 / serial: 327BBA51C67A8AB40811E2BF0D7D9195CB7139EB / notbefore: Oct  3 22:08:25 2022 GMT / notafter: Sep 30 22:08:25 2032 GMT]
[Sat Jan 14 00:00:02.348853 2023] [ssl:error] [pid 808:tid 547632530496] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Sat Jan 14 00:00:02.351909 2023] [mpm_event:notice] [pid 808:tid 547632530496] AH00489: Apache/2.4.54 (Debian) OpenSSL/1.1.1n configured -- resuming normal operations
[Sat Jan 14 00:00:02.351971 2023] [core:notice] [pid 808:tid 547632530496] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

tail: cannot open '/var/log/mysql/*.log' for reading: No such file or directory

```
</details>

ZendaiOwl, 2023-01-15:
The issue comes from the php-json package that is installed and when doing an `apt update` installs a couple php 8.2 packages, a fix is in the pipeline for the next release [here](https://github.com/nextcloud/nextcloudpi/commit/0d9d62f132f61ad7ffa8680fc2857e9e7e608e2d)

ZendaiOwl, 2023-01-16:
Can you try `sudo apt remove php-json php8.2-*`?

OfficialMuffin, 2023-01-16:
This won't break anything will it? It is my production environment

ZendaiOwl, 2023-01-16:
It shouldn't break anything, it didn't for me when I tested it out, but you can always interrupt it if you see it removes anything other than the `php-json` & a couple `php8.2` packages. 

A backup is always key to have ←

The test instance I tried with was clean, if there is other things installed for you I have no way of knowing that, so read closely what it will try to remove and don't use the `-y` flag

theCalcaholic, 2023-01-21:
Closing as duplicate of #1692

closed 2023-01-21 by theCalcaholic

OfficialMuffin, 2023-01-22:
This seems to still be an issue even with the php-json and some php8.2 packages uninstalled. Also tried restarting server but the WebUI now states an incompatibility with NC version 24.0.9. My server is currently on 24.0.5.
The new update for NCPi should bump it to 24.0.8, am I correct?

theCalcaholic, 2023-01-23:
@OfficialMuffin Correct, which version of NCP are you on (check the version number in the upper left in the web UI)?

OfficialMuffin, 2023-01-23:
@theCalcaholic Unable to find the version number in the Web UI but upon accessing server with SSH, Nextcloud Pi version is 1.51.0 and "up to date"

theCalcaholic, 2023-01-23:
Wait a second - just saw that you wrote "web updater" - are you referring to Nextcloud's internal updater or the NCP web UI? Because the former shouldn't be used and should actually be disabled on NCP. Instead, connect to the terminal (either SSH or directly) and run ncp-update-nc from the `ncp-config` command line tool.

theCalcaholic, 2023-01-23:
According to your error message you have PHP 8.2 modules installed. If `sudo apt remove php8.2*` doesn't resolve this, you may have to create and restore a backup

OfficialMuffin, 2023-01-23:
I have tried `sudo apt remove php8.2*` already. I just ignore the Web UI updater then? Or even wait for the next NCPi update for it to fix the issue.

OfficialMuffin, 2023-01-23:
FYI I don't use the Web UI to update my NextcloudPi instance. It was purely to check everything was up to date

theCalcaholic, 2023-01-23:
The issue is already fixed in the latest version of NCP (1.51.0). What can't be fixed on our side, however, is the packages, that were installed due to this bug, because we cannot know if users installed them on purpose.

Therefore you will either have to restore your instance or fix the situation yourself, manually.

You can also attempt to upgrade nextcloud (this is best done from `ncp-config` in the terminal) and see if this helps - but I doubt it'll work as long as there are php 8.2 packages installed.

OfficialMuffin, 2023-01-26:
@theCalcaholic Thanks. I used ncp-config to update NC version to latest. All works fine now except the Web UI, nextcloud icons have gone missing. Not sure if that's PHP related but I'll check that tomorrow.

theCalcaholic, 2023-01-26:
@OfficialMuffin If it's just icons, then it sounds like a caching issue. Try another browser, if the icons appear there, it's your browser cache, if not, try restarting the web server or just wait for it to resolve by itself :)

OfficialMuffin, 2023-01-27:
@theCalcaholic 
Rebooted server, now getting `503 Service Unavaliable` when trying to access WebUI. When I checked the `/var/log/apache2/error.log`, this is what it contained:

```
[Fri Jan 27 00:02:38.530185 2023] [ssl:warn] [pid 826:tid 547736794176] AH01906: localhost:4443:0 server certificate is a CA certificate (BasicConstraints: CA == TRUE !?)
[Fri Jan 27 00:02:38.530219 2023] [ssl:warn] [pid 826:tid 547736794176] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Fri Jan 27 00:02:38.530401 2023] [ssl:error] [pid 826:tid 547736794176] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: emailAddress=REDACTED,CN=next,O=Internet Widgits Pty Ltd,ST=Some-State,C=GB / issuer: emailAddress=REDACTED,CN=next,O=Internet Widgits Pty Ltd,ST=Some-State,C=GB / serial: 5422C5EB08DDD55F1644948AC76221A8A638736D / notbefore: Jan 26 23:57:56 2023 GMT / notafter: Jan 26 23:57:56 2024 GMT]
```

Do I need to open another issue for this?

OfficialMuffin, 2023-01-27:
After 5 hours of trying to find the issue and reissuing multiple SSL certificcates, I have finally found a solution to the above.

`sudo systemctl reload apache2.service`
`sudo systemctl restart apache2.service`