nextcloud/nextcloudpi #1704: NCP reports -  Nextcloud is currently down - but my nextcloud isn't

stoneubi, 2023-01-15:
I updated from 1.50.4 to 1.50.5. My nextcloud version is 24.0.5. I tried to update nc to the latest version in ncp-web-interface I get the following message: Nextcloud is currently down. 

Trying to update all nc-apps also bia the web-interface shows: [ nc-update-nc-apps ] (Sun Jan 15 20:40:22 CET 2023)
This version of Nextcloud is not compatible with PHP>=8.2.<br/>You are currently running 8.2.1.

I havn't manually updated to version 8.2.1. 

Can you please let me know what I should do to update to latest version of nc 24.0.8?

Many thanks
Stoneubi

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.50.5
NextCloudPi image    NextCloudPi_07-17-22
OS                   Debian GNU/Linux 11. 5.10.0-20-amd64 (x86_64)
automount            no
USB devices          none
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      11G/63G
rootfs usage         11G/63G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
public IP            ***REMOVED SENSITIVE VALUE***
public IP            ***REMOVED SENSITIVE VALUE***
Port check 80        closed
Port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               0min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

This version of Nextcloud is not compatible with PHP>=8.2.<br/>You are currently running 8.2.1.
```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Sun Jan 15 19:48:04.275101 2023] [ssl:warn] [pid 554:tid 139732686380352] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Sun Jan 15 19:48:04.279286 2023] [:notice] [pid 554:tid 139732686380352] ModSecurity for Apache/2.9.3 (http://www.modsecurity.org/) configured.
[Sun Jan 15 19:48:04.279299 2023] [:notice] [pid 554:tid 139732686380352] ModSecurity: APR compiled version="1.7.0"; loaded version="1.7.0"
[Sun Jan 15 19:48:04.279306 2023] [:notice] [pid 554:tid 139732686380352] ModSecurity: PCRE compiled version="8.39 "; loaded version="8.44 2020-02-12"
[Sun Jan 15 19:48:04.279310 2023] [:warn] [pid 554:tid 139732686380352] ModSecurity: Loaded PCRE do not match with compiled!
[Sun Jan 15 19:48:04.279313 2023] [:notice] [pid 554:tid 139732686380352] ModSecurity: LUA compiled version="Lua 5.1"
[Sun Jan 15 19:48:04.279317 2023] [:notice] [pid 554:tid 139732686380352] ModSecurity: YAJL compiled version="2.1.0"
[Sun Jan 15 19:48:04.279324 2023] [:notice] [pid 554:tid 139732686380352] ModSecurity: LIBXML compiled version="2.9.10"
[Sun Jan 15 19:48:04.279328 2023] [:notice] [pid 554:tid 139732686380352] ModSecurity: Original server signature: Apache
[Sun Jan 15 19:48:04.279410 2023] [:notice] [pid 554:tid 139732686380352] ModSecurity: StatusEngine call: "2.9.3,Apache,1.7.0/1.7.0,8.39/8.44 2020-02-12,Lua 5.1,2.9.10,ae"
[Sun Jan 15 19:48:14.286807 2023] [:notice] [pid 554:tid 139732686380352] ModSecurity: StatusEngine call failed. Query: GIXDSLRTFRAXAYLDNBSSYMJOG4XDALZR.FY3S4MBMHAXDGOJPHAXDINBAGIYDEMBN.GAZC2MJSFRGHKYJAGUXDCLBSFY4S4MJQ.FRQWK.1673808484.status.modsecurity.org
[Sun Jan 15 19:48:14.351852 2023] [ssl:warn] [pid 1896:tid 139732686380352] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Sun Jan 15 19:48:14.355180 2023] [mpm_event:notice] [pid 1896:tid 139732686380352] AH00489: Apache/2.4.54 (Debian) OpenSSL/1.1.1n   configured -- resuming normal operations
[Sun Jan 15 19:48:14.355214 2023] [core:notice] [pid 1896:tid 139732686380352] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

tail: '/var/log/mysql/*.log' kann nicht zum Lesen geöffnet werden: Datei oder Verzeichnis nicht gefunden

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

```
</details>

ZendaiOwl, 2023-01-15:
The cause is due to the deprecated `php-json` package being installed (it's now included in the base php), this causes `apt update` to install some php 8.2 packages that is incompatible with the nextcloud version

A fix is in the pipeline in `devel`

stoneubi, 2023-01-15:
Thank you for the explanation.

closed 2023-01-15 by stoneubi