nextcloud/nextcloudpi #1705: Issues with upgrade 1.50.2 to 1.50.5 on docker

feutl, 2023-01-17:
I tried to replace my 1.50.2 docker container today with 150.5 docker. I am using portainer to manage my containers, so a simple "recreate" pulled the new image and with keeping the settings the new on got installed.
Perhaps it is important that I migrated from 1.50.2 native install on debian to the docker version some weeks age (was a hazzle free migration ;) and a hazzle free NC experience from the beginning ;) )

The install script / bootup scripts ran without an error, nextcloud was available but I got greeted with some errors.
First one was, that my user was not able to list the data directory. Opening the files tab always threw the same error, that this directory is not available, see the logs - well the logs did not raise anything useful.
My data directory is mounted as a persistent volume to /media/nextcloud-data and the config.php is adopted to it, it worked perfectly with my native NCP installation and also with 1.50.2 on docker it worked.

The other problem popping up ... see screenshot - it is in German but I hope you get the problem
[image]

Because I did not know where to start with the data thingy issue ;) I rolled back to 1.50.2 docker image and voila - all worked again.
I do not know what has changed on 1.50.5 but it breaks at least my installation and I have no clue where to start fixing it.

Thanks

ZendaiOwl, 2023-01-17:
The Docker data directory has changed I believe :) 

Best @theCalcaholic takes a look at it when he's feeling better from his fever and isn't sick

theCalcaholic, 2023-01-17:
@feutl A custom location for the data directory is unsupported for the docker instance and can be overwritten or assumed to be somewhere else by the ncp administration tools.

The intended way to run ncp docker is to mount your data location to `/data` inside the container, which will contain not only your user files, but also the database, the nextcloud directory (including configuration) and backups (by default at least). 

But if you want to separate the user files from everything else, you can just mount a host directory to `/data/ncdata/data` for ncp versions >= 1.50.5 or to `/data/nextcloud/data` for older versions of ncp.

feutl, 2023-01-17:
Oh, that would at least explain the problems.
But I could not find any documentation telling me that the mounted volume MUST be /data
I mean the docker example shows /data as the directory, but I assumed, because I can change the config.php information anyhow this does not matter that much.

But ok, I will try migrating the information to be /data and mount my volume accordingly. Especially with 1.50.5 I can at least separate the user data ;) which is actually the reason I mounted the data directory to a different mountpoint.

Thanks and I will give it a try and let you know if this worked out as intended.

@theCalcaholic get well soon :)

theCalcaholic, 2023-01-17:
Thank you :) 

Basically, ncp never expects (and discourages) fiddling with configuration files.
The concept of ncp is to have those maintained for you, which means they are auto generated (or auto edited) and your changes can be overwritten.

Be careful not to loose your database and nextcloud files when you change the mount point! You need to migrate everything from `/data` first.

theCalcaholic, 2023-01-17:
Btw, I assume you still have a docker volume mounted to `/data`, right? Otherwise you would've lost your database during the update

feutl, 2023-01-17:
sure /data is mounted as a volume, and all the necessary files are in there including the database ;)
thats not the problem, I just mounted my NC-Data-Dir to a different mountpoint so I can resuse the existing config.php 
As I mentioned already, I migrated from a native ncp installation to a docker based one, imported the ncp-backup to recreate ncps config in docker.

but no problem moving the data-dir is something I will do in the evening and let you know

feutl, 2023-01-17:
Seems to work right now, have upgrade to 1.50.5 and changed the data directory and the system seems to be up and running. What I do not see is NC 24.0.8 installed as mentioned in the release notes for 1.50.5. I am still on 24.0.5

And the question is still what "Install various templates on docker container start (improves config persistence)" means, I see that something with opcache is getting executed wit 1.50.5
Is this somehow related to https://github.com/nextcloud/nextcloudpi/issues/1480

theCalcaholic, 2023-01-21:
@feutl You need to perform the nextcloud update manually (or have ncp-autoupdate-nc enabled).

Regarding the automated template provisioning - that just means that some settings that would not survive a container replacement previously, do now :)

closed 2023-01-21 by theCalcaholic