nextcloud/nextcloudpi #1706: Problem when trying to create docker container with v1.50.5

bj-rn, 2023-01-18:
I am running NCP in ContainerStation on a Qnap Nas. Worked flawlessly up to _1.49.1_, then I borked it somehow by trying to update to _1.50.5_. Now I am trying to create a container from scratch (hopefully getting my latest backup to run) but encounter the following problem:

```
Initializing empty volume..
Making /usr/local/etc/ncp-config.d persistent ...
Making /etc/letsencrypt persistent ...
Making /etc/shadow persistent ...
Making /etc/cron.d persistent ...
Making /etc/cron.daily persistent ...
Making /etc/cron.hourly persistent ...
Making /etc/cron.weekly persistent ...
Making /usr/local/bin persistent ...
Installing template 'php/opcache.ini.sh'...
Installing template 'php/90-ncp.ini.sh'...
Installing template 'php/pool.d.www.conf.sh'...
Installing template 'ncp-metrics.cfg.sh'...
Making /etc/apache2/sites-available persistent ...
Making /etc/apache2/sites-enabled persistent ...
Starting PHP-fpm
Starting Apache
Installing template 'mysql/90-ncp.cnf.sh'...
INFO: Docker container detected.
Starting mariaDB
2023-01-18 14:15:51 0 [Note] mysqld (mysqld 10.5.18-MariaDB-0+deb11u1) starting as process 462 ...
2023-01-18 14:15:51 0 [Warning] The parameter innodb_file_format is deprecated and has no effect. It may be removed in future releases. See https://mariadb.com/kb/en/library/xtradbinnodb-file-format/
2023-01-18 14:15:51 0 [Warning] The parameter innodb_buffer_pool_instances is deprecated and has no effect.
2023-01-18 14:15:51 0 [Note] InnoDB: Uses event mutexes
2023-01-18 14:15:51 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
2023-01-18 14:15:51 0 [Note] InnoDB: Number of pools: 1
2023-01-18 14:15:51 0 [Note] InnoDB: Using crc32 + pclmulqdq instructions
2023-01-18 14:15:51 0 [Note] mysqld: O_TMPFILE is not supported on /tmp (disabling future attempts)
2023-01-18 14:15:51 0 [Note] InnoDB: Using Linux native AIO
2023-01-18 14:15:51 0 [Note] InnoDB: Initializing buffer pool, total size = 268435456, chunk size = 134217728
2023-01-18 14:15:51 0 [Note] InnoDB: Completed initialization of buffer pool
2023-01-18 14:15:52 0 [Note] InnoDB: 128 rollback segments are active.
2023-01-18 14:15:52 0 [Note] InnoDB: Creating shared tablespace for temporary tables
2023-01-18 14:15:52 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wait ...
2023-01-18 14:15:52 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
2023-01-18 14:15:52 0 [Note] InnoDB: 10.5.18 started; log sequence number 1187908; transaction id 2173
2023-01-18 14:15:52 0 [Note] Plugin 'FEEDBACK' is disabled.
2023-01-18 14:15:52 0 [Note] InnoDB: Loading buffer pool(s) from /data/database/ib_buffer_pool
2023-01-18 14:15:52 0 [Note] Server socket created on IP: '127.0.0.1'.
2023-01-18 14:15:52 0 [Note] Reading of all Master_info entries succeeded
2023-01-18 14:15:52 0 [Note] Added new Master_info '' to hash table
2023-01-18 14:15:52 0 [Note] mysqld: ready for connections.
Version: '10.5.18-MariaDB-0+deb11u1'  socket: '/run/mysqld/mysqld.sock'  port: 3306  Debian 11
2023-01-18 14:15:52 0 [Note] InnoDB: Buffer pool(s) load completed at 230118 14:15:52
Starting Redis
Starting Cron
Starting Postfix
postfix/postfix-script: starting the Postfix mail system
Provisioning
Provisioning Redis password
Updating NextCloud config with Redis password
Provisioning MariaDB password
Updating NextCloud config with MariaDB password
Running nc-limits
Using 6177223680 for PHP max memory
Installing template 'php/90-ncp.ini.sh'...
Using 6 PHP threads
Installing template 'php/pool.d.www.conf.sh'...
Installing template 'mysql/91-ncp.cnf.sh'...
Stopping MariaDB database server: mariadbd2023-01-18 14:16:04 0 [Note] mysqld (initiated by: root[root] @ localhost []): Normal shutdown
2023-01-18 14:16:04 0 [Note] Event Scheduler: Purging the queue. 0 events
2023-01-18 14:16:04 0 [Note] InnoDB: FTS optimize thread exiting.
2023-01-18 14:16:04 0 [Note] InnoDB: Starting shutdown...
2023-01-18 14:16:04 0 [Note] InnoDB: Dumping buffer pool(s) to /data/database/ib_buffer_pool
2023-01-18 14:16:04 0 [Note] InnoDB: Buffer pool(s) dump completed at 230118 14:16:04
2023-01-18 14:16:04 0 [Note] InnoDB: Removed temporary tablespace data file: "ibtmp1"
2023-01-18 14:16:04 0 [Note] InnoDB: Shutdown completed; log sequence number 1187920; transaction id 2176
[... 77 lines cut ...]
2023-01-18 14:31:28 0 [Note] InnoDB: Dumping buffer pool(s) to /data/database/ib_buffer_pool
2023-01-18 14:31:28 0 [Note] InnoDB: Buffer pool(s) dump completed at 230118 14:31:28
2023-01-18 14:31:28 0 [Note] InnoDB: Removed temporary tablespace data file: "ibtmp1"
2023-01-18 14:31:28 0 [Note] InnoDB: Shutdown completed; log sequence number 1102166; transaction id 1926
2023-01-18 14:31:28 0 [Note] mysqld: Shutdown complete

.
Starting MariaDB database server: mariadbd.
Stopping redis-server: redis-server.
Starting redis-server: /etc/init.d/redis-server: 51: ulimit: error setting limit (Operation not permitted)
redis-server.
Starting notify_push daemon
Configuring Domain
System config value trusted_domains => 11 set to string 178.202.6.21
System config value trusted_domains => 1 set to string 192.168.178.222
System config value trusted_domains => 14 set to string Bolt
Nextcloud version 24.0.5.1. NextCloudPi version v1.50.2
Init done
```

Reading through the releases changelog I assume it might be related releated to:
_* Move default location of data directory out of nextcloud directory (partial fix for https://github.com/nextcloud/nextcloudpi/issues/1417)_

In both cases `data `and `var` are mounted as follows:
[image]

theCalcaholic, 2023-01-19:
You don't need to worry much about this error.

Is anything else broken? Please include the output of ncp-report in this issue

bj-rn, 2023-01-21:
Sorry for the late answer. Was totally fixated on getting my data back and forgot about this issue. Haven't got the non-working container anymore and unfortunately can't reproduce the issue...

closed 2023-01-21 by bj-rn