nextcloud/nextcloudpi #1707: Nextcloud 27 support

sunjam, 2023-01-18:
[Removal of PSR 0 class loader](https://help.nextcloud.com/t/breaking-removal-of-psr-0-class-loader-in-nextcloud-27/153916) announcement.
- https://github.com/nextcloud/server/pull/36114

sunjam, 2023-04-27:
update from the developers on shortened release cycle:

> Hi there, coming with some news today.
> If you stalk https://github.com/nextcloud/server/wiki/Maintenance-and-Release-Schedule you might have noticed already. But Nextcloud 27 is just around the corner. It was decided to go with a shorter Nextcloud 27 cycle and a longer Nextcloud 28 cycle to better fit some internal schedules.
> 
> 27 Beta 1 aka. Feature freeze is on 4th May (exactly 1 week from now)
> 27 RC 1 aka. String and UI freeze is on 17th May (moved to Wednesday as Thursday is public holiday in Germany and more)
> The final date is not yet settled, but should at least be 2 weeks after RC1

sunjam, 2023-05-06:
[27.0.0beta1](https://github.com/nextcloud/server/tags) available for testing.

janvlug, 2023-08-26:
I just upgraded to 27.0.1. Should this issue be closed? Version 27.0.2 is available as well though.

theCalcaholic, 2023-08-31:
@janvlug  Yes, this should be closed

closed 2023-08-31 by theCalcaholic