nextcloud/nextcloudpi #1715: Update leads to rollback with all files deleted.

monsieurhannes, 2023-01-23:
Hi,

I have a strange behavior that should get mentioned here, I think. I lose all files when trying to update. 
I'll post the log-output later, when I get back to my computer (sorry, I'm a bit limited when the cloud doesn't work). 

## In short:

I have `NC 24.0.7` installed and wanted to update via CLI (`ncp-config`) with the version set to `0`. Which in my case meant `NC 24.0.8`, as it doesn't seem to automatically jump to `NC 25.0.3`
First it checks if there is enough space and everything, all good, it creates a backup and the next thing that happens is, that it complains that there is already a `nextcloud-old` directory.
Instead of overwriting the directory or creating `nextcloud-old-old`, it does a rollback. 

Unfortunately the rollback is a proper rollback to the bare minimum. ***All files get deleted***, there is nothing left, the folder goes from 3,5 TB to 17GB - funny, the backup before the update is still in there 😄 

At the moment I extract my main backup, that's why the information below is not complete.
It's the second time this happens. I was using the latest Docker image ([v1.51.0](https://hub.docker.com/layers/ownyourbits/nextcloudpi/v1.51.0/images/sha256-0ea6164efbd554b1f5d870a6f366a72c0e9ca9074b19a7345adb86809887affe?context=explore)), the time before (a couple of weeks ago) I was using the image from back then ([v1.50.2](https://hub.docker.com/layers/ownyourbits/nextcloudpi/v1.50.2/images/sha256-2459e0fc3ace666edc2bffff193112fff5a5d5e7e14f4cd1175a0a32da48eb1d?context=explore)).

Any way to reduce the risk of losing files?

## System information

ODROID N2, Armbian Bullseye
NextcloudPi on Docker
Can't do a `ncp-report` right now as I'm working on transferring my backup.

theCalcaholic, 2023-01-23:
Hey, sorry to hear that. I'm not yet sure how that happened, but if you initialize nextcloudpi from version 1.50.5 or later, the risk for something like this will be minimal, because the data directory isn't kept inside the nextcloud directory anymore (which is probably why everything was removed at once in your case).

There should be a backup, however, that was created during the update and you should ve able to restore that

monsieurhannes, 2023-01-23:
Hi,
no worries, I like my borgbackup and it's quite fast so I got used to do it before any bigger update. It just takes forever to extract...

Now I'm at home and got my copy of the install logs ready:

```
Running nc-update-nextcloud
Current   Nextcloud version 24.0.7.1
Available Nextcloud version 24.0.8
Download Nextcloud 24.0.8...
Back up current instance...
check free space...
Maintenance mode enabled
backup database...
backup files...
backup /data/nextcloud-bkp_20230122_1674418767.tar.gz generated
Maintenance mode disabled
Install Nextcloud 24.0.8...
mv: cannot move 'nextcloud' to 'nextcloud-old': Directory not empty
Abort
Simple roll back...
```

Once the `simple roll back` is done, everything is gone. Your comment with initialization doesn't apply to me I guess, as I don't want to initialize this 5 year old nextcloud - is it even possible from the backup? 
Next time I update, do I need to think of something? I'll obviously rename or delete the `nextcloud-old` folder, but is there anything else I can do to prevent this? It just takes so much time to extract my backup :D 

And in terms of "future-proofness", would it be good to change the simple roll back to something which doesn't delete everything?

theCalcaholic, 2023-01-23:
> [...] as I don't want to initialize this 5 year old nextcloud - is it even possible from the backup?

Sure, every NCP instance can be recreated from an ncp-backup at any time :)

So after I've seen your logs, I suspect, that it failed to copy the nextcloud directory to nextcloud-old and then restored things from the *old* nextcloud-old directory.
In basically all other steps the updater should have been able to restore your instance to its previous state :D

I'll make sure to add a check for this.

theCalcaholic, 2023-01-23:
Nevermind, I found the issue. The updater didn't respect the different base directory on docker (namely `/data` instead of `/var/www`). I'll push a fix soon.

closed 2023-01-23 by theCalcaholic

monsieurhannes, 2023-01-25:
Thank you for your help!
I tried to update yesterday, before that I deleted the previous `nextcloud-old`-directory. But now my Nextcloud is not initialized. I wrote about it in the forum: https://help.nextcloud.com/t/ncp-update-led-to-not-initialized-nextcloud/154378

Is that due to the previous problem, or something different?

PR #1726 (2023-03-26): Fix various issues related to updating Nextcloud
Fixes #1715
Fixes #1633
Fixes #1694