nextcloud/nextcloudpi #1716: NextcloudPi installation destroys root user

gilbert-grape, 2023-01-23:
**System information**
Raspberry 4 with 8GB ram
Fresh installation of the last Raspbian OS 64bit.

I only activaed the SSH connection and started with the installation script:
```
sudo su
curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | sudo bash

```

After the installation, the root user is broken.
```
pi@mycloud:~ $ sudo su
This account is currently not available.
```

if I open a second shell and test the command "sudo su" where the installation script is running, I see somewhere in the middle it happens, that the systemuser get killed.
Do you have an idea, what the problem is?

jiridienstbier, 2023-01-23:
I have the same problem, but no solution yet. As I was wondering if this is an issue at my config, it helps to see it seems to be generic.

gilbert-grape, 2023-01-23:
The big question is: what do we do now?
Can we install an older version?

gilbert-grape, 2023-01-23:
Update: if I follow the installation logs on the screen, i see "installing nextcloud", the user root exists. Then follows "installing ncp" and some texts with root user... And then it is broken. It is definitly the ncp installation.

jiridienstbier, 2023-01-23:
Thanks, with this info I can continue :-)

gilbert-grape, 2023-01-23:
After about 10-20 retries. I have found my solution: I use an old image, the version 1.50 works very well. 1.51 is full of errors. there are some more errors like the ip address is not in trusted domains. And a lot of errors after missing root user like documents .md or .doc could not edited.

theCalcaholic, 2023-01-23:
@gilbert-grape Sounds like your installation failed badly and resulted in a corrupted system.
There's not much you can do, but retry the installation (with the same or another version, as you have done)

closed 2023-01-23 by theCalcaholic

iT-Boyer, 2023-02-13:
I have the same problem when installing her on the vultr host. However, this problem causes `SSH` login failure.
```
ssh root@64.*.*.254
```
[image]

v1ckxy, 2023-02-27:
About this topic:

The point is that ncp installer doesn't check for sudo being installed nor available, and upon a basic debian install, you'll be on a lockdown status.

At least, installer should check for sudo being installed and install if it's not found.

luxzg, 2023-03-13:
> This is indeed intended behavior to avoid security issues on some platforms (e. g. armbian).
> 
> You can still get a root shell if you must with `sudo su -s /bin/bash`

While this gets me to root prompt on Raspberry Pi 3, it's not exactly a welcome feature. I was hoping to use RPi with both NextCloudPi and some minor stuff (like using LAMP that's already there anyway, setting up certs, and using few minor *nix tools). Now it looks like I'll need to change years of "muscle memory" typing usual (and very unsecure) "sudo su" then working unobstructed... Guess I'll need to simply add command to run as soon as I login to SSH

Can't we get it back same way it was destroyed and disabled? For us stuborn mules that like our "sudo su"?

maltepost, 2023-06-08:
I have been a very happy user of Nextcloud for many years and am very grateful for the excellent work done here by everyone involved.

And yet, I have to express my criticism one time. I cannot understand why it is suddenly no longer possible to access Nextcloud as root. I understand that this is to avoid a security risk. However, I have the impression that the consideration did not properly take into account that this will result in a great deal of additional work for countless previously satisfied users. For example, it is no longer possible to simply edit config.php with an SSH client or even to automatically download ncp backups.

I assume that the users who activate SSH can decide for themselves whether they are willing to take a ( supposed ) security risk or not. I think it is wrong to suddenly decide that. Hence my request: Reconsider this and, if possible, please reactivate the root user!

theCalcaholic, 2023-06-11:
@luxzg @maltepost This is just a default setting - nothing prevents you from re-enabling the root user (especially if you're already fiddling with config files).

It's mainly meant to protect those users who don't need it or don't have a lot of experience in using a CLI.

Reenabling root login is as easy as `sudo chsh -s /bin/bash root`

guusec, 2023-08-01:
leaving a device running as root that you don't have root access to is definitely a choice. i don't know if i'd call it a "good" one though :)