nextcloud/nextcloudpi #1717: [Update NC 25 / PHP 8.1] Temp path in php.ini reverted to default

johndoe0815, 2023-01-23:
After updating Nextcloud to 25.0.3. I could not upload any files in anymore, always an error about different expected file size.

Digging through the Nextcloud forums, I found the issue:

In the php.ini, the default temp file path was set:
upload_tmp_dir = /var/www/nextcloud/data/tmp

But I had and have a different temp path. The files are on a mounted, external drive, and the temp files in its sub folder, both maintained in the config file.
So obviously the upgrade does not consider this and just sets the temp dir in the php.ini back to the default value.
Adjusting this manually in the php.ini fixes the issue, so the solution is easy, but finding it not.

The upgrade needs to consider the currently set temp file path and use that in the php.ini according.y

theCalcaholic, 2023-01-23:
Are you running ncp via docker? NCP always sets the temp path to be in the same place as the NC data directory, because it assumes that wherever your data is, should be enough disk space for caching.

If your data is not in /var/www/nextcloud/data that could mean, that we have a bug in the templating script for php.ini

Are you running NCP as docker container or otherwise?

johndoe0815, 2023-01-23:
I have it running in a LXC container in Proxmox. My datadir was and is 

/mnt/myCloudDrive/ncdata

and tmp dir therefore

/mnt/myCloudDrive/ncdata/tmp

johndoe0815, 2023-01-23:
Sorry, I have to correct myself. I looked in into a backup from 24.0.x, and also there the file path in php.ini was the default. I guess I just never adjusted it.
I had installed NextcloudPi under Proxmox and adjusted the paths directly in the config.php, as ncp-config under Proxmox does not have the option to adjust the datadir.
The odd thing is, that the default path in the php.ini did never matter, until I did the Nextcloud upgrade, then it led to upload errors with messages like
_Expected filesize of 10485760 bytes but read (from Nextcloud client) and wrote (to Nextcloud storage) 8192 bytes. Could either be a network problem on the sending side or a problem writing to the storage on the server side._
Adjusting the php.ini fixes that, I am just puzzled why this was not necessary before?!

theCalcaholic, 2023-01-23:
Alright, glad you could find the cause :)

closed 2023-01-23 by theCalcaholic