nextcloud/nextcloudpi #1719: tmpl_php_max_filesize and tmpl_php_threads: command not found when updaeting to v1.51.0

facorazza, 2023-01-24:
# Issue

When updating the docker image from 1.49.1 to 1.51.1 I get:

```
nextcloudpi                  | Install default configurations for new ncp apps
nextcloudpi                  | Installing template 'php/opcache.ini.sh'...
nextcloudpi                  | Installing template 'php/90-ncp.ini.sh'...
nextcloudpi                  | /usr/local/etc/ncp-templates/php/90-ncp.ini.sh: line 12: tmpl_php_max_filesize: command not found
nextcloudpi                  | Installing template 'php/pool.d.www.conf.sh'...
nextcloudpi                  | /usr/local/etc/ncp-templates/php/pool.d.www.conf.sh: line 14: tmpl_php_threads: command not found
nextcloudpi                  | Installing template 'ncp-metrics.cfg.sh'...
nextcloudpi                  | Starting PHP-fpm
nextcloudpi                  | [24-Jan-2023 20:27:48] ERROR: No pool defined. at least one pool section must be specified in config file
nextcloudpi                  | [24-Jan-2023 20:27:48] ERROR: failed to post process the configuration
nextcloudpi                  | [24-Jan-2023 20:27:48] ERROR: FPM initialization failed
nextcloudpi                  | Starting Redis
nextcloudpi                  | Starting Cron
nextcloudpi                  | Starting Postfix
nextcloudpi                  | ERROR 2002 (HY000): Can't connect to local MySQL server through socket '/run/mysqld/mysqld.sock' (2)
nextcloudpi                  | Back up current instance...
nextcloudpi                  | Error reading data directory. Is NextCloud running and configured?
nextcloudpi                  | WARN: Backup creation failed
nextcloudpi                  | Init done
```

There could be 3 potential issues:

* `tmpl_php_max_filesize` and `tmpl_php_threads` not found
* php-fpm: No pool defined
* Can't connect to local MySQL server

Let's start from the first one and let's see if that solves the other ones as well.

`tmpl_php_max_filesize` and `tmpl_php_threads` are called in [90-ncp.ini.sh](https://github.com/nextcloud/nextcloudpi/blob/master/etc/ncp-templates/php/90-ncp.ini.sh) and are defined in `nc-limits.sh`. The two functions have been introduced since [here](https://github.com/nextcloud/nextcloudpi/commit/b675d61e61e11f14581eb82faeb5407cf61e0137#diff-ff2649985a98286ed35017d93cb76d7be0df3864d7863ff9a9c9fb2379686300) since ncp version 1.50.1 so after the version I'm upgrading from (1.49.1).

The template is loaded in [000ncp](https://github.com/nextcloud/nextcloudpi/blob/master/build/docker/nextcloudpi/000ncp). Since it seems that the `/data/bin` directory contains the old version (1.49.1) of `nc-limits.sh` without the two functions the piece of code to further examine is the following:

```
# from 000ncp

# reset bin if we pull a more recent container
[[ -d /data/bin ]] && {
  date_img=$(date "+%s" -r /.docker-image)
  date_per=$(date "+%s" -r /data/bin)
  [[ $date_img -gt $date_per ]] && {
    rm -r /data/bin
  }
}
persistent_cfg /usr/local/bin /data/bin
```

I'll add more info if I can.

**System information**

ncp-report using the image 1.49.1

```
NextCloudPi version  v1.49.1
NextCloudPi image    NextCloudPi_docker_09-14-22
OS                   Debian GNU/Linux 11. 5.4.0-137-generic (x86_64)
automount            no
USB devices          sde 
datadir              /ncdata
data in SD           no
data filesystem      zfs
data disk usage      1.8T/2.8T
rootfs usage         75G/233G
swapfile             /swap.img
[... 134 lines cut ...]
```

{"reqId":"xmR12NLfp52XUJttFXwJ","level":1,"time":"2023-01-24T21:25:08+01:00","remoteAddr":"","user":"--","app":"passwords","method":"","url":"--","message":"Passwords runs /var/www/nextcloud/occ in global mode","userAgent":"--","version":"24.0.3.2","data":{"app":"passwords"}}
{"reqId":"LGwvwG7WNluHDRNqgc56","level":0,"time":"2023-01-24T21:25:14+01:00","remoteAddr":"","user":"--","app":"carnet","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"24.0.3.2","data":{"app":"carnet"}}
{"reqId":"LGwvwG7WNluHDRNqgc56","level":0,"time":"2023-01-24T21:25:14+01:00","remoteAddr":"","user":"--","app":"extract","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"24.0.3.2","data":{"app":"extract"}}
{"reqId":"LGwvwG7WNluHDRNqgc56","level":0,"time":"2023-01-24T21:25:14+01:00","remoteAddr":"","user":"--","app":"files_mindmap","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"24.0.3.2","data":{"app":"files_mindmap"}}
{"reqId":"LGwvwG7WNluHDRNqgc56","level":0,"time":"2023-01-24T21:25:14+01:00","remoteAddr":"","user":"--","app":"files_readmemd","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"24.0.3.2","data":{"app":"files_readmemd"}}
{"reqId":"LGwvwG7WNluHDRNqgc56","level":0,"time":"2023-01-24T21:25:14+01:00","remoteAddr":"","user":"--","app":"geoblocker","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"24.0.3.2","data":{"app":"geoblocker"}}
{"reqId":"LGwvwG7WNluHDRNqgc56","level":0,"time":"2023-01-24T21:25:14+01:00","remoteAddr":"","user":"--","app":"metadata","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"24.0.3.2","data":{"app":"metadata"}}
{"reqId":"LGwvwG7WNluHDRNqgc56","level":0,"time":"2023-01-24T21:25:14+01:00","remoteAddr":"","user":"--","app":"music","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"24.0.3.2","data":{"app":"music"}}
{"reqId":"LGwvwG7WNluHDRNqgc56","level":0,"time":"2023-01-24T21:25:14+01:00","remoteAddr":"","user":"--","app":"qownnotesapi","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"24.0.3.2","data":{"app":"qownnotesapi"}}
{"reqId":"LGwvwG7WNluHDRNqgc56","level":0,"time":"2023-01-24T21:25:14+01:00","remoteAddr":"","user":"--","app":"socialsharing_diaspora","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"24.0.3.2","data":{"app":"socialsharing_diaspora"}}
{"reqId":"LGwvwG7WNluHDRNqgc56","level":0,"time":"2023-01-24T21:25:14+01:00","remoteAddr":"","user":"--","app":"socialsharing_email","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"24.0.3.2","data":{"app":"socialsharing_email"}}
{"reqId":"LGwvwG7WNluHDRNqgc56","level":0,"time":"2023-01-24T21:25:14+01:00","remoteAddr":"","user":"--","app":"socialsharing_facebook","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"24.0.3.2","data":{"app":"socialsharing_facebook"}}
{"reqId":"LGwvwG7WNluHDRNqgc56","level":0,"time":"2023-01-24T21:25:14+01:00","remoteAddr":"","user":"--","app":"socialsharing_telegram","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"24.0.3.2","data":{"app":"socialsharing_telegram"}}
{"reqId":"LGwvwG7WNluHDRNqgc56","level":0,"time":"2023-01-24T21:25:14+01:00","remoteAddr":"","user":"--","app":"socialsharing_twitter","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"24.0.3.2","data":{"app":"socialsharing_twitter"}}
{"reqId":"LGwvwG7WNluHDRNqgc56","level":0,"time":"2023-01-24T21:25:14+01:00","remoteAddr":"","user":"--","app":"video_converter","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgen [... cut ...]
{"reqId":"LGwvwG7WNluHDRNqgc56","level":0,"time":"2023-01-24T21:25:15+01:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"new parser added : FeedIo\\Standard\\Rdf","userAgent":"--","version":"24.0.3.2","data":{"app":"news"}}
{"reqId":"LGwvwG7WNluHDRNqgc56","level":1,"time":"2023-01-24T21:25:15+01:00","remoteAddr":"","user":"--","app":"passwords","method":"","url":"--","message":"Passwords runs /var/www/nextcloud/occ in global mode","userAgent":"--","version":"24.0.3.2","data":{"app":"passwords"}}

```
</details>

facorazza, 2023-01-24:
Looking at the modification date of `/data/bin` it is more recent than the one contained in the image so the old `/data/bin`is never removed and the new one is never symlinked in-place.

facorazza, 2023-01-24:
I could manually fix it with:

```
touch -t <timestamp> /data/bin
```

facorazza, 2023-01-24:
Perhaps it could be possible to get the modification date of a file like .docker-image inside /data/bin rather than the directory itself. This is a very edge case that can be probably ignored and fixed manually. For me this issue can be closed.

facorazza, 2023-01-24:
P.S.
Fixing this issue also fixed the one related to php-fpm and mysql.

theCalcaholic, 2023-01-26:
Hey, thanks for the troubleshooting efforts. Do you think this could also be the cause for #1671?

facorazza, 2023-01-27:
> Hey, thanks for the troubleshooting efforts. Do you think this could also be the cause for the missing scripts in #1671?

To be honest I didn't check the web UI. Once resolved the issue, NC worked fine even though NCP didn't and kept saying that my NC version (24.0.3.2) was too old for the php version in use (8.2.1). This happened with both NCP version 1.50.5 and 1.51.0. For the moment I reverted the snapshot and went back to NCP 1.49.1.