nextcloud/nextcloudpi #1720: NextcloudPi logos do not show up correctly

OfficialMuffin, 2023-01-27:
When I iupdated my NextcloudPi's Nextcloud version using `ncp-config` to the latest version (25.0.3.2) then restarted the Nextcloud logos in the WebUI are missing. Even using an alternate browser does not help.

I have inspected the missing Nextcloud logos through the browser and the path of the logos are `/apps/theming/image/logo?useSvg=1&v=3`. Upon checking this in `/var/www/nextcloud/apps/theming/` the directory image doesnt exist. Could this be the issue?

EDIT:

Changing `/apps/theming/image/logo?useSvg=1&v=3` to `/core/img/logo/logo.svg` in the browsers developer mode seems to work. I wonder if anyone else is affected?

ttavenard, 2023-01-27:
I've had the same problem. I solved it by downloading the ncp logo from here, and reapplying it via nexcloud admin page/ theming / change logo

OfficialMuffin, 2023-01-27:
> I've had the same problem. I solved it by downloading the ncp logo from here, and reapplying it via nexcloud admin page/ theming / change logo

Thanks for the tip :)

closed 2023-01-27 by OfficialMuffin

dengesCU, 2023-01-28:
I had the same issue and could solve it by following @ttavenard 's advice (Thank you!) For the next person with this issue, here is the links from this repo where the files can be found (for me background was missing too):
https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/ncp-web/img/ncp-logo.svg
https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/ncp-web/img/background.png
Cheers

svenb1234, 2023-02-14:
Should this issue not be fixed by the next version of NCP? Not every user will find this report and be able to repair it by himself.

dengesCU, 2023-02-14:
@svenb1234 i second that request.

OfficialMuffin, 2023-02-14:
I found that a fresh install of NCP version 1.51.0 fixed the issue.