nextcloud/nextcloudpi #1723: Stop /etc/resolv.conf from being overwritten on reboot

OfficialMuffin, 2023-01-28:
**Description**  
When rebooting NectcloudPi, the /etc/resolv.conf file gets overwritten with:

`nameserver 127.0.0.1`

This prevents external network access and prevents DNS from working. Therefore all NextcloudPi features that require an internet connection stop working. I usually have `resolv.conf` that works, set up as:

```
nameserver 1.1.1.1
nameserver 1.0.0.1
```

Figured this shouldn't be a bug report as a setting in the NextcloudPi panel could be added to stop the system from overwriting the `resolv.conf` file. Just a suggestion.

theCalcaholic, 2023-01-30:
Manually editing resolv.conf is generally not advised anymore, afaik, but it depends on the exact system you are running. How are your running ncp?

OfficialMuffin, 2023-01-30:
It's just the basic NCP installation using the curl script.

theCalcaholic, 2023-01-31:
@OfficialMuffin In that case, you should use `systemd-resolved` (and it's cli `resolvectl`) to manage DNS resolution

closed 2023-01-31 by theCalcaholic