nextcloud/nextcloudpi #1724: Pulling and restarting Docker Image 1.51 will not upgrade NC

BstAA, 2023-01-29:
Latest NCP version supports NC 25.0.3.
It's not recommended to upgrade NC withing running Docker container.
But pulling latest image und restarting will leave NC at 24.0.8.2

ZendaiOwl, 2023-01-29:
You can upgrade NC within the Docker container but not NCP :)

BstAA, 2023-01-30:
THX! After once getting into trouble when updating NCP within the container, I thought NC will also be updated by puling the latest image.
Now after again looking for [Staying up to date](https://help.nextcloud.com/t/staying-up-to-date/126259) I've seen it's in the docs _but_ 

1. The docs on this point should be really reworked. 
2. As Docker is mentioned and needs special action, the Docker tag should be added

svenb1234, 2023-01-30:
From the docs you linked.

`Note Docker users should not update internally but rather update when the new image becomes available in the docker resistry on the docker hub. Update by pulling the new image, stopping the old and starting with the new, then removing the old one else the old images will start to pile up`

Reads to me as if one should pull a new image rather than update internally. So where do you read that one should update the NC version in another way?

And more important. If the user reads that information at the top of the docs page, why should one read further?

BstAA, 2023-01-30:
> Reads to me as if one should pull a new image rather than update internally. So where do you read that one should update the NC version in another way?

See [ZendaiOwl's](https://github.com/ZendaiOwl) comment on my question [two up](https://github.com/nextcloud/nextcloudpi/issues/1724#issuecomment-1407715174)

> 
> And more important. If the user reads that information at the top of the docs page, why should one read further?

As pulling the new Docker image just updates NCP _not_ NC! And _this_ should be noted in simple words.

backupandforkacc, 2023-01-31:
Can I update my NC without making a backup? It always wants to make a backup, but I do my backups with ZFS snaphots. I cannot make an update as NCP want to do it. Regular methods of NC to update (web interface, calling  "php /var/www/nextcloud/updater/updater.phar" do not work.

ZendaiOwl, 2023-01-31:
Correct :tada: 

And yes, the docs really need an overhaul .. 

Many of them are out-of-date at this point and have information not really relevant anymore such as the `pi` user for one, & how to update the Docker image needs to be reworked is another, due to lack of time and a lot of stuff to do for everyone they haven't been updated

ZendaiOwl, 2023-01-31:
@backupandforkacc 

Can you try using the CLI instead and see how that works if you have access to the terminal? 

`sudo ncp-config`

Are your backups stored elsewhere so they're safe in case something goes wrong? This is important so they aren't deleted by something going wrong or by accident

backupandforkacc, 2023-01-31:
@ZendaiOwl 
Backups/Snapshots are synced on an external drive that I can pull.

I am running a docker container, but `ncp-config`  wants to backup files before upgrading. Can I somehow skip that step? 
Is there an alternative to ncp-config? From what I know, the web interface does not work for NCP instances in general.

ZendaiOwl, 2023-01-31:
@backupandforkacc 

You are running Nextcloud's internal updater, NCP doesn't use that one for updates, if you go through with that one you will break your instance

The update steps that NCP does when updating NC are in this script 

https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp-update-nc

ZendaiOwl, 2023-01-31:
Can't you let it do the backup and then remove the backup afterwards tho?

backupandforkacc, 2023-01-31:
That means waiting 6 hours to update to 24.0.9 and then 6 hours to update to 25. It works, but its not very useful

ZendaiOwl, 2023-01-31:
I see, well I guess the alternative at the moment is to do the steps outlined in the script manually yourself

theCalcaholic, 2023-01-31:
> That means waiting 6 hours to update to 24.0.9 and then 6 hours to update to 25. It works, but its not very useful
> 

I have two remarks on that:

1. You can skip 24.0.9 without issues. Just don't skip any major versions (the updater won't let you do that anyways)

2. It's not supposed to take that long, unless you have a huge amount of NC app data (not user files) in your instance. Do you know which step takes so long? It's likely a bug.

closed 2023-02-07 by theCalcaholic