nextcloud/nextcloudpi #1728: 503 Service Unavailable when deleting files/folders with Nextcloud desktop client

OfficialMuffin, 2023-02-03:
When deleting files/folders, the Nextcloud desktop client returns a `503 Service Unavailable` error. 
Upon inspecting the logs in `/var/log/daemon.log` I see an entry from MariaDB:

`Feb  3 13:59:27 next mariadbd[1848]: 2023-02-03 13:59:27 4017 [Warning] Aborted connection 4017 to db: 'nextcloud' user: 'ncadmin' host: 'localhost' (Got an error reading communication packets)
`

I dont recall a user `ncadmin` on my server. So I inspected `/etc/passwd` and `ncadmin` does not exist. I then had a look at MariaDB and showed grants:

`show grants for ncadmin@localhost;`

the output:

```
+----------------------------------------------------------------------------------------------------------------+
| Grants for ncadmin@localhost                                                                                   |
+----------------------------------------------------------------------------------------------------------------+
| GRANT USAGE ON *.* TO `ncadmin`@`localhost` IDENTIFIED BY PASSWORD 'REDACTED' |
| GRANT ALL PRIVILEGES ON `nextcloud`.* TO `ncadmin`@`localhost`                                                 |
+----------------------------------------------------------------------------------------------------------------+
2 rows in set (0.000 sec)

```

I tried looking at the current users of the nextcloud instance in MariaDB:

1. `USE nextcloud;`
2. `SELECT * FROM oc_users;`

Output: 

```
+------+-------------+-----------------------------------------------------------------------------------------------------+-----------+
| uid  | displayname | password                                                                                            | uid_lower |
+------+-------------+-----------------------------------------------------------------------------------------------------+-----------+
| ncp  | NULL        | REDACTED                                                                                            | ncp       |
| nick | NULL        | REDACTED                                                                                            | nick      |
+------+-------------+-----------------------------------------------------------------------------------------------------+-----------+
2 rows in set (0.001 sec)

```

ZendaiOwl, 2023-02-03:
`ncadmin` is the database admin

https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/CONFIG/nc-init.sh#L11

OfficialMuffin, 2023-02-03:
Ah, my mistake @ZendaiOwl.

Schniefel, 2023-02-03:
I have had the exact same problem. Nextcloud was completely unusable.
I think it might be related to this: https://github.com/nextcloud/nextcloudpi/issues/1608

I fixed it by reinstalling NCP and deleting memcache locking from config.php

OfficialMuffin, 2023-02-03:
Thanks. I'll use this if there isn't any other fix without reinstallation.

Schniefel, 2023-02-03:
Additionally I found out that the oc_files_lock-table was full with locked files so it seemed as if the file locks won't be released correctly. Buf if I removed all entries from the table I still wasn't able to delete any file, so I am not really sure if it is related. 

Further more, on my erroneous first install I installed imagick (I have no idea if this could be related to this problem), my (working) reinstallation was without imagick. Those two things (imagick + config.php) where the only two differences between both installations. I hope this helps investigating the problem further.

OfficialMuffin, 2023-02-05:
Could I possibly use `nc-init` to restore NCP to a clean installation? Rather than imaging the drive again?

Schniefel, 2023-02-06:
Try it out. It won't take long. I think that I tried it, but not 100% sure...

OfficialMuffin, 2023-02-07:
Didn't work for me. Apparently it showed an error saying Nextcloud was not installed.
I'm closing this now as I'm wiping disk and reinstalling NCPi.

Schniefel, 2023-02-08:
Why do you close it? I think it is still a bug.

OfficialMuffin, 2023-02-08:
> Why do you close it? I think it is still a bug. 

Did you update to the latest NCPi version `1.51.0`? I'm thinking that this upgrade caused the issue. My server worked fine before the update.

Schniefel, 2023-02-08:
No, i did not upgrade. I had a completely new installation of Raspberry Pi OS and Nextcloudpi (Curl installer)

OfficialMuffin, 2023-02-08:
Do you have Nexcloud android installed? If you attempt to delete something, does it return an error saying that the `server is in maintenance mode`?

Schniefel, 2023-02-09:
Yes, exactly the same.

OfficialMuffin, 2023-02-09:
I'll reopen issue

esseivan, 2023-02-20:
I have the same issue. I had a brand new 1.51.0 NextCloudPi OS installed on my raspberry pi 4 2GB. Everything worker fine until I tried uploading multiples files from my phone (Android). Almost all the upload failed saying something like connection error, and then this issue appeared and I couldn't rename nor delete any file, even the ones that were already on the cloud.
I'm using an external HDD to store the data. I did a backup of the clean install and restoring that backup once the issue appeared reverted it.

svenb1234, 2023-02-25:
For 503 issues this is something to try:

```
sudo bash -c 'source /usr/local/etc/library.sh && clear_opcache' 
sudo service php8.1-fpm restart
```

feutl, 2023-03-09:
I need to jump onto this as well, I do see the error popping up but in combination with the contacts up. As soon as I open the contacts web app I see the error in the logs and the contacts stay empty.
I am using NCP Docker with the latest build available on hub.docker.com also NC is up to date 25.0.3

Can someone help me out on this?! or can I help to figure this out.

Because I started the whole endeavor in the contacts github repo - I link it here
https://github.com/nextcloud/contacts/issues/3253#

feutl, 2023-03-28:
Had the same error message as @OfficialMuffin and fixed it by extending the valus for `max_allowed_packet` in the mysql config.
https://github.com/nextcloud/contacts/issues/3253#issuecomment-1486719612

OfficialMuffin, 2023-04-09:
A backup of all nextcloud data and a reinstallation of NextcloudPi fixes the issue for me. I'm closing this.

closed 2023-04-09 by OfficialMuffin