nextcloud/nextcloudpi #1729: ncp-web backups.php crashes

tahoar, 2023-02-06:
NextCloud v25.0.3
NextCloudPi v1.51.0
Debian 11.6
PHP 8.1-fpm

Problem: When one or more backup files with proper name syntax (nextcloud-bkp_yyyymmdd_dddddddddd.tar.gz) are present in the configured nc-backup.cfg DESTDIR directory, Nextcloudpi crashes.

Result: 
Initial System Info screen collects information (spinning mouse icon). It then shows status information but immediately goes blank with the message: "Something went wrong. Try refreshing the page"
I can navigate left menu to other apps, but returning to System Info screen is always blank. After hard browser refresh (ctrl-shift-R with Firefox), the screen collects information and then goes blank with message again.

Error message in error.log:
[Sun Feb 05 20:03:17.301164 2023] [proxy_fcgi:error] [pid 4438] [client 192.168.50.1:52012] AH01071: Got error 'PHP message: PHP Fatal error:  Uncaught TypeError: array_key_exists(): Argument #2 ($array) must be of type array, null given in /var/www/ncp-web/backups.php:76\nStack trace:\n#0 /var/www/ncp-web/ncp-launcher.php(146): include()\n#1 {main}\n  thrown in /var/www/ncp-web/backups.php on line 76'

Note: removing the backup files from the configured directory, followed by hard browser refresh clears the error and displays system information.

tahoar, 2023-03-21:
Ping... Any suggestions? The problem continues and the solution to remove the backup file from the configured folder still works. However, that's not the same as having it work properly.

tahoar, 2023-04-12:
Ping... This bug continues in NextcloudPi version v1.51.1 update.

Dani-Pi, 2023-06-09:
I see the same problem in  v1.52.0

theCalcaholic, 2023-06-14:
Thanks for reporting! I'm currently very short on time but will try to look into this issue on the weekend.

Dani-Pi, 2023-07-19:
The bug continues in NCP v1.52.1

Dani-Pi, 2023-07-22:
The bug continues in NCP v1.52.2

Halloran, 2023-09-13:
in NCP v1.51.4 a similar issue occurs when the system has a corrupted `backup-info-cache.cfg` file. I was able to resolve this by renaming the `backup-info-cache.cfg` file, thought I don't know how the file got corrupted in the first place.

Looking at the code, it seems like adding a check to ensure that `backup-info-cache.cfg` contains valid json as is done for nc-backup-auto.cfg, might resolve the issue.

Thanks for your hard work on a great project!

Dani-Pi, 2023-09-22:
> in NCP v1.51.4 a similar issue occurs when the system has a corrupted `backup-info-cache.cfg` file. I was able to resolve this by renaming the `backup-info-cache.cfg` file, thought I don't know how [...]

Solved the problem for me. Thank you, @Halloran !!