nextcloud/nextcloudpi #1730: Debian 12 Bookworm Support

sunjam, 2023-02-11:
For internal testing and notes. **Do not expect support!!!!!!** Please do report testing issues here.

[Debian 12 Bookworm](https://www.debian.org/releases/bookworm/) - [release schedule](https://wiki.debian.org/DebianBookworm)
Enters soft freeze 2/12/2023, but no formal release date has been set.
Assume mid-2023 for formal stable.

apblack96, 2023-06-13:
Now that Debian 12 is officially out is this support confirmed? From what I can tell it should be but want official confirmation before I run the upgrade.

theCalcaholic, 2023-06-14:
@theoriginalguy Debian 12 is not supported by NCP yet. If you are upgrading manually, you are doing so on your own account (and things are likely to break).

I doubt that we will upgrade to Debian 12 until we have progressed on the ongoing architectural changes (i.e. containerizing most of the included services), because that will make system upgrades a lot less risky.

tom-i, 2023-08-06:
I've tried it today on clean Debian 12 - bookworm (Armbian 23.5.1) and after commenting section about checking version install script crashed after dependancy issue due tu php-8.1 version.
So it looks that I have to w8 till you support bookworm.. :-(

tom-i, 2023-08-06:
Finally, I’ve managed to do it myself :-)
Few problems occurs, but it was ok I’ve solve them without problem, mainly about different name of php libs, exfat utils and so on.
Some problem with installing and starting up dnsmasq service.
One issue after installing with activating, but this thread helped: [NextCloudPlus not yet initialized, trying again in a few seconds](https://help.nextcloud.com/t/nextcloudplus-not-yet-initialized-trying-again-in-a-few-seconds/31019)
So actually I’m on NCP 27.0.1 on Debian 12 - Bookworm.

SHA-NDY, 2023-08-28:
> Finally, I’ve managed to do it myself :-) Few problems occurs, but it was ok I’ve solve them without problem, mainly about different name of php libs, exfat utils and so on. Some problem with [...]

Would you share your modified script?

SHA-NDY, 2023-08-28:
Meanwhile I upgraded from Bullseye to Bookworm. Everything seems to be fine. NC instance is reachable and all NCP settings are ok.

[image]

morgan-germain, 2023-10-15:
Bookworm is now the default release with raspberry pi os. But NextcloudPi installer script fails to launch because of an "Error: distro not supported" message. I will try to bypass this check

theCalcaholic, 2023-12-27:
Update: If all goes well with the next releases, I see ourselves supporting Debian Bookworm in 2024-03

FloKoll, 2024-01-04:
Thx for your efforts! I appreciate your work, but It feels not good to be without security updates for 3 months.

theCalcaholic, 2024-01-07:
@FloKoll Don't worry! Debian bullseye has security support until 2024-05 (including May)

FloKoll, 2024-01-09:
@theCalcaholic Thank you for replying to my post. Yes, but - as far as I understand it correctly - support for Nextcloud 25 expired in October 2023. That means there are no more security fixes coming. Probably tens of thousands or more Raspberry Pis can't upgrade to Netxcloud 26 or higher because of the PHP version under Bullseye, which - it seems to me - makes the whole thing a veritable security risk. But maybe my worries are unfounded?

theCalcaholic, 2024-01-09:
@FloKoll NextcloudPi supports Nextcloud 27, though

FloKoll, 2024-01-09:
I'm no expert, but the problem seems to be that all machines running  Bullseye with php 7.4 are not offered the update to NC 26 at all. This should affect a lot of people who, like me, have their small server running at home. I could of course update to Bookworm, but I'm afraid that there will be problems with Nextcloudpi.

Best regards

theCalcaholic, 2024-01-09:
@FloKoll NextcloudPi uses PHP 8.1 currently, not 7.4. The PHP version will automatically be updated when updating nextcloud to v24 or later

FloKoll, 2024-01-09:
Interesting! I have always worked with "sudo -u www-data php occ upgrade" and not with ncp-config. That will probably be the reason why PHP has not been pulled from the Sury repos. I'll try it with your tool. Thanks for your patience and advice.

FloKoll, 2024-01-12:
Running nc-update-nextcloud
Upgrade cannot skip major versions. Please upgrade one major version at a time
Done. Press any key...

I read https://help.nextcloud.com/t/staying-up-to-date/126259 and tried it with the code 0. But it didn't work. At the moment my nc-version ist NextCloudPi 25.0.13 (Debian Bullseye)

FloKoll, 2024-01-12:
Is there another code, I could choose?

theCalcaholic, 2024-01-12:
@FloKoll The updater doesn't currently support selecting the next major version automatically. Try putting `26.0.10`, then `27.1.5` (or 0, which should update to 27.1.3 atm)

FloKoll, 2024-01-13:
`running nc-update-nextcloud
Current   Nextcloud version 25.0.13.2
Available Nextcloud version 26.0.10
Download Nextcloud 26.0.10...
Back up current instance...
check free space...
Maintenance mode enabled
backup database...
backup files...
backup /var/www/nextcloud-bkp_20240113_1705107484.tar.gz generated
Maintenance mode disabled
Storing backup at '/var/www/nextcloud-bkp_20240113_1705107484-25.0.13.2.tar.gz'...
Install Nextcloud 26.0.10...
Fix permissions...
Upgrade...
This version of Nextcloud requires at least PHP 8.0<br/>You are currently running 7.4.33. Please update your PHP version.Abort
Clean up...
Rolling back to backup /var/www/nextcloud-bkp_20240113_1705107484-25.0.13.2.tar.gz...
check free space...
extracting backup file /var/www/nextcloud-bkp_20240113_1705107484-25.0.13.2.tar.gz...
restore files...
restore database...
No datadir found in backup
Maintenance mode disabled
Starting scan for user 1 out of 4 [...]`

@theCalcaholic It seems to me that the updater does not automatically fetch PHP 8.1 from the Sury repository. I understood you above to mean that this is what you intended.

theCalcaholic, 2024-01-13:
@FloKoll huh, weird... Could you run `DBG=X ncp-update 26.0.10` be post the output here?

FloKoll, 2024-01-14:
Sure!

sudo DBG=X ncp-update 26.0.10
/usr/local/bin/ncp-update: line 5: set: -X: invalid option
set: usage: set [-abefhkmnptuvxBCHP] [-o option-name] [--] [arg ...]
INFO: updating to development branch '26.0.10'
Downloading updates
warning: Could not find remote branch 26.0.10 to clone.
fatal: Remote branch 26.0.10 not found in upstream origin
No internet connectivity

FloKoll, 2024-01-14:
I am amazed that it says there is no internet connection available. The server is online and I can access it. App updates and ncp-updates also work without any problems. Perhaps useful additional information could be that in my case the DynDNS account is set up on the server itself and not on the router, as the combination of STRATO provider and AVM Fritzbox made this necessary.

theCalcaholic, 2024-01-14:
@FloKoll Sorry, that was my autocorrect messing with me 🙈
The x is supposed to be lower case, so `DBG=x ncp-update 26.0.10`

FloKoll, 2024-01-14:
@theCalcaholic 

sudo DBG=x ncp-update-nc 26.0.10
+ VER=26.0.10
+ BIN=ncp-update-nc
+ source /usr/local/etc/library.sh
++ export CFGDIR=/usr/local/etc/ncp-config.d
++ CFGDIR=/usr/local/etc/ncp-config.d
++ export BINDIR=/usr/local/bin/ncp
++ BINDIR=/usr/local/bin/ncp
++ export NCDIR=/var/www/nextcloud
++ NCDIR=/var/www/nextcloud
++ export ncc=/usr/local/bin/ncc
++ ncc=/usr/local/bin/ncc
++ export NCPCFG=etc/ncp.cfg
++ NCPCFG=etc/ncp.cfg
+++ dpkg --print-architecture
++ export ARCH=arm64
++ ARCH=arm64
++ [[ arm64 =~ ^(armhf|arm)$ ]]
++ [[ arm64 == \a\r\m\6\4 ]]
++ ARCH=aarch64
++ [[ aarch64 == \a\m\d\6\4 ]]
++ export SYSTEMD_PAGER=
++ SYSTEMD_PAGER=
++ [[ -f etc/ncp.cfg ]]
++ export NCPCFG=/usr/local/etc/ncp.cfg
++ NCPCFG=/usr/local/etc/ncp.cfg
++ [[ -f /usr/local/etc/ncp.cfg ]]
+++ ps -p 1 --no-headers -o %c
++ [[ systemd == \s\y\s\t\e\m\d ]]
++ [[ -d /run/systemd/system ]]
++ [[ -d /run/systemd/system ]]
++ INIT_SYSTEM=systemd
++ export INIT_SYSTEM
++ command -v jq
+++ jq -r .nextcloud_version
++ NCLATESTVER=27.1.3
+++ jq -r .php_version
++ PHPVER=7.4
+++ jq -r .release
++ RELEASE=bullseye
++ grep -Eh '^deb ' /etc/apt/sources.list
++ grep bullseye-security
++ RELEASE=bullseye-security
++ command -v ncc
+++ ncc status
+++ grep version:
+++ awk '{ print $3 }'
++ NCVER=25.0.13.2
+ [[ 26.0.10 == '' ]]
+ [[ -f /.docker-image ]]
+ BASEDIR=/var/www
+ cd /var/www
++ get_nc_config_value datadirectory
++ sudo -u www-data php -r 'include("/var/www/nextcloud/config/config.php"); echo($CONFIG["datadirectory"]);'
+ DATADIR=/media/Daten-Festplatte/ncdata/data
+ ncc status
+ [[ -d /var/www/nextcloud-old ]]
+ [[ -d /var/www/nextcloud ]]
+ [[ -d /media/Daten-Festplatte/ncdata/data ]]
+ [[ 0 -eq 0 ]]
+ SUDO='sudo -u www-data'
++ sudo -u www-data php /var/www/nextcloud/occ status
++ awk '{ print $3 }'
++ grep version:
+ CURRENT=25.0.13.2
++ cut -d. -f1
+ MAJOR_CUR=25
++ cut -d. -f1
[... 156 lines cut ...]
Starting scan for user 4 out of 4 [censored]
+---------+-------+--------------+
| Folders | Files | Elapsed time |
+---------+-------+--------------+
| 15160   | 96812 | 00:02:30     |
+---------+-------+--------------+
Installing template 'php/opcache.ini.sh'...
System config value tempdirectory set to string /media/Daten-Festplatte/ncdata/data/tmp
System config value logfile set to string /media/Daten-Festplatte/ncdata/data/nextcloud.log
System config value trusted_domains => 11 set to string 2a01:c23:c4d6:b600:fba5:a881:b460:a634
System config value trusted_domains => 1 set to string 192.168.178.20
System config value trusted_domains => 14 set to string nextcloudpi
Scanning AppData for files

+---------+-------+--------------+
| Folders | Files | Elapsed time |
+---------+-------+--------------+
| 93836   | 21383 | 00:07:50     |
+---------+-------+--------------+
Cleanup...
++ [[ /media/Daten-Festplatte/ncdata/data == \/\v\a\r\/\w\w\w\/\n\e\x\t\c\l\o\u\d\/\d\a\t\a ]]
++ rm /var/www/nextcloud-bkp_20240114_1705240186-25.0.13.2.tar.gz
++ echo 'Rollback successful. Nothing was updated'
Rollback successful. Nothing was updated
++ exit 1

pkra, 2024-01-23:
@theCalcaholic I seem to have the same issue as @FloKoll (no PHP 8).  I'm on NC v25.0.9.2 and debugging output looks fairly identical to the comment just above this one.  Happy to provide additional information if that's useful.

christoph-bittmann, 2024-03-06:
@pkra, @FloKoll 
Have you tried to install or switch to php8.1 manuell? Maybe it's already installed on your system and you only must switch to php8.1 as default (see https://think.unblog.ch/en/php-versions-switching-on-debian-with-apache2/).
Maybe update packaged before: `sudo apt update && sudo apt upgrade -y `

FloKoll, 2024-03-12:
@christoph-bittmann @theCalcaholic 

Thank you very much for your answer. I had feared that my request had come to a dead end. PHP 8.1 isn't installed and not in the bullseye repositories. I fear that things might break in the future if I use other sources.

`apt list --installed | grep php

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

php-apcu-bc/oldstable,now 1.0.5-5 arm64 [installed,automatic]
php-apcu/oldstable,now 5.1.19+4.0.11-3 arm64 [installed]
php-common/oldstable,oldstable,now 2:76 all [installed,automatic]
php-igbinary/oldstable,now 3.2.1+2.0.8-2 arm64 [installed,automatic]
php-imagick/oldstable,now 3.4.4+php8.0+3.4.4-2+deb11u2 arm64 [installed]
php-redis/oldstable,now 5.3.2+4.3.0-2+deb11u1 arm64 [installed]
php7.4-bcmath/oldstable,oldstable-security,now 7.4.33-1+deb11u4 arm64 [installed]
php7.4-bz2/oldstable,oldstable-security,now 7.4.33-1+deb11u4 arm64 [installed]
php7.4-cli/oldstable,oldstable-security,now 7.4.33-1+deb11u4 arm64 [installed]
php7.4-common/oldstable,oldstable-security,now 7.4.33-1+deb11u4 arm64 [installed]
php7.4-curl/oldstable,oldstable-security,now 7.4.33-1+deb11u4 arm64 [installed]
php7.4-fpm/oldstable,oldstable-security,now 7.4.33-1+deb11u4 arm64 [installed]
php7.4-gd/oldstable,oldstable-security,now 7.4.33-1+deb11u4 arm64 [installed]
php7.4-gmp/oldstable,oldstable-security,now 7.4.33-1+deb11u4 arm64 [installed]
php7.4-intl/oldstable,oldstable-security,now 7.4.33-1+deb11u4 arm64 [installed]
php7.4-json/oldstable,oldstable-security,now 7.4.33-1+deb11u4 arm64 [installed]
php7.4-ldap/oldstable,oldstable-security,now 7.4.33-1+deb11u4 arm64 [installed]
php7.4-mbstring/oldstable,oldstable-security,now 7.4.33-1+deb11u4 arm64 [installed]
php7.4-mysql/oldstable,oldstable-security,now 7.4.33-1+deb11u4 arm64 [installed]
php7.4-opcache/oldstable,oldstable-security,now 7.4.33-1+deb11u4 arm64 [installed]
php7.4-readline/oldstable,oldstable-security,now 7.4.33-1+deb11u4 arm64 [installed,automatic]
php7.4-xml/oldstable,oldstable-security,now 7.4.33-1+deb11u4 arm64 [installed]
php7.4-zip/oldstable,oldstable-security,now 7.4.33-1+deb11u4 arm64 [installed]
php7.4/oldstable,oldstable,oldstable-security,oldstable-security,now 7.4.33-1+deb11u4 all [installed]
`

pkra, 2024-03-13:
@christoph-bittmann thanks for following up. I have a similar apt list output as @FloKoll, i.e., only php 7.4. 

A while back (when nextcloud first required it), I had tried to manually upgrade to php 8.1 and my inexperience had messed up my system. If there's a recommended guide, I'd be willing to try again.

kevinestel, 2024-03-16:
> Update: If all goes well with the next releases, I see ourselves supporting Debian Bookworm in 2024-03

I would be very interested too. Can you tell if it's coming soon ?

Thanks for what you're doing !

blueskyfeeling, 2024-03-27:
Can anyone tell us, when nextcloudpi will be available on bookworm approximately?

theCalcaholic, 2024-03-28:
> Can anyone tell us, when nextcloudpi will be available on bookworm approximately?

Expect preliminary bookworm support in 1-2 weeks

Artorias38, 2024-04-18:
> > Can anyone tell us, when nextcloudpi will be available on bookworm approximately?
> 
> Expect preliminary bookworm support in 1-2 weeks

Anything new? I am eager to use my new pi 5 :D

sunjam, 2024-04-20:
https://github.com/nextcloud/nextcloudpi/pull/1887

theCalcaholic, 2024-04-20:
> > > Can anyone tell us, when nextcloudpi will be available on bookworm approximately?
> > 
> > 
> > Expect preliminary bookworm support in 1-2 weeks
> 
> Anything new? I am eager to use my new pi 5 :D

I'm still working on it. The armbian builds are giving a bit trouble still, but I hope I can release new images for testing soon

nbendall34, 2024-04-20:
Thanks @theCalcaholic, looking forward to it, very eager to use my new RPI5 8G :)

blueskyfeeling, 2024-04-21:
I can't wait any longer :-) and will be very grateful if it works. A big compliment to @theCalcaholic

theCalcaholic, 2024-04-22:
There's now a test release based on Debian 12: https://github.com/nextcloud/nextcloudpi/releases/tag/v1.54.0-rc3

If you are able to test any of the images, please consider giving feedback :)

blueskyfeeling, 2024-04-23:
Great. In the meantime, I was able to test the version for a Raspi5. So far no problems. I'll continue testing extensively over the weekend.

szarpaj, 2024-04-27:
> There's now a test release based on Debian 12: https://github.com/nextcloud/nextcloudpi/releases/tag/v1.54.0-rc3
> 
> If you are able to test any of the images, please consider giving feedback :)

I have one installation that was upgraded to Debian 12 while nextcloud on 23.x and it stuck there. No go to 24.x. With this -rc3 I've upgraded to 26.x (due to timeline I had to stop there for few days). But it seems that the bigest issue isn't there anymore. Great job!

blueskyfeeling, 2024-04-28:
Now I tested the NextcloudPi_RaspberryPi5_v1.54.0-rc3.zip image and it seems, that it works very well. I also copied the entire Nextcloud from a Raspi4 and everything worked there too. If nothing speaks against it, I would then make the real move soon, is that possible?

Maybe a small (ridiculous) question: Is there a reason,
[image]
 why my Raspi 4 appears green on putty and white on Raspi5 (see attached picture)? Is there anything else that needs to be corrected here? I don't know that much about it excatly.

theCalcaholic, 2024-05-02:
Support for Debian 12 is now officially release with v1.54.0 and will be rolled out to users successively via staged auto updates (10% of users as of now).

Manual updates via `ncp-update` are of course always possible

closed 2024-05-02 by theCalcaholic

lxto, 2024-05-02:
I tried to upgrade to ncp 1.54.0 which worked normally but when I tried the dist-upgrade to Bookworm
I got the massage that php 7.4 still being installed although php 8.3.3 is installed and activated on my system.
How can I safely remove php 7.4 files?

> update to v1.54.0 through 'ncp-config' or type 'sudo ncp-update'
Last login: Wed May  1 23:49:58 2024 from 192.168.1.80
pi@nextcloudpi:~ $ sudo ncp-update
Downloading updates
Performing updates
Update root login prevention method...
done.
Fixing trusted proxies list...
done.
Updating PHP package signing key...
OK:1 http://deb.debian.org/debian bullseye InRelease
OK:2 http://deb.debian.org/debian-security bullseye-security InRelease               
OK:3 http://archive.raspberrypi.org/debian bullseye InRelease                        
OK:4 http://deb.debian.org/debian bullseye-updates InRelease                         
OK:5 https://packages.sury.org/php bullseye InRelease                                
Paketlisten werden gelesen… Fertig                           
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Die folgenden Pakete wurden automatisch installiert und werden nicht mehr benötigt:
  php8.1-igbinary php8.1-imagick php8.1-phpdbg php8.1-redis php8.1-smbclient php8.2-igbinary php8.2-imagick php8.2-redis
Verwenden Sie »sudo apt autoremove«, um sie zu entfernen.
Die folgenden NEUEN Pakete werden installiert:
  gnupg2
0 aktualisiert, 1 neu installiert, 0 zu entfernen und 163 nicht aktualisiert.
Es müssen 434 kB an Archiven heruntergeladen werden.
Nach dieser Operation werden 453 kB Plattenplatz zusätzlich benutzt.
Holen:1 http://deb.debian.org/debian bullseye/main arm64 gnupg2 all 2.2.27-2+deb11u2 [434 kB]
Es wurden 434 kB in 0 s geholt (958 kB/s).
Vormals nicht ausgewähltes Paket gnupg2 wird gewählt.
(Lese Datenbank ... 50714 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../gnupg2_2.2.27-2+deb11u2_all.deb ...
Entpacken von gnupg2 (2.2.27-2+deb11u2) ...
gnupg2 (2.2.27-2+deb11u2) wird eingerichtet ...
Trigger für man-db (2.9.4-2) werden verarbeitet ...
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
Executing: /tmp/apt-key-gpghome.4buN94m7ml/gpg.1.sh --fetch-keys https://packages.sury.org/php/apt.gpg
gpg: fordere Schlüssel von https://packages.sury.org/php/apt.gpg an
gpg: Schlüssel B188E2B695BD4743: "DEB.SURY.ORG Automatic Signing Key <deb@sury.org>" nicht geändert
gpg: Anzahl insgesamt bearbeiteter Schlüssel: 1
gpg:                             unverändert: 1
done.
Installing dependencies...
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Die folgenden Pakete wurden automatisch installiert und werden nicht mehr benötigt:
  php8.1-igbinary php8.1-imagick php8.1-phpdbg php8.1-redis php8.1-smbclient php8.2-igbinary php8.2-imagick php8.2-redis
Verwenden Sie »sudo apt autoremove«, um sie zu entfernen.
Die folgenden zusätzlichen Pakete werden installiert:
  libutempter0
Die folgenden NEUEN Pakete werden installiert:
  libutempter0 tmux
0 aktualisiert, 2 neu installiert, 0 zu entfernen und 163 nicht aktualisiert.
Es müssen 347 kB an Archiven heruntergeladen werden.
Nach dieser Operation werden 881 kB Plattenplatz zusätzlich benutzt.
Holen:1 http://deb.debian.org/debian bullseye/main arm64 libutempter0 arm64 1.2.1-2 [8.972 B]
Holen:2 http://deb.debian.org/debian bullseye/main arm64 tmux arm64 3.1c-1+deb11u1 [338 kB]
Es wurden 347 kB in 0 s geholt (862 kB/s).
Vormals nicht ausgewähltes Paket libutempter0:arm64 wird gewählt.
(Lese Datenbank ... 50721 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../libutempter0_1.2.1-2_arm64.deb ...
Entpacken von libutempter0:arm64 (1.2.1-2) ...
Vormals nicht ausgewähltes Paket tmux wird gewählt.
Vorbereitung zum Entpacken von .../tmux_3.1c-1+deb11u1_arm64.deb ...
Entpacken von tmux (3.1c-1+deb11u1) ...
[... 33 lines cut ...]
BUG_REPORT_URL="https://bugs.debian.org/"
pi@nextcloudpi:~ $ sudo reboot
Connection to 192.168.1.70 closed by remote host.
Connection to 192.168.1.70 closed.

>sudo ncp-dist-upgrade: 

>This is a dangerous process that is only guaranteed to work properly if you
have not made manual changes in the system. Backup the SD card first and
proceed at your own risk.

>Note that this is not a requirement for NCP to continue working properly.
The current distribution will keep receiving updates for some time.

>Do you want to continue? [y/N]
>y
>You still have PHP version 7.4 installed. Please update to the latest supported version of nextcloud (which will also update your PHP version) before proceeding with the distribution upgrade.
Exiting.
root@nextcloudpi:/home/pi# 
root@nextcloudpi:/home/pi# php -v
PHP 8.3.3-1+0~20240216.17+debian11~1.gbp87e37b (cli) (built: Feb 16 2024 10:33:07) (NTS)
Copyright (c) The PHP Group
Zend Engine v4.3.3, Copyright (c) Zend Technologies
    with Zend OPcache v8.3.3-1+0~20240216.17+debian11~1.gbp87e37b, Copyright (c), by Zend Technologies
root@nextcloudpi:/home/pi# <

FloKoll, 2024-08-20:
@theCalcaholic 

Unfortunately my problem is still not solved, although the thread has been closed. 
It is not possible to upgrade the Debian 11 / NC25 / PHP 7.4 machine:

`
ncp-dist-upgrade
>>> ATTENTION <<<
This is a dangerous process that is only guaranteed to work properly if you
have not made manual changes in the system. Backup the SD card first and
proceed at your own risk.

Note that this is not a requirement for NCP to continue working properly.
The current distribution will keep receiving updates for some time.

Do you want to continue? [y/N]
y
You still have PHP version 7.4 installed. Please update to the latest supported version of nextcloud (which will also update your PHP version) before proceeding with the distribution upgrade.
`
`ncp-update-nextcloud
Current   Nextcloud version 25.0.13
Available Nextcloud version 26.0.13
INFO: You have requested an update to '26.0.10', but a direct update to '26.0.10' cannot be performed, so the latest available version that can be updated to has been selected automatically.
Download Nextcloud 26.0.13...
Back up current instance...
check free space...
Maintenance mode enabled
backup database...
backup files...
backup /var/www/nextcloud-bkp_20240820_1724156651.tar.gz generated
Maintenance mode disabled
Storing backup at '/var/www/nextcloud-bkp_20240820_1724156651-25.0.13.tar.gz'...
Install Nextcloud 26.0.13...
Fix permissions...
Upgrade...
This version of Nextcloud requires at least PHP 8.0<br/>You are currently running 7.4.33. Please update your PHP version.Abort
Clean up...
Rolling back
`
I now need a clear statement whether the problem is still being fixed or whether a new installation is unavoidable. 

Thank you very much for your efforts!

sunjam, 2024-08-20:
@FloKoll  please post this support ticket to help.nextcloud.com with tag `ncp` 
Thank you

theCalcaholic, 2024-08-21:
@FloKoll The next update for NCP will (internally) decouple the php upgrade and NC update, so it should be easier to give you instructions for fixing your situation aftwr said update.

Please ping me once again when v1.55.0 is out

FloKoll, 2024-08-22:
@sunjam done --> https://help.nextcloud.com/t/upgrade-of-nextcloud-25-server-not-possible-debian-11-php-7-4/201435
@theCalcaholic --> I will, thank you!