nextcloud/nextcloudpi #1731: Service Unavailable The server is temporarily unable to service your request due to maintenance downtime or capacity problems. Please try again later.

sebasop320, 2023-02-13:
**System information**

<details>
<summary>NextcloudPi diagnostics</summary>

```

NextcloudPi version  v1.51.0
NextcloudPi image    NextcloudPi_01-29-23
OS                   Debian GNU/Linux 11. 5.15.84-v8+ (aarch64)
automount            yes
USB devices          sda 
datadir              /media/USBdrive/ncdata/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      6.0G/29G
rootfs usage         6.0G/29G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    25.0.3.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              closed
Port 443             closed
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               28min

```
</details>

when i try to acces nextcloudPi dashboard this error appears 

Service Unavailable
The server is temporarily unable to service your request due to maintenance downtime or capacity problems. Please try again later.

please help!

closed 2023-02-13 by sebasop320

Schniefel, 2023-04-16:
How did you solve this?

sebasop320, 2023-04-18:
on the admin UI on the tools section there is a place to delete the PHP cache just press it and the error should vansh

it worked for me

have a nice day :)

Schniefel, 2023-04-18:
Worked for me too, thank you