nextcloud/nextcloudpi #1732: ssh login fail ：This account is currently not available.

iT-Boyer, 2023-02-13:
**System information**
Linux vultr 5.10.0-21-amd64 #1 SMP Debian 5.10.162-1 (2023-01-21) x86_64 GNU/Linux

<--! Paste this in GitHub report -->
<details>
<summary>NextcloudPi diagnostics</summary>

```
NextcloudPi version  v1.51.0
NextcloudPi image    NextcloudPi_02-13-23
OS                   Debian GNU/Linux 11. 5.10.0-21-amd64 (x86_64)
automount            no
USB devices          none
datadir              /opt/ncdata/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      14G/24G
rootfs usage         14G/24G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    25.0.3.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              closed
Port 443             closed
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            enp1s0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               56min
```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "7": "nextcloudpi",
            "5": "nextcloudpi.local",
            "8": "nextcloudpi.lan",
            "3": "nextcloudpi.lan"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "25.0.3.2",
        "overwrite.cli.url": "https:\/\/nextcloudpi\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
[... 50 lines cut ...]
<summary>Nextcloud logs</summary>

```

{"reqId":"1LQxfpYfEo3ixwnd8Tkt","level":3,"time":"2023-02-13T15:40:02+08:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Not installed","userAgent":"--","version":"","exception":{"Exception":"Exception","Message":"Not
installed","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/base.php","line":680,"function":"checkInstalled","class":"OC","type":"::"},{"file":"/var/www/nextcloud/lib/base.php","line":1144,"function":"init","class":"OC","type":"::"},{"file":"/var/www/nextcloud/cron.php","line":43,"args":["/var/www/nextcloud/lib/base.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/base.php","Line":284,"CustomMessage":"--"}}
{"reqId":"v9QhciyH0yn6f4XNQ5Tw","level":3,"time":"2023-02-13T15:40:43+08:00","remoteAddr":"","user":"ncp","app":"PHP","method":"","url":"--","message":"exif_read_data(): File not supported at /var/www/nextcloud/lib/private/Metadata/Provider/ExifProvider.php#59","userAgent":"--","version":"25.0.3.2","exception":{"Exception":"Error","Message":"exif_read_data(): File not supported at
/var/www/nextcloud/lib/private/Metadata/Provider/ExifProvider.php#59","Code":0,"Trace":[{"function":"onError","class":"OC\\Log\\ErrorHandler","type":"::"},{"file":"/var/www/nextcloud/lib/private/Metadata/Provider/ExifProvider.php","line":59,"function":"exif_read_data"},{"file":"/var/www/nextcloud/lib/private/Metadata/MetadataManager.php","line":68,"function":"execute","class":"OC\\Metadata\\Provider\\ExifProvider","type":"->"},{"file":"/var/www/nextcloud/lib/private/Metad\
ata/FileEventListener.php","line":101,"function":"generateMetadata","class":"OC\\Metadata\\MetadataManager","type":"->"},{"file":"/var/www/nextcloud/lib/private/EventDispatcher/ServiceEventListener.php","line":87,"function":"handle","class":"OC\\Metadata\\FileEventListener","type":"->"},{"file":"/var/www/nextcloud/3rdparty/symfony/event-dispatcher/EventDispatcher.php","line":251,"function":"__invoke","class":"OC\\EventDispatcher\\ServiceEventListener","type":"->"},{"file"\
:"/var/www/nextcloud/3rdparty/symfony/event-dispatcher/EventDispatcher.php","line":73,"function":"callListeners","class":"Symfony\\Component\\EventDispatcher\\EventDispatcher","type":"->"},{"file":"/var/www/nextcloud/lib/private/EventDispatcher/EventDispatcher.php","line":88,"function":"dispatch","class":"Symfony\\Component\\EventDispatcher\\EventDispatcher","type":"->"},{"file":"/var/www/nextcloud/lib/private/EventDispatcher/EventDispatcher.php","line":100,"function":"di\
spatch","class":"OC\\EventDispatcher\\EventDispatcher","type":"->"},{"file":"/var/www/nextcloud/lib/private/Files/Node/HookConnector.php","line":118,"function":"dispatchTyped","class":"OC\\EventDispatcher\\EventDispatcher","type":"->"},{"file":"/var/www/nextcloud/lib/private/legacy/OC_Hook.php","line":106,"function":"postWrite","class":"OC\\Files\\Node\\HookConnector","type":"->"},{"file":"/var/www/nextcloud/lib/private/Files/View.php","line":641,"function":"emit","class"\
:"OC_Hook","type":"::"},{"file":"/var/www/nextcloud/lib/private/Files/View.php","line":692,"function":"emit_file_hooks_post","class":"OC\\Files\\View","type":"->"},{"file":"/var/www/nextcloud/lib/private/Files/Node/File.php","line":71,"function":"file_put_contents","class":"OC\\Files\\View","type":"->"},{"file":"/var/www/nextcloud/lib/private/legacy/OC_Util.php","line":257,"function":"putContent","class":"OC\\Files\\Node\\File","type":"->"},{"file":"/var/www/nextcloud/lib\
/private/legacy/OC_Util.php","line":216,"function":"copyr","class":"OC_Util","type":"::"},{"file":"/var/www/nextcloud/lib/private/User/Session.php","line":553,"function":"copySkeleton","class":"OC_Util","type":"::"},{"file":"/var/www/nextcloud/lib/private/User/Session.php","line":403,"function":"prepareUserLogin","class":"OC\\User\\Session","type":"->"},{"file":"/var/www/nextcloud/lib/private/User/Session.php","line":616,"function":"completeLogin","class":"OC\\User\\Sessi\
on","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/lib/private/User/Session.php","line":355,"function":"loginWithPassword","class":"OC\\User\\Session","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/lib/private/Setup.php","line":433,"function":"login","class":"OC\\User\\Session","type":"->","args":["*** sensitive parameters replaced
***"]},{"file":"/var/www/nextcloud/core/Command/Maintenance/Install.php","line":105,"function":"install","class":"OC\\Setup","type":"->","args":["*** sensitive parameters replaced
***"]},{"file":"/var/www/nextcloud/3rdparty/symfony/console/Command/Command.php","line":255,"function":"execute","class":"OC\\Core\\Command\\Maintenance\\Install","type":"->"},{"file":"/var/www/nextcloud/3rdparty/symfony/console/Application.php","line":1009,"function":"run","class":"Symfony\\Component\\Console\\Command\\Command","type":"->"},{"file":"/var/www/nextcloud/3rdparty/symfony/console/Application.php","line":273,"function":"doRunCommand","class":"Symfony\\Compone\
nt\\Console\\Application","type":"->"},{"file":"/var/www/nextcloud/3rdparty/symfony/console/Application.php","line":149,"function":"doRun","class":"Symfony\\Component\\Console\\Application","type":"->"},{"file":"/var/www/nextcloud/lib/private/Console/Application.php","line":213,"function":"run","class":"Symfony\\Component\\Console\\Application","type":"->"},{"file":"/var/www/nextcloud/console.php","line":100,"function":"run","class":"OC\\Console\\Application","type":"->"}\
,{"file":"/var/www/nextcloud/occ","line":11,"args":["/var/www/nextcloud/console.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/private/Log/ErrorHandler.php","Line":92,"CustomMessage":"--"}}
{"reqId":"jGVopDSO9CTdzfYBnzib","level":2,"time":"2023-02-13T15:41:15+08:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"25.0.3.2","data":[]}
{"reqId":"jGVopDSO9CTdzfYBnzib","level":2,"time":"2023-02-13T15:41:15+08:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"25.0.3.2","data":[]}
[... cut ...] ser":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"25.0.3.2","data":[]}
{"reqId":"mhTCcIxB5jyaIEeIKemJ","level":2,"time":"2023-02-13T15:41:15+08:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"25.0.3.2","data":[]}

```
</details>

OfficialMuffin, 2023-02-15:
This is for security purposes. Root login is disabled by default. Have you made a user for SSH? You will need to login to the NextcloudPi Panel and enable SSH for that particular user in order to use SSH.

OfficialMuffin, 2023-02-15:
Related to #1669

SHA-NDY, 2023-03-11:
Same problem here on Odroid HC4. With NCP 1.51.0 I can't activate SSH as the user "root" can't be used. A new user can't be added via the NCP WebUI. An existing user "ncp" will be disconnected as soon as logged in via ssh.
The user "pi" isn't available on Odroid HC4 because that image is armbian based.

theCalcaholic, 2023-04-16:
Should be solved with 1.51.1: New users can now be created from the SSH configuration

closed 2023-04-16 by theCalcaholic

OfficialMuffin, 2023-04-16:
Great, thanks so much!