nextcloud/nextcloudpi #1733: Cannot login to nc after activating static ip address

jpilgrim, 2023-02-14:
I have created a new SSD card from  `NextCloudPi_RaspberryPi_v1.51.0.img`.
I was able to log in to nc and NextCloudPi

After booting the PI and activating nc/ncp, I changed the *Set up a static IP address* to static in NextCloudPi panel.
After that, I was not able to login to nc anymore. Changing the password with nc-admin did not help.

Actually, this is my second attempt. In the first attempt, I had the same problem but a message to change "trusted_domains" in config.php. This message did not appear in the second attempt, and the static ip is already listed there.

So, how to reproduce: 
1. Create ssd from NextCloudPi_RaspberryPi_v1.51.0.img`
2. Start system, activate etc.
3. Set up a static IP address to some static IP

<details>
<summary>Set static IP address output</summary>
```
[ nc-static-IP ] (Tue Feb 14 21:17:31 GMT 2023)
System config value trusted_domains => 1 set to string 192.168.178.200
warning: No domain found. Defaulting to 'nextcloudpi'
System config value trusted_domains => 3 set to string nextcloudpi
System config value overwrite.cli.url set to string https://nextcloudpi/
System config value trusted_proxies => 11 set to string 127.0.0.1
System config value trusted_proxies => 12 set to string ::1
System config value trusted_proxies => 13 set to string nextcloudpi
System config value trusted_proxies => 14 set to string 192.168.178.28
192.168.178.200
Setup notify_push (attempt 1/5)
✓ redis is configured
✓ push server is receiving redis messages
✓ push server is receiving redis messages
✓ push server can load mount info from database
✓ push server can connect to the Nextcloud server
✓ push server is a trusted proxy
✓ push server is running the same version as the app
configuration saved
Static IP set to 192.168.178.200
```
</details>

**System information**

<details>
<summary>NextcloudPi diagnostics</summary>

```

NextcloudPi version  v1.51.0
NextcloudPi image    NextCloudPi_RaspberryPi_v1.51.0.img
OS                   Debian GNU/Linux 11. 5.15.61-v8+ (aarch64)
automount            no
USB devices          none
datadir              /opt/ncdata/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      2.5G/59G
rootfs usage         2.5G/59G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    25.0.3.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              closed
[... 93 lines cut ...]

<details>
<summary>Database logs</summary>

```

tail: cannot open '/var/log/mysql/*.log' for reading: No such file or directory

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"jCKHe6O8u4wnpTCCkYd3","level":3,"time":"2023-01-22T01:55:01+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Not installed","userAgent":"--","version":"","exception":{"Exception":"Exception","Message":"Not installed","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/base.php","line":680,"function":"checkInstalled","class":"OC","type":"::"},{"file":"/var/www/nextcloud/lib/base.php","line":1144,"function":"init","class":"OC","type":"::"},{"file":"/var/www/nextcloud/cron.php","line":43,"args":["/var/www/nextcloud/lib/base.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/base.php","Line":284,"CustomMessage":"--"}}
{"reqId":"hietpzdjSBJdaxhEho5e","level":3,"time":"2023-01-22T01:57:04+00:00","remoteAddr":"","user":"ncp","app":"PHP","method":"","url":"--","message":"exif_read_data(): File not supported at /var/www/nextcloud/lib/private/Metadata/Provider/ExifProvider.php#59","userAgent":"--","version":"25.0.3.2","exception":{"Exception":"Error","Message":"exif_read_data(): File not supported at /var/www/nextcloud/lib/private/Metadata/Provider/ExifProvider.php#59","Code":0,"Trace":[{"function":"onError","class":"OC\\Log\\ErrorHandler","type":"::"},{"file":"/var/www/nextcloud/lib/private/Metadata/Provider/ExifProvider.php","line":59,"function":"exif_read_data"},{"file":"/var/www/nextcloud/lib/private/Metadata/MetadataManager.php","line":68,"function":"execute","class":"OC\\Metadata\\Provider\\ExifProvider","type":"->"},{"file":"/var/www/nextcloud/lib/private/Metadata/FileEventListener.php","line":101,"function":"generateMetadata","class":"OC\\Metadata\\MetadataManager","type":"->"},{"file":"/var/www/nextcloud/lib/private/EventDispatcher/ServiceEventListener.php","line":87,"function":"handle","class":"OC\\Metadata\\FileEventListener","type":"->"},{"file":"/var/www/nextcloud/3rdparty/symfony/event-dispatcher/EventDispatcher.php","line":251,"function":"__invoke","class":"OC\\EventDispatcher\\ServiceEventListener","type":"->"},{"file":"/var/www/nextcloud/3rdparty/symfony/event-dispatcher/EventDispatcher.php","line":73,"function":"callListeners","class":"Symfony\\Component\\EventDispatcher\\EventDispatcher","type":"->"},{"file":"/var/www/nextcloud/lib/private/EventDispatcher/EventDispatcher.php","line":88,"function":"dispatch","class":"Symfony\\Component\\EventDispatcher\\EventDispatcher","type":"->"},{"file":"/var/www/nextcloud/lib/private/EventDispatcher/EventDispatcher.php","line":100,"function":"dispatch","class":"OC\\EventDispatcher\\EventDispatcher","type":"->"},{"file":"/var/www/nextcloud/lib/private/Files/Node/HookConnector.php","line":118,"function":"dispatchTyped","class":"OC\\EventDispatcher\\EventDispatcher","type":"->"},{"file":"/var/www/nextcloud/lib/private/legacy/OC_Hook.php","line":106,"function":"postWrite","class":"OC\\Files\\Node\\HookConnector","type":"->"},{"file":"/var/www/nextcloud/lib/private/Files/View.php","line":641,"function":"emit","class":"OC_Hook","type":"::"},{"file":"/var/www/nextcloud/lib/private/Files/View.php","line":692,"function":"emit_file_hooks_post","class":"OC\\Files\\View","type":"->"},{"file":"/var/www/nextcloud/lib/private/Files/Node/File.php","line":71,"function":"file_put_contents","class":"OC\\Files\\View","type":"->"},{"file":"/var/www/nextcloud/lib/private/legacy/OC_Util.php","line":257,"function":"putContent","class":"OC\\Files\\Node\\File","type":"->"},{"file":"/var/www/nextcloud/lib/private/legacy/OC_Util.php","line":216,"function":"copyr","class":"OC_Util","type":"::"},{"file":"/var/www/nextcloud/lib/private/User/Session.php","line":553,"function":"copySkeleton","class":"OC_Util","type":"::"},{"file":"/var/www/nextcloud/lib/private/User/Session.php","line":403,"function":"prepareUserLogin","class":"OC\\User\\Session","type":"->"},{"file":"/var/www/nextcloud/lib/private/User/Session.php","line":616,"function":"completeLogin","class":"OC\\User\\Session","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/lib/private/User/Session.php","line":355,"function":"loginWithPassword","class":"OC\\User\\Session","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/lib/private/Setup.php","line":433,"function":"login","class":"OC\\User\\Session","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/core/Command/Maintenance/Install.php","line":105,"function":"install","class":"OC\\Setup","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/3rdparty/symfony/console/Command/Command.php","line":255,"function":"execute","class":"OC\\Core\\Command\\Maintenance\\Install","type":"->"},{"file":"/var/www/nextcloud/3rdparty/symfony/console/Application.php","line":1009,"function":"run","class":"Symfony\\Component\\Console\\Command\\Command","type":"->"},{"file":"/var/www/nextcloud/3rdparty/symfony/console/Application.php","line":273,"function":"doRunCommand","class":"Symfony\\Component\\Console\\Application","type":"->"},{"file":"/var/www/nextcloud/3rdparty/symfony/console/Application.php","line":149,"function":"doRun","class":"Symfony\\Component\\Console\\Application","type":"->"},{"file":"/var/www/nextcloud/lib/private/Console/Application.php","line":213,"function":"run","class":"Symfony\\Component\\Console\\Application","type":"->"},{"file":"/var/www/nextcloud/console.php","line":100,"function":"run","class":"OC\\Console\\Application","type":"->"},{"file":"/var/www/nextcloud/occ","line":11,"args":["/var/www/nextcloud/console.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/private/Log/ErrorHandler.php", [... cut ...]
{"reqId":"fIemf2PKbuxhXOBpdDae","level":2,"time":"2023-01-22T01:59:58+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"25.0.3.2","data":[]}
{"reqId":"fIemf2PKbuxhXOBpdDae","level":2,"time":"2023-01-22T01:59:58+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"25.0.3.2","data":[]}

```
</details>

OfficialMuffin, 2023-02-15:
Can you not connect to the new IP at all? What exactly happens when you type in `https://192.168.178.200`?

jpilgrim, 2023-02-15:
I can connect to the new IP via SSH and via the NectCloutPi Panel. The only thing that doesn't work anymore is log into nc.

OfficialMuffin, 2023-02-15:
What's the browser error when attempting to connect to nc?

jpilgrim, 2023-02-15:
There is no browser error, I just cannot login nextcloud.

jpilgrim, 2023-02-15:
Forget it. 

I tried again, this time setting up Debian manually (starting from the Raspberry Pi OS image created via Raspberry Pi Imager). I had configured the static IP manually. Then installed NextcloudPi via the curl-script. (oh, fixed some locale problem before...).

But I'm deleting that NextCloudPi installation now. Bugs over bugs. Starts with missing logo (which does not really build up trust) and ends with a corrupt calendar application which is a show stopper: I imported some calendar data, and it looked nice -- but most entries were gone after login/logout. And it is super slow (compared to a previous version I had installed manually which was running nicely on my Pi 3 Model B+. That one uses nginx instead of Apache -- although I'm not quite sure whether that is the problem).

Will give it another try in a year ;-)

closed 2023-02-15 by jpilgrim

OfficialMuffin, 2023-02-15:
My installation on an Raspberry Pi 4B 4GB is running smoothly with no issues after a clean install using the pebuilt image on Github.

oskar2517, 2023-02-19:
Having the exact same issue on my Raspberry Pi 4B 4GB. I have tried both flashing the image and using the installation script with the same result.