nextcloud/nextcloudpi #1734: Icon missing on Nextcloud v25

kojid0, 2023-02-16:
Seems like the icon at the left side of the top bar is somehow gone. Fresh install with latest versions
[image]

Schniefel, 2023-02-17:
I jad the same problem

REAPERSbattlecry, 2023-02-20:
The reason is that the way the logo is used changed.
The logo can be changed at settings > Design > Logo
This was used only for the login page. With NC25 it is used for the login page and the logo in the left upper corner.

nenderlein, 2023-02-20:
I don't know why, but uninstalling php-imagick / php8.1-imagick also fixed the issue for me.

kojid0, 2023-02-21:
> The reason is that the way the logo is used changed. The logo can be changed at settings > Design > Logo This was used only for the login page. With NC25 it is used for the login page and the logo in the left upper corner.

Thanks for the information. I just changed the logo by uploading a picture at settings > Design > Logo, then reset the logo to the default. Now I see the correct "original" logo.

I keep the issue open as this is only a workaround.

schoetju, 2023-02-24:
There has been a similar issue reported as part of nextcloud/server itself
https://github.com/nextcloud/server/issues/35871
so i don't think this is related to Nextcloud**Pi**

svenb1234, 2023-02-25:
You can just go to the admin's settings in the web ui and reset the logo and background to default, i.e. not ncp.

theCalcaholic, 2023-04-08:
This should now be fixed in version v1.51.1

closed 2023-04-08 by theCalcaholic