nextcloud/nextcloudpi #1736: Update to minor release 25.0.4

svenb1234, 2023-02-25:
Update to 25.0.4 went fine on the docker image.

Only culprit is that HPB is down, but that was already the case before the update:
:/# ncc notify_push:self-test
✓ redis is configured
✓ push server is receiving redis messages
✓ push server can load mount info from database
✓ push server can connect to the Nextcloud server
✓ push server is a trusted proxy
🗴 push server (version 0.5.2) is not the same version as the app (version 0.6.0).

Schniefel, 2023-03-01:
Worked without any problem for me. Everything what was working before (e.g. HPB) is still working now on 25.0.4

svenb1234, 2023-03-01:
Docker installation?

Schniefel, 2023-03-01:
> Docker installation?

No, normal installation using the curl script on Raspberry Pi OS.

svenb1234, 2023-03-01:
Ok, so maybe the HPB mismatch is a docker issue.

REAPERSbattlecry, 2023-03-22:
Curl installation here as well. No issues so far.
The HPB was up an running after update. Clear OP-cache was necessary.

svenb1234, 2023-03-30:
> @svenb1234 Does it resolve over time? A restart of the notify_push service should resolve the issue. That should happen once per day if not triggered outside of that schedule.

Seems so:
```
/# ncc notify_push:self-test
✓ redis is configured
✓ push server is receiving redis messages
✓ push server can load mount info from database
✓ push server can connect to the Nextcloud server
✓ push server is a trusted proxy
✓ push server is running the same version as the app
```

svenb1234, 2023-04-03:
The update to 25.0.5. seems to have worked flawlessly as well.

lordz0rn, 2023-04-05:
> A [critical security bug](https://github.com/nextcloud/security-advisories/security/advisories/GHSA-h3c9-cmh8-7qpj) was fixed in version 25.0.4, so it should really be a top priority to release the update to NextcloudPi users.

Hey there,

I also was a little startled that the nc-update-nextcloud tool in the ncp webadmin didnt „know“ that there is an update 25.0.4 when searching for version „0“, as I always did.
But you can put in „25.0.4“ instead of „0“ and the Update will be performed as intended.

Hope this helps.

sibu7, 2023-04-05:
> 

Thanks, I already did this. But beginner users might not know and I think the NextcloudPi team should "release" Nextcloud version 25.0.4 or 25.0.5 as the default version "0" as soon as possible, given it is a critical security bug.

sibu7, 2023-04-10:
@theCalcaholic I think this issue can be closed, as NCP v1.51.1 was released.

closed 2023-04-10 by theCalcaholic