nextcloud/nextcloudpi #1737: Duck DNS stopped working

coreybruce, 2023-02-26:
My DuckDNS domain stopped working, I checked the public domain, I renewed the key and I even deactivated and re activated duckdns and it still won't work but the local IP address 

<details>
<summary>NextcloudPi diagnostics</summary>

```

NextcloudPi version  v1.51.0
NextcloudPi image    NextCloudPi_RaspberryPi_v1.51.0.img
OS                   Debian GNU/Linux 11. 5.15.84-v8+ (aarch64)
automount            yes
USB devices          sda 
datadir              /media/myCloudDrive/ncdata/data
data in SD           no
data filesystem      btrfs
data disk usage      18G/1.9T
rootfs usage         4.4G/59G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    25.0.3.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
Internet check       no
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              open
Port 443             open
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         yes
Uptime               14:50

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "7": "nextcloudpi",
            "5": "nextcloudpi.local",
            "8": "nextcloudpi.lan",
            "3": "nextcloudpi",
            "11": "49.177.44.178",
            "1": "192.168.0.24",
            "14": "nextcloudpi",
            "20": "coreynextcloud.duckdns.org",
            "2": "coreynextcloud.duckdns.org"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "25.0.3.2",
        "overwrite.cli.url": "https:\/\/coreynextcloud.duckdns.org\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
[... 77 lines cut ...]
```

{"reqId":"buVhqxaorxAQDdFIJ9OW","level":4,"time":"2023-02-21T02:55:51+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Could not boot notify_push: No such file or directory","userAgent":"--","version":"25.0.3.2","exception":{"Exception":"RedisException","Message":"No such file or directory","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/RedisFactory.php","line":137,"function":"pconnect","class":"Redis","type":"->"},{"file":"/var/www/nextcloud/lib/private/RedisFactory.php","line":178,"function":"create","class":"OC\\RedisFactory","type":"->"},{"file":"/var/www/nextcloud/apps/notify_push/lib/AppInfo/Application.php","line":62,"function":"getInstance","class":"OC\\RedisFactory","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":162,"function":"OCA\\NotifyPush\\AppInfo\\{closure}","class":"OCA\\NotifyPush\\AppInfo\\Application","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php","line":122,"function":"OC\\AppFramework\\Utility\\{closure}","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":129,"function":"offsetGet","class":"Pimple\\Container","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/DependencyInjection/DIContainer.php","line":459,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/DependencyInjection/DIContainer.php","line":437,"function":"queryNoFallback","class":"OC\\AppFramework\\DependencyInjection\\DIContainer","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":89,"function":"query","class":"OC\\AppFramework\\DependencyInjection\\DIContainer","type":"->"},{"function":"OC\\AppFramework\\Utility\\{closure}","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":108,"function":"array_map"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":116,"function":"buildClass","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":133,"function":"resolve","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/DependencyInjection/DIContainer.php","line":465,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/DependencyInjection/DIContainer.php","line":437,"function":"queryNoFallback","class":"OC\\AppFramework\\DependencyInjection\\DIContainer","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":57,"function":"query","class":"OC\\AppFramework\\DependencyInjection\\DIContainer","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Bootstrap/FunctionInjector.php","line":50,"function":"get","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"function":"OC\\AppFramework\\Bootstrap\\{closure}","class":"OC\\AppFramework\\Bootstrap\\FunctionInjector","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Bootstrap/FunctionInjector.php","line":67,"function":"array_map"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Bootstrap/BootContext.php","line":51,"function":"injectFn","class":"OC\\AppFramework\\Bootstrap\\FunctionInjector","type":"->"},{"file":"/var/www/nextcloud/apps/notify_push/lib/AppInfo/Application.php","line":70,"function":"injectFn","class":"OC\\AppFramework\\Bootstrap\\BootContext","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Bootstrap/Coordinator.php","line":190,"function":"boot","class":"OCA\\NotifyPush\\AppInfo\\Application","type":"->"},{"file":"/var/www/nextcloud/lib/private/legacy/OC_App.php","line":208,"function":"bootApp","class":"OC\\AppFramework\\Bootstrap\\Coordinator","type":"->"},{"file":"/var/www/nextcloud/lib/private/legacy/OC_App.php","line":141,"function":"loadApp","class":"OC_App","type":"::"},{"file":"/var/www/nextcloud/lib/private/Console/Application.php","line":120,"function":"loadApps","class":"OC_App","type":"::"},{"file":"/var/www/nextcloud/console.php","line":99,"function":"loadCommands","class":"OC\\Console\\Application","type":"->"},{"file":"/var/www/nextcloud/occ","line":11,"args":["/var/www/nextcloud/console.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/private/RedisFactory.php","Line":137,"message":"Could not boot notify_push: No such file or directory","exception":{},"CustomMessage":"Could not boot notify_push: No such file or directory"}}
{"reqId":"buVhqxaorxAQDdFIJ9OW","level":4,"time":"2023-02-21T02:55:51+00:00","remoteAddr":"","user":"--","app":"oauth2","method":"","url":"--","message":"Error during app loading: Redis server /var/run/redis/redis.sock:0 went away","userAgent":"--","version":"25.0.3.2","exception":{"Exception":"RedisException","Message":"Redis server /var/run/redis/redis.sock:0 went away","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/Memcache/Redis.php","line":48,"function":"get","class":"Redis","type":"->"},{"file":"/var/www/nextcloud/lib/autoloader.php","line":154,"function":"get","class":"OC\\Memcache\\Redis","type":"->"},{"function":"load","class":"OC\\Autoloader","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Bootstrap/Coordinator.php","line":207,"function":"class_exists"},{"file":"/var/www/nextcloud/lib/private/legacy/OC_App.php","line":173,"function":"isBootable","class":"OC\\AppFramework\\Bootstrap\\Coordinator","type":"->"},{"file":"/var/www/nextcloud/lib/private/legacy/OC_App.php","line":141,"function":"loadApp","class":"OC_App","type":"::"},{"file":"/var/www/nextcloud/lib/private/Console/Application.php","line":120,"function":"loadApps","class":"OC_App","type":"::"},{ [... cut ...] loud/console.php","line":99,"function":"loadCommands","class":"OC\\Console\\Application","type":"->"},{"file":"/var/www/nextcloud/occ","line":11,"args":["/var/www/nextcloud/console.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/private/Memcache/Redis.php","Line":48,"message":"Error during app loading: Redis server /var/run/redis/redis.sock:0 went away","exception":{},"CustomMessage":"Error during app loading: Redis server /var/run/redis/redis.sock:0 went away"}}

```
</details>

coreybruce, 2023-02-26:
Figured it out, for some reason my router wasn't letting it push to the domain and restarting the router fixed it, is there anything in the logs showing any issues at all?

ZendaiOwl, 2023-02-26:
I don't know, nothing in particular that I can see at a glance at least

Internet check shows `no` while ports show as `open` which is interesting, there are also logs of losing the connection to the redis database, however I will venture out and do a guess that these appeared when a reboot was done perhaps? 

Is it working for you now?

coreybruce, 2023-02-27:
yeah that is weird

It is all working now but I have no idea what was going on or why my router was doing this even tho it was port forwarded and setup all correctly.

closed 2023-02-27 by coreybruce