nextcloud/nextcloudpi #1740: Redis exception - WRONGPASS invalid username-password pair / NOAUTH Authentication required.

Tchiimy, 2023-02-28:
Hello,

I'm using the docker version of NextcloudPi on my RPI4 (OS 64 bits), I use an external HDD to store my files. Everything was well, but recently I have troubles to launch/use it and I don't know why…

When I launch the docker, my logs are this:

```
02/28/2023 11:16:34 AM
Install default configurations for new ncp apps
02/28/2023 11:16:35 AM
Installing template 'php/opcache.ini.sh'...
02/28/2023 11:16:38 AM
Installing template 'php/90-ncp.ini.sh'...
02/28/2023 11:16:44 AM
Installing template 'php/pool.d.www.conf.sh'...
02/28/2023 11:16:49 AM
Installing template 'ncp-metrics.cfg.sh'...
02/28/2023 11:17:05 AM
Starting PHP-fpm
02/28/2023 11:17:06 AM
Starting Apache
02/28/2023 11:17:06 AM
Installing template 'mysql/90-ncp.cnf.sh'...
02/28/2023 11:17:09 AM
INFO: Docker container detected.
02/28/2023 11:17:09 AM
Starting mariaDB
02/28/2023 11:17:09 AM
2023-02-28 10:17:09 0 [Note] mysqld (mysqld 10.5.18-MariaDB-0+deb11u1) starting as process 457 ...
02/28/2023 11:17:09 AM
2023-02-28 10:17:09 0 [Warning] The parameter innodb_file_format is deprecated and has no effect. It may be removed in future releases. See https://mariadb.com/kb/en/library/xtradbinnodb-file-format/
02/28/2023 11:17:09 AM
2023-02-28 10:17:09 0 [Warning] The parameter innodb_buffer_pool_instances is deprecated and has no effect.
02/28/2023 11:17:09 AM
2023-02-28 10:17:09 0 [Note] InnoDB: Uses event mutexes
02/28/2023 11:17:09 AM
2023-02-28 10:17:09 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
02/28/2023 11:17:09 AM
2023-02-28 10:17:09 0 [Note] InnoDB: Number of pools: 1
02/28/2023 11:17:09 AM
2023-02-28 10:17:09 0 [Note] InnoDB: Using ARMv8 crc32 instructions
02/28/2023 11:17:09 AM
2023-02-28 10:17:09 0 [Note] mysqld: O_TMPFILE is not supported on /tmp (disabling future attempts)
02/28/2023 11:17:09 AM
2023-02-28 10:17:09 0 [Note] InnoDB: Using Linux native AIO
02/28/2023 11:17:09 AM
2023-02-28 10:17:09 0 [Note] InnoDB: Initializing buffer pool, total size = 268435456, chunk size = 134217728
02/28/2023 11:17:09 AM
2023-02-28 10:17:09 0 [Note] InnoDB: Completed initialization of buffer pool
02/28/2023 11:17:09 AM
2023-02-28 10:17:09 0 [Note] InnoDB: Starting crash recovery from checkpoint LSN=10866669497,10866669497
02/28/2023 11:17:09 AM
2023-02-28 10:17:09 0 [Note] InnoDB: Starting final batch to recover 194 pages from redo log.
02/28/2023 11:17:10 AM
2023-02-28 10:17:10 0 [Note] InnoDB: 128 rollback segments are active.
02/28/2023 11:17:10 AM
2023-02-28 10:17:10 0 [Note] InnoDB: Removed temporary tablespace data file: "ibtmp1"
02/28/2023 11:17:10 AM
2023-02-28 10:17:10 0 [Note] InnoDB: Creating shared tablespace for temporary tables
02/28/2023 11:17:10 AM
2023-02-28 10:17:10 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wait ...
02/28/2023 11:17:10 AM
2023-02-28 10:17:10 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
02/28/2023 11:17:10 AM
2023-02-28 10:17:10 0 [Note] InnoDB: 10.5.18 started; log sequence number 10867225184; transaction id 19391101
02/28/2023 11:17:10 AM
2023-02-28 10:17:10 0 [Note] Plugin 'FEEDBACK' is disabled.
02/28/2023 11:17:10 AM
2023-02-28 10:17:10 0 [Note] InnoDB: Loading buffer pool(s) from /data/database/ib_buffer_pool
02/28/2023 11:17:10 AM
[... 517 lines cut ...]
> 02/28/2023 11:36:58 AM
> 
>     app=coreexception.Code=0exception.CustomMessage=--exception.Exception=RedisExceptionexception.File=/data/nextcloud/lib/private/Memcache/Redis.phpexception.Line=48exception.Message=NOAUTH Authentication required.exception.Trace=[object Object],[object Object],[object Object],[object Object],[object Object],[object Object],[object Object],[object Object],[object Object],[object Object],[object Object],[object Object]level=3message=NOAUTH Authentication required.method=GETremoteAddr=172.17.0.1reqId=Y_3ZSqQFb2Hl8CozYzbMewAAAAEtime=2023-02-28T10:36:58+00:00url=/user=--userAgent=axios/0.27.2version=24.0.5.1
> 
>     add app=core
>     exception=
>         add Code=0
>         add CustomMessage=--
>         add Exception=RedisException
>         add File=/data/nextcloud/lib/private/Memcache/Redis.php
>         add Line=48
>         add Message=NOAUTH Authentication required.
>         add Trace=[ { "class": "Redis", "file": "/data/nextcloud/lib/private/Memcache/Redis.php", "function": "get", "line": 48, "type": "->" } ,{ "class": "OC\\Memcache\\Redis", "file": "/data/nextcloud/lib/private/App/InfoParser.php", "function": "get", "line": 58, "type": "->" } ,{ "class": "OC\\App\\InfoParser", "file": "/data/nextcloud/lib/private/App/AppManager.php", "function": "parse", "line": 506, "type": "->" } ,{ "class": "OC\\App\\AppManager", "file": "/data/nextcloud/lib/private/App/AppManager.php", "function": "getAppInfo", "line": 521, "type": "->" } ,{ "class": "OC\\App\\AppManager", "file": "/data/nextcloud/lib/private/legacy/OC_App.php", "function": "getAppVersion", "line": 568, "type": "->" } ,{ "class": "OC_App", "file": "/data/nextcloud/lib/private/legacy/OC_App.php", "function": "getAppVersion", "line": 891, "type": "::" } ,{ "class": "OC_App", "file": "/data/nextcloud/lib/private/legacy/OC_Util.php", "function": "shouldUpgrade", "line": 1225, "type": "::" } ,{ "class": "OC_Util", "file": "/data/nextcloud/lib/public/Util.php", "function": "needUpgrade", "line": 579, "type": "::" } ,{ "class": "OCP\\Util", "file": "/data/nextcloud/lib/private/legacy/OC_Template.php", "function": "needUpgrade", "line": 116, "type": "::" } ,{ "class": "OC_Template", "file": "/data/nextcloud/lib/private/legacy/OC_Template.php", "function": "initTemplateEngine", "line": 78, "type": "::" } ,{ "class": "OC_Template", "file": "/data/nextcloud/lib/private/legacy/OC_Template.php", "function": "__construct", "line": 324, "type": "->" } ,{ "class": "OC_Template", "file": "/data/nextcloud/index.php", "function": "printExceptionErrorPage", "line": 74, "type": "::" } ] 
>     add level=3
>     add message=NOAUTH Authentication required.
>     add method=GET
>     add remoteAddr=172.17.0.1
>     add reqId=Y_3ZSqQFb2Hl8CozYzbMewAAAAE
>     add time=2023-02-28T10:36:58+00:00
>     add url=/
>     add user=--
>     add userAgent=axios/0.27.2
>     add version=24.0.5.1

I don't really know what could be the cause, someone has an idea ? I maybe presented the logs badly, is there a better to present them to facilitate the understanding ?

feutl, 2023-03-13:
I need to back this, this started to happen on my docker installation too and have no clue why.
Any updated on this?

feutl, 2023-03-22:
Any news on this, a hint, just something to start debugging?
It is really annoying and effect the reliability of nextcloud.

Thanks

twojstaryzdomu, 2023-03-23:
Outside of docker, these issues had to do with failing to restart redis if the configuration (redis.cfg or config.php) has been updated with fresh credentials. It may be easily avoided, comparing the redis systemd start time to the config file modification time and acting as needed. See:
https://github.com/twojstaryzdomu/nextcloudpi/commit/68bc7ee173591d339b9f7f2efdc2e02fab732286

feutl, 2023-03-23:
For me, in NCP-Docker this (https://github.com/nextcloud/nextcloudpi/issues/1652#issuecomment-1445141639) fixed my problem of seeing auth errors with redis.
Adding the user "default" into the config.php redis configuration.

feutl, 2023-03-28:
I tried to get around this again, after a reboot of my NCP docker I am having the issues again. with or without the redis default user. Also tried to swith of protectionmode - same problem. I am clueless, but it makes more and more problems.

Any help appreciated.

feutl, 2023-03-28:
I did some more testing, and actually not the default user fixed the problem, but restarting php after the docker instance was booted. It seems that the password change, which is done on every restart for the docker installation is propagated to the redis config, but not all services are rebooted afterwards.

For now I do a restart of redis and php and the problems on my NCP docker instance disappeared.
`service redis-server restart`
`service php8.1-fpm restart`

Tommy-01, 2023-04-15:
> I did some more testing, and actually not the default user fixed the problem, but restarting php after the docker instance was booted. It seems that the password change, which is done on every [...]

Thank you! That solved the Problem for me.
Will there be an permanent solution?

feutl, 2023-04-18:
This is something we need to ask @theCalcaholic :)