nextcloud/nextcloudpi #1744: E-mail for security issue

REAPERSbattlecry, 2023-03-15:
The github issue categories recommend to contact nacho in case of security issues by mail. 
Does it make sense to change that mail to a more general one or even to one of @theCalcaholic 
[image]

theCalcaholic, 2023-03-15:
Good find, thank you

joshtrichards, 2025-02-04:
If someone wants their first PR... the change needs to happen in the template:

https://github.com/nextcloud/nextcloudpi/blob/master/.github/ISSUE_TEMPLATE/1_Bug_report.md?plain=1

I suggest pointing at the same resource as the GH Security Policy link:

https://github.com/nextcloud/.github/blob/master/SECURITY.md