nextcloud/nextcloudpi #1746: Errors in curl installer on RPi / Debian Bullseye

luxzg, 2023-03-17:
**System information**

<--! Paste this in GitHub report -->

<details>
<summary>NextcloudPi diagnostics</summary>

```

NextcloudPi version  v1.51.0
NextcloudPi image    NextcloudPi_03-17-23
OS                   Debian GNU/Linux 11. 5.15.84-v8+ (aarch64)
automount            no
USB devices          none
datadir              /opt/ncdata/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      2.8G/3.4G
rootfs usage         2.8G/3.4G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    25.0.3.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              closed
Port 443             closed
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               50min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "7": "nextcloudpi",
            "5": "nextcloudpi.local",
            "8": "nextcloudpi.lan",
            "3": "nextcloudpi",
            "11": "178.17.116.16",
            "1": "192.168.0.13",
            "14": "nextcloudpi"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "25.0.3.2",
        "overwrite.cli.url": "https:\/\/nextcloudpi\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
[... 173 lines cut ...]
noip2.c:1838:20: warning: variable ‘y’ set but not used [-Wunused-but-set-variable]
 1838 |         int     x, y, z;
      |                    ^
noip2.c: In function ‘autoconf’:
noip2.c:2538:19: warning: ‘%s’ directive writing up to 255 bytes into a region of size 247 [-Wformat-overflow=]
 2538 |  sprintf(line, "%s%s%s%s", USTRNG, login, PWDSTRNG, password);
      |                   ^~               ~~~~~
noip2.c:2538:2: note: ‘sprintf’ output between 16 and 526 bytes into a destination of size 256
 2538 |  sprintf(line, "%s%s%s%s", USTRNG, login, PWDSTRNG, password);
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/tmp/nextcloudpi.UVDIce/nextcloudpi
Installing samba
...
```

Now this looks like more serious, and I can see it's actual error when doing make on external project.
Script in question is:
https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/NETWORKING/no-ip.sh
line: 19 : `wget -O- --content-disposition https://github.com/nachoparker/noip-DDNS/archive/master/latest.tar.gz \`
then error gets triggered by line 23 `make`

I've tried manually downloading said tar file, extracting, and it threw same error on `make`. But that github repo is 5 years old.
I haven't tried but I guess "no-ip" functionality would be broken. Is there a fix? Better substitution?

Anyway, I hope some of these errors are useful. I've captured whole SSH session if it's needed, and I can replicate same on my Pi. I'm still just playing and trying out stuff on RPi + NextcloudPi, so I can easily erase and start from scratch at this point. But I'm uncertain about using such setup for real data, it's hard for me to judge as these three could be insignificant to me at this point. I'll probably wait for Nextcloud 26 to try again anyway, but if someone can either check these or just reassure me that they're not significant errors, I'd be thankful, before setting up the Pi + NextcloudPi for "production" (home cloud). Thanks!

twojstaryzdomu, 2023-03-23:
I've likely hit the same errors you do whilst installing NC. For specific fixes, you might want to review the commits in the bookworm branch in my repo: https://github.com/twojstaryzdomu/nextcloudpi
It shouldn't be too difficult to cherry picking the commits needed to fix those bugs.
NC 26 runs nicely on Rpi3B.

luxzg, 2023-03-24:
Thanks for the infos. In the meantime I've switched my focus a little, mostly due to getting reminded that RPi3B has 100Mbps ethernet, and my library is quite huge. I've had another SFF x86 PC lying around, with more RAM and way better specs overall, and will try running NCPi or standard NC from it this weekend. I still appreciate the comments, both as a learning experience, and maybe I encounter them again on my x86 alternative.