nextcloud/nextcloudpi #1748: Since Update I have no files list

Domifry, 2023-03-22:
I did ncp-update and to the newest version and since this time it told me that I can not use things like filescan because I use PHP 8.2 - so I deleted PHP 8.2 because 8.1 was already on it. For some reason the update also installed 8.2

Since I removed it my Filelist is not visible anymore. Can you help me how to solve this problem? Thanks!
Now I installed the packages again to see if it is because missing PHP 8.2 put no change. Can you help me?

**System information**

<summary>NextcloudPi diagnostics</summary>

```

NextcloudPi version  v1.51.0
NextcloudPi image    NextCloudPi_09-14-22
OS                   Debian GNU/Linux 11. 5.15.84-v8+ (aarch64)
automount            no
USB devices          sda sdc 
datadir              /media/pi/cloud/data
data in SD           no
data filesystem      ext2/ext3
data disk usage      34G/916G
rootfs usage         7.8G/15G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              closed
Port 443             closed
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               20:18

```
</details>

[image]

<details>
<summary>Nextcloud configuration</summary>

```

This version of Nextcloud is not compatible with PHP>=8.2.<br/>You are currently running 8.2.4.
```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Wed Mar 22 19:04:55.698857 2023] [mpm_event:notice] [pid 675:tid 547796350016] AH00492: caught SIGWINCH, shutting down gracefully
[Wed Mar 22 19:04:56.879233 2023] [ssl:warn] [pid 145185:tid 548638561344] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Mar 22 19:04:56.922608 2023] [ssl:warn] [pid 145186:tid 548638561344] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Mar 22 19:04:56.931819 2023] [mpm_event:notice] [pid 145186:tid 548638561344] AH00489: Apache/2.4.56 (Debian) OpenSSL/1.1.1n configured -- resuming normal operations
[Wed Mar 22 19:04:56.931990 2023] [core:notice] [pid 145186:tid 548638561344] AH00094: Command line: '/usr/sbin/apache2'
[Wed Mar 22 19:08:29.426010 2023] [mpm_event:notice] [pid 145186:tid 548638561344] AH00492: caught SIGWINCH, shutting down gracefully
[Wed Mar 22 19:08:29.893237 2023] [ssl:warn] [pid 147734:tid 547738166336] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Mar 22 19:08:29.932799 2023] [ssl:warn] [pid 147735:tid 547738166336] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Mar 22 19:08:29.941804 2023] [mpm_event:notice] [pid 147735:tid 547738166336] AH00489: Apache/2.4.56 (Debian) OpenSSL/1.1.1n configured -- resuming normal operations
[Wed Mar 22 19:08:29.941973 2023] [core:notice] [pid 147735:tid 547738166336] AH00094: Command line: '/usr/sbin/apache2'
[Wed Mar 22 19:45:51.515249 2023] [mpm_event:notice] [pid 147735:tid 547738166336] AH00492: caught SIGWINCH, shutting down gracefully
[Wed Mar 22 19:45:52.290921 2023] [ssl:warn] [pid 154226:tid 548204299328] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Mar 22 19:45:52.358077 2023] [ssl:warn] [pid 154227:tid 548204299328] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Mar 22 19:45:52.374685 2023] [mpm_event:notice] [pid 154227:tid 548204299328] AH00489: Apache/2.4.56 (Debian) OpenSSL/1.1.1n configured -- resuming normal operations
[Wed Mar 22 19:45:52.374900 2023] [core:notice] [pid 154227:tid 548204299328] AH00094: Command line: '/usr/sbin/apache2'
[Wed Mar 22 19:51:50.259546 2023] [mpm_event:notice] [pid 154227:tid 548204299328] AH00492: caught SIGWINCH, shutting down gracefully
[Wed Mar 22 19:51:50.481824 2023] [ssl:warn] [pid 155676:tid 548057773120] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Mar 22 19:51:50.522957 2023] [ssl:warn] [pid 155677:tid 548057773120] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Mar 22 19:51:50.533394 2023] [mpm_event:notice] [pid 155677:tid 548057773120] AH00489: Apache/2.4.56 (Debian) OpenSSL/1.1.1n configured -- resuming normal operations
[Wed Mar 22 19:51:50.533596 2023] [core:notice] [pid 155677:tid 548057773120] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

tail: cannot open '/var/log/mysql/*.log' for reading: No such file or directory

```
</details>

<details>
<summary>Nextcloud logs</summary>

Domifry, 2023-03-22:
Maybe this line is the mistake? 
exif_read_data(): File not supported at /var/www/nextcloud/lib/private/Metadata/Provider/ExifProvider.php#59

Domifry, 2023-03-22:
[image]

I found the mistakes. Can you help me maybe how I can fix them?

svenb1234, 2023-03-22:
Have you tried to clear the php cache?

```
sudo bash -c 'source /usr/local/etc/library.sh && clear_opcache' 
sudo service php8.1-fpm restart
```

If this is really a bug it is an upstream bug and you would have to report it at nextcloud https://github.com/nextcloud/server

Domifry, 2023-03-23:
I solved it with: 
sudo -u www-data php /var/www/nextcloud/occ maintenance:repair
reboot

closed 2023-03-23 by Domifry