nextcloud/nextcloudpi #1750: Upload avatar goes in error

dierre72, 2023-03-24:
Hi,
I burned the nextcloudpi image into a Raspberry PI 4.
After the configuration phase I updated the SO and then browse the nextcloudpi web pages.
As first thing I updated my information and tried to upload a picture but without success, this is the error returner in nextcloud.log

{"reqId":"ZB253xYSwwshe9s1xhqS8AAACAg","level":2,"time":"2023-03-24T14:55:27+00:00","remoteAddr":"192.168.1.60","user":"ncp","app":"core","method":"POST","url":"/index.php/avatar","message":"Missing a temporary folder","userAgent":"Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/111.0","version":"25.0.3.2","data":{"app":"core"}}
{"reqId":"ZB253xYSwwshe9s1xhqS8AAACAg","level":3,"time":"2023-03-24T14:55:27+00:00","remoteAddr":"192.168.1.60","user":"ncp","app":"PHP","method":"POST","url":"/index.php/avatar","message":"File upload error - unable to create a temporary file at Unknown#0","userAgent":"Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/111.0","version":"25.0.3.2","data":{"app":"PHP"}}

I checked these files to check the Upload directory
**/etc/php/8.1/fpm/php.ini**
**/etc/php/8.1/cli/php.ini**

In the first one there is the right directory
**upload_tmp_dir = /media/myCloudDrive/ncdata/data/tmp**

in the second one the variable name is wrong and the path too
**uploadtmp_dir = /opt/ncdata/data/tmp**

I replaced with the first and it works

my ncp-report
[ncpreport.txt](https://github.com/nextcloud/nextcloudpi/files/11063744/ncpreport.txt)
-report

nixoncello, 2023-04-09:
I had a similar issue where all my uploads were failing, and all I did was a clean ncp install and one nc-datadir. The variable name/path fix in the nc-datadir.sh fixed the issue.
I made a github account just to make this comment and I don't know github so I don't know what exactly I need to do, just wanted to give a heads up on this issue

theCalcaholic, 2025-03-24:
Fix is in devel and will be released shortly, thanks to @joshtrichards

commit ce2406d2 (2025-03-24): fix(CONFIG): temp directory typo in nc-datadir.sh

Fixes #1857
Fixes #1750
closed 2025-03-24 by theCalcaholic