nextcloud/nextcloudpi #1756: No apache2 folder under /etc/php8.1

Oberlurch3000, 2023-04-09:
First: I'm not sure if this is a bug or a problem. If not I'm sorry to post here.

I've set up a new NC instance on my pi4 via curL install script. I am used to see a apache2 folder in /etc/php/ where I have to set up memory limit and opcache options for php. But the instance is working without this. How/where do I change these values with NCpi if necessary? Currently php memory limit ist set to 768M but I don't know where.

theCalcaholic, 2023-04-10:
Hi,

You can go to the admin web ui and configure the PHP memory limit through the option `nc-limits`

closed 2023-04-10 by theCalcaholic

Oberlurch3000, 2023-04-16:
Ok, where are those settings stored? I'm just curious.

theCalcaholic, 2023-04-18:
The ncp web UI settings are stored in `/usr/local/etc/ncp-config.d` (but mostly have no effect when edited by hand). The PHP memory limit itself is configured via config files in `/etc/php/<VERSION>/fpm/`