nextcloud/nextcloudpi #1758: can not update 25.0.3.1 to 25.0.5

FadeFx, 2023-04-09:
Hi, today i tried to update from 25.0.3.1 to 25.0.5, the webinterface does not do this (btw, no output in the console, only red point at the end) and the logs then showed that PHP needs to be upgraded, which can not be done in webinterface and has to be done via console, so i ssh'ed in to my RPi4 to do the update via ncp-config, but this however tries to backup my whole instance including data, which is neither neccessary imo nor possible. 

This is my console output, as you can see it wants 4GB of space for the backup, which should actually be no problem with the normal backup location i have configured which is on my USBDrive, but it wants to store the backup in /var/www/nextcloud which is on a sdcard obviously which is limited in storage obviously...

`Running nc-update-nextcloud
Current   Nextcloud version 25.0.3.2
Available Nextcloud version 25.0.5
Download Nextcloud 25.0.5...
Back up current instance...
check free space...
free space check failed. Need 4024957751 Bytes
Maintenance mode already disabled
Error backing up
Clean up...
Done. Press any key...
`

i managed to free up enough space for the backup on the sdcard and now the install is running well as it seems, but this backup imho should respect the path set as backup location as well....

theCalcaholic, 2023-04-10:
What's being backed up is not your data, but your database and nextcloud directory. Because it is a lot faster to create and restore the backup (and it needs to work even if you never configured any backups) the backup location is based on the location of the nextcloud directory, not your configured backup location.

It's good advice to have at least 1.5 times as much free space on your root disk as your nextcloud directory + database uses

FadeFx, 2023-04-11:
I know that, and the same is being backed up during scheduled backup, but this goes to the configured backup location on USBDrive, also if I update through webinterface. However, this is not the case for the backup done when updating through ncp-config.

theCalcaholic, 2023-04-11:
the script executed by the web UI is exactly the same as the one executed by ncp-config.

Scheduled backups are different from backups during upgrades, however.

closed 2023-04-11 by theCalcaholic