nextcloud/nextcloudpi #1764: Reser server secret

Oberlurch3000, 2023-04-14:
I have set up a new nextcloud instance and restored all data and the database from an old instance (same NC version).
Everything is fine, but the passwords app can't show my passwords any more. I think this happens because the server secret in config.php has changed?

Is is possible to set back the secret, since I have a backup of the old config file?

Oberlurch3000, 2023-04-15:
Changing the values of "secret" and "passwordsalt" in config.php did not work out. I just could not access NC any more.

SOLUTION:
1. Making a backup from the current instance "/var/www/nextcloud" with cp -r and put it anywhere.
2. Delete "/var/www/nextcloud".
3. Also restore "/var/www/nextcloud" from the old instances backup.
4. Then change the values of "dbpassword" and "redis password" in config.php to the ones backed up in 1.
5. Disable maintanance mode and run occ upgrade via browser.

Everything is working and the passwords are back again.

closed 2023-04-15 by Oberlurch3000

theCalcaholic, 2023-04-15:
Thank you for sharing your solution!