nextcloud/nextcloudpi #1767: LXC Container - support of ZFS file system for data drive

hurrikan77, 2023-04-22:
Hi Team

I tried the LXC Container under Proxmox.
The installation is like all other images very easy.

After installation I want to change the path for the data folder to an second drive.
Actually the drive is formated with ZFS as file system (because it is the standard file system of Proxmox).
There is an info message the ZFS is actually not supported. 

Will there be a Support for ZFS in NextcloudPi?

Thx

theCalcaholic, 2023-04-22:
Thanks for the suggestion. I think it should be easy to add support for zfs. I'll test this and add zfs to the list if there are no issues

theCalcaholic, 2023-04-24:
I just checked myself and found the following: The container filesystem is not zfs on Proxmox. It's recognized by the guest as ext2/ext3 for me which makes me wonder: how did you add the volume to the container? :thinking: 

TLDR: As far as I understand it, ZFS support should not be required for running containers on Proxmox, even if the storage pool on the host is using ZFS

theCalcaholic, 2023-04-24:
Can you run the following commands and post their output here (replace `<path-to-ncp-data-directory>` with the actual path to the data directory)?

```sh
df -h
stat --file-system --format=%T <path-to-ncp-data-directory>
```

ZendaiOwl, 2023-04-24:
The filesystem can be ZFS, I use BTRFS on my Proxmox installation, but you can use ZFS :slightly_smiling_face: there are a couple of storage options available. 

[Proxmox Wiki - Storage][proxmox-storage]

[Proxmox Wiki - Container Settings][wiki-ct-settings]

[Proxmox VE Docs - Storage Types][pve-storage-types]

[wiki-ct-settings]: https://pve.proxmox.com/wiki/Linux_Container#pct_settings 

[proxmox-storage]: https://pve.proxmox.com/wiki/Storage

[pve-storage-types]: https://pve.proxmox.com/pve-docs/chapter-pvesm.html#_storage_types

Is there a reason why you aren't mounting and installing to the drive you want to keep the data in during creation? Or maybe I'm misunderstanding or missing something, below is some of my Proxmox storage configs as an example.

---

A container image with storage configured to my storage-pool (local-btrfs), in my case it says btrfs instead because that is what I use with my Proxmox installation, the HDD does not use BTRFS and I forgot what it uses since it was a while since I installed it as a LVM thinpool. You can move the volume here as well, if you select the drive to highlight it and then click on 'Volume Action' & then 'Move Storage' you can select another storage pool or device to use.

[image]

The available storage devices for 'Datacenter'

[image]

The mounted devices in the storage overview for the node 'lab'

[image]

I mount the storage during container creation, step 2 & 3 configures the storage options where I can choose a storage mountpoint that is available. To make a storage mountpoint available I first need to configure it in the 'Disks' section of the node in Proxmox web interface as well as the 'Datacenter' > 'Storage' tab & finally for the container as well during creation.  

Step 1

[image]

Step 2

[image]

Step 3

[image]

hurrikan77, 2023-04-24:
Hi all

to be a little bit clearer. The LXC container runs without any problem.

My setup actually is that the storage device, which is on an other hard drive, is formated as ZFS File System.
I have added a part of this disk to the LXC container as described in step 3 @ZendaiOwl . 

And then I got the message in NCP panel.

I am a little bit a newbie in Proxmox so I tried as described in some tutorials. I assume that it is possible to format the storage hard drive with btrfs and then have no problems. 

@theCalcaholic 

df -h
stat --file-system --format=%T <path-to-ncp-data-directory>

the  commands got the following results:
[image]

`export DISABLE_FS_CHECK=1
ncp-config`

This works and the datadir has been changed successfully.
I will check if all things are working.

Best regards

ZendaiOwl, 2023-04-24:
You should preferably use the same filsystem partitioning scheme you installed Proxmox with, I installed mine with btrfs, which is experimental on Proxmox

theCalcaholic, 2023-04-24:
Okay, interesting. That means that volumes in zfs storage pools are handled differently to the ones I was working with (btrfs and dir storage drivers mainly), in which the host fs would have no effect on the container volume fs.

I think, we can probably just white-list zfs if you don't encounter any problems.

I'll double check all fs checks in the code base, though

ZendaiOwl, 2023-04-24:
@theCalcaholic it's not ncp specific, it's Proxmox specific, not related to ncp

When you install Proxmox you can use different storage types such as zfs or btrfs, btrfs however is experimental for Proxmox. More info is in the articles I linked above

theCalcaholic, 2023-04-24:
Okay, if you meant just that, I disagree :D

Imo the point of Proxmox storage volumes is that you are not limited to your host configuration.

ZendaiOwl, 2023-04-24:
It's still experimental with btrfs on Proxmox :) it's only a technical preview at the moment

theCalcaholic, 2023-04-24:
I'm aware of that and I think we're missing each other's point right now. Anyways, I'll have a fresh look at it tomorrow 😁

ZendaiOwl, 2023-04-25:
> I think we're missing each other's point right now

Quite possible 😅

dkolgrre, 2023-06-25:
@theCalcaholic any updates on whitelisting zfs? I have the same challenge. I have moved the datadir using the `export DISABLE_FS_CHECK=1` trick, but this does not seem to affect moving dbdir, which I would like to do as well.

monkeyclass, 2023-06-25:
@theCalcaholic any updates on whitelisting zfs? I have the same challenge. I have moved the datadir using the export DISABLE_FS_CHECK=1 trick. However, for this to also be possible for the DBDIR I had to modify `nc-database.sh` to look at the DISABLE_FS_CHECK, which is does not by default. I will open a PR for that shortly.

monkeyclass, 2023-06-25:
#1788 https://github.com/nextcloud/nextcloudpi/pull/1788

theCalcaholic, 2023-07-17:
ZFS support is added in v1.52.1

closed 2023-07-17 by theCalcaholic