nextcloud/nextcloudpi #1773: Web interface for setting "default_phone_region"

marcuskabele, 2023-05-03:
**Description**  
On every new installation of Nextcloud the Admin Settings - Overview complain about the missing "default phone region", even with NCP. There is no GUI for this setting. You have to edit nextcloud/config/config.php by CLI.

Nextcloudpi should provide an interface to set this value, like trusted domains.

Thanks, Marcus

**Example**  
Your installation has no default phone region set. This is required to validate phone numbers in the profile settings without a country code. To allow numbers without a country code, please add "default_phone_region" with the respective ISO 3166-1 code ↗ of the region to your config file.

naldoxable, 2024-01-29:
Where is located that file? I'm not finding the folder Nextcloud.

RavanH, 2024-01-31:
> Where is located that file? I'm not finding the folder nextcloud.

Usually /var/www/nextcloud/config/config.php