nextcloud/nextcloudpi #1778: No Docker 1.52.0 image, but ncp-config within Docker asks for update

BstAA, 2023-05-17:
**System information**

When starting ncp-config within a running 1.51.4 container I get the message 
[image]

1. Why isn't a actual Docker image available?
2. to my knowledge ncp-update should not be available within Docker

Klaus

ZendaiOwl, 2023-05-17:
@theCalcaholic tagging you for visibility :) 

---

I don't know however it appears the workflow stepped up the version number for the Docker image tag instead, going from 1.51.3 to 1.51.4 instead of going to 1.52.0 like the other IMG releases

theCalcaholic, 2023-05-17:
Thank you for reporting - yes, that update should *not* be offered on docker!

I'll look into it

ChronoWerX82, 2023-06-17:
Have to renew my container and can't update to 1.52.0 with "ALLOW_UPDATE_SCRIPT=1 ncp-update"

Following output:

> ALLOW_UPDATE_SCRIPT=1 ncp-update
Downloading updates
fatal: unable to access 'https://github.com/nextcloud/nextcloudpi.git/': Could not resolve host: github.com
No internet connectivity

Ping to github from pi worked

JOHAAANNS, 2023-06-20:
Hi, same problem !
I don't find any solution :/
Thx

BstAA, 2023-06-20:
The problem has already been fixed in the current image. So workaround: * stop running container* delete container (will not delete volumes)* delete image (docker image prune)* pull current image* start container Without deleting the current image - and this can only be done without a running container - at least for my situation, I could pull a new image. Klaus —Dr. Klaus-F. Beckstette

JOHAAANNS, 2023-06-20:
Sorry, that's work : 
I do this : ALLOW_UPDATE_SCRIPT=1 ncp-update
Thx for all ;)

theCalcaholic, 2023-07-21:
Version 1.52.0 and later should not be offered anymore starting with NCP docker version 1.51.1

closed 2023-07-21 by theCalcaholic

BstAA, 2023-07-21:
> Version 1.52.0 and later should not be offered anymore starting with NCP docker version 1.51.1

Please open again, as Docker Image 1.51.5 (i.e. latest) still has got this issue.